espressif/esp-idf #4718: Very long wait for a response from the SNTP server

Eugene-Ilyushchyts, 2020-02-05:
There is a chance that the module will not receive a response from the SNTP server for long time.
Sometimes I have to wait up to 25 seconds. A delay of 10 to 15 seconds is repeated more often.
In my case, even a 12-second wait is critical.
I use sntp example from esp-idf release/v4.1.

In addition, I added the following workaround to the code example that improved the situation but didn't solve it:
```
while (sntp_get_sync_status() == SNTP_SYNC_STATUS_RESET && ++retry < retry_count)
{
    ESP_LOGI(TAG, "Waiting for system time to be set... (%d/%d)", retry, retry_count);
    vTaskDelay(1000 / portTICK_PERIOD_MS);
    if (retry % 3 == 0) {
        sntp_restart(); // Instead of initialize_sntp() you can use this function - sntp_restart().
    }
}
```
**Example of console output:**
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7452
load:0x40078000,len:14848
load:0x40080400,len:4568
entry 0x400806f0
I (29) boot: ESP-IDF v4.1-dev-1961-g73384c1-dirty 2nd stage bootloader
I (29) boot: compile time 00:22:13
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1a02c (106540) map
I (152) esp_image: segment 1: paddr=0x0002a054 vaddr=0x3ffb0000 size=0x03974 ( 14708) load
I (158) esp_image: segment 2: paddr=0x0002d9d0 vaddr=0x40080000 size=0x00400 (  1024) load
I (159) esp_image: segment 3: paddr=0x0002ddd8 vaddr=0x40080400 size=0x02240 (  8768) load
I (171) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x7b0f0 (504048) map
I (357) esp_image: segment 5: paddr=0x000ab118 vaddr=0x40082640 size=0x13d04 ( 81156) load
I (391) esp_image: segment 6: paddr=0x000bee24 vaddr=0x400c0000 size=0x00064 (   100) load
I (391) esp_image: segment 7: paddr=0x000bee90 vaddr=0x50000000 size=0x00004 (     4) load
I (411) boot: Loaded app from partition at offset 0x10000
I (411) boot: Disabling RNG early entropy source...
I (412) cpu_start: Pro cpu up.
I (416) cpu_start: Application information:
I (420) cpu_start: Project name:     sntp
I (425) cpu_start: App version:      1
I (429) cpu_start: Compile time:     Feb  5 2020 00:22:20
I (436) cpu_start: ELF file SHA256:  83c6a812cc29c05d...
I (442) cpu_start: ESP-IDF:          v4.1-dev-1961-g73384c1-dirty
I (448) cpu_start: Starting app cpu, entry point is 0x40081278
I (0) cpu_start: App cpu up.
I (459) heap_init: Initializing. RAM available for dynamic allocation:
I (466) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (472) heap_init: At 3FFB92C0 len 00026D40 (155 KiB): DRAM
I (478) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (484) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (491) heap_init: At 40096344 len 00009CBC (39 KiB): IRAM
I (497) cpu_start: Pro cpu start user code
I (515) spi_flash: detected chip: gd
I (516) spi_flash: flash io: dio
W (516) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (526) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (537) example: Boot count: 1
I (537) example: Time is not set yet. Connecting to WiFi and getting time over NTP.
I (597) wifi: wifi driver task: 3ffc0e90, prio:23, stack:3584, core=0
I (597) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (597) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (627) wifi: wifi firmware version: f1dc391
I (627) wifi: config NVS flash: enabled
I (627) wifi: config nano formating: disabled
I (627) wifi: Init dynamic tx buffer num: 32
I (637) wifi: Init data frame dynamic rx buffer num: 32
I (637) wifi: Init management frame dynamic rx buffer num: 32
I (647) wifi: Init management short buffer num: 32
I (647) wifi: Init static rx buffer size: 1600
I (657) wifi: Init static rx buffer num: 10
I (657) wifi: Init dynamic rx buffer num: 32
I (657) example_connect: Connecting to Butternut...
I (757) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (757) wifi: mode : sta (bc:dd:c2:cf:83:28)
I (767) example_connect: Waiting for IP
I (1367) wifi: new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (2137) wifi: state: init -> auth (b0)
I (2137) wifi: state: auth -> assoc (0)
I (2147) wifi: state: assoc -> run (10)
I (2157) wifi: connected with Butternut, aid = 9, channel 6, BW20, bssid = 4a:a4:72:3b:fd:fd
I (2157) wifi: security type: 3, phy: bgn, rssi: -29
I (2157) wifi: pm start, type: 1

I (2167) wifi: AP's beacon interval = 102400 us, DTIM period = 3
I (3087) esp_netif_handlers: sta ip: 192.168.137.176, mask: 255.255.255.0, gw: 192.168.137.1
I (3087) example_connect: Got IP event!
I (3587) example_connect: Got IPv6 event!
I (3587) example_connect: Connected to Butternut
I (3587) example_connect: IPv4 address: 192.168.137.176
I (3587) example_connect: IPv6 address: fe80:0000:0000:0000:bedd:c2ff:fecf:8328
I (3597) example: Initializing SNTP
I (3607) example: Waiting for system time to be set... (1/25)
I (4607) example: Waiting for system time to be set... (2/25)
I (5607) example: Waiting for system time to be set... (3/25)
I (6607) example: Waiting for system time to be set... (4/25)
I (7607) example: Waiting for system time to be set... (5/25)
I (8607) example: Waiting for system time to be set... (6/25)
I (9607) example: Waiting for system time to be set... (7/25)
I (10607) example: Waiting for system time to be set... (8/25)
I (11607) example: Waiting for system time to be set... (9/25)
I (12607) example: Waiting for system time to be set... (10/25)
I (13607) example: Waiting for system time to be set... (11/25)
I (14607) example: Waiting for system time to be set... (12/25)
I (15607) example: Waiting for system time to be set... (13/25)
I (16607) example: Waiting for system time to be set... (14/25)
I (17607) example: Waiting for system time to be set... (15/25)
I (18607) example: Waiting for system time to be set... (16/25)
I (19607) example: Waiting for system time to be set... (17/25)
I (20607) example: Waiting for system time to be set... (18/25)
I (21607) example: Waiting for system time to be set... (19/25)
I (22607) example: Waiting for system time to be set... (20/25)
I (23607) example: Waiting for system time to be set... (21/25)
I (24607) example: Waiting for system time to be set... (22/25)
I (25607) example: Waiting for system time to be set... (23/25)
I (26607) example: Waiting for system time to be set... (24/25)
```

- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.1-dev-1961-g73384c1-dirty
- Build System: Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:  Linux
- Power Supply:         USB

### Expected Behavior
There was no such problem in release 3.3. In release 4.1 (and release 4.0 as well), I expected sntp to sync the time in no more than 4 seconds.
### Code to reproduce this issue
This is an example code of esp-idf. This is path:
esp-idf/examples/protocols/sntp

Alvin1Zhang, 2020-03-06:
@Eugene-Ilyushchyts Thanks for reporting and sorry for slow reply.

KonstantinKondrashov, 2020-03-10:
Hi @Eugene-Ilyushchyts !
Sorry for the slow reply.
Could you attach your sdkconfig? Do you have some special settings in it?
I suggest applying this patch to enable SNTP_debug [enable_sntp_debug.zip](https://github.com/espressif/esp-idf/files/4310987/enable_sntp_debug.zip). I think it will allow showing additional information which can help to understand the root cause of this behavior. 

An additional suggestion is to try to connect to another WiFi, for example, Wi-Fi hotspot your mobile phone. I use the v4.1-dev-1961-g73384c1 commit as you and do not observe that behavior.
Thanks.

Eugene-Ilyushchyts, 2020-03-11:
Hi @KonstantinKondrashov 
Please let us some time to retest everything. I updated branches yesterday and can't reproduce the issue. I need a couple of days to make sure that the issue doesn't reproduce anymore.

Eugene-Ilyushchyts, 2020-03-17:
Hi @KonstantinKondrashov 
It looks like the problem is no longer there. But I have a couple of times observed a long synchronization via laptop's wifi hotspot. Probably in this case there were problems with laptop's hotspot. I think the issue can be closed. 
Could you clarify? As I understand it, the synchronization request is sent via UDP. What happens if the package is lost ? The next sync package will be sent in at least 15 seconds ? Please correct me if I wrote something wrong.
Thank you in advance.

KonstantinKondrashov, 2020-03-18:
Hi @Eugene-Ilyushchyts!
The synchronization request is sent via UDP. If the package is lost then the next package will be sent in 15 sec. You are correct.
FYI, some links:
RFC-4330: https://tools.ietf.org/html/rfc4330#section-10
SNTP of LWIP: https://www.nongnu.org/lwip/2_1_x/group__sntp__opts.html

Alvin1Zhang, 2020-04-21:
@KonstantinKondrashov Thanks for support. @Eugene-Ilyushchyts Thanks for reporting and updates, seems your issue has been resolved. Feel free to reopen if the issue still happens. Thanks.

closed 2020-04-21 by Alvin1Zhang