espressif/esp-idf #4719: 'btController' task stuck in loop with WiFi/BLE coexistence + 'Bluetooth modem sleep'

hb-kaertech, 2020-02-05:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v4
- Module or chip used: ESP32-WROOM-32D
- IDF version: v4.1-dev-2089-g50466a5e4
- Build System: idf.py
- Compiler version: (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

When using BLE+Wifi coexistence, the 'btController' task gets stuck and takes all CPU time.

More specifically, the 'btController' task gets stuck and read indefinitely from `0x3FF7101C` in function `r_ea_time_get_halfslot_rounded`:
```
400158b0:	0020c0        	memw
400158b3:	0828      	l32i.n	a2, a8, 0
400158b5:	ff7296        	bltz	a2, 400158b0 <0x400158b0>
```

Note: disabling 'Bluetooth modem sleep' fixes the issue.
Note: possibly related: https://github.com/espressif/esp-idf/issues/4691 ?

### Expected Behavior

BLE and WiFi work correctly.

### Actual Behavior

'btController' is stuck in a loop, resulting in BLE to not work, and WiFi performance to be vastly reduced.

### Steps to reproduce

1- Add STA WiFi credentials to your NVS partition
2- Flash&Run my code below

### Code to reproduce this issue

`main.c`: https://gist.github.com/hb-kaertech/696e6cc4931cb55c189d6b073db84256
`sdkconfig`: https://gist.github.com/hb-kaertech/288f673bb32f0c46371b537d9ebe510d

## Debug Logs

I have added debug logs in 'freertos/tasks.c' in 'xTaskIncrementTick' to log the PC of the 'btController' task on each tick. This is the `[TICK CORE1: ...` lines below.

```
(549) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (557) app: App starting
I (597) app: Starting wifi_task
I (597) app: Starting ble_task
I (597) BTDM_INIT: BT controller compile version [5fda27f]
I (597) system_api: Base MAC address is not set
I (597) system_api: read default base MAC address from EFUSE
I (607) wifi: wifi driver task: 3ffcdf28, prio:23, stack:3584, core=0
I (627) wifi: wifi firmware version: 94efb94
I (627) wifi: config NVS flash: enabled
I (627) wifi: config nano formating: disabled
I (627) wifi: Init dynamic tx buffer num: 32
I (627) wifi: Init data frame dynamic rx buffer num: 32
I (627) wifi: Init management frame dynamic rx buffer num: 32
I (627) wifi: Init management short buffer num: 32
I (627) wifi: Init static tx buffer num: 16
I (627) wifi: Init static rx buffer size: 1600
I (627) wifi: Init static rx buffer num: 10
I (627) wifi: Init dynamic rx buffer num: 32
I (727) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (737) wifi: mode : sta (24:6f:28:5b:ce:38)
I (737) app: try to connect to the AP
[TICK CORE1: 'btController' pc:0x400160d6 a0:0x80016369]
[TICK CORE1: 'btController' pc:0x400160d8 a0:0x80016369]
[TICK CORE1: 'btController' pc:0x400160f1 a0:0x8001630d]
[TICK CORE1: 'btController' pc:0x40015c35 a0:0x800160b0]
[TICK CORE1: 'btController' pc:0x40016018 a0:0x800161bc]
[TICK CORE1: 'btController' pc:0x40016101 a0:0x80016388]
[TICK CORE1: 'btController' pc:0x40016050 a0:0x800161fa]
[TICK CORE1: 'btController' pc:0x400160fa a0:0x80016570]
[TICK CORE1: 'btController' pc:0x400160dc a0:0x800162b0]
[TICK CORE1: 'btController' pc:0x40015c2f a0:0x80015e4d]
I (857) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (857) wifi: state: init -> auth (b0)
[TICK CORE1: 'btController' pc:0x400161b9 a0:0x800166a7]
[TICK CORE1: 'btController' pc:0x40015c58 a0:0x80015e4d]
[TICK CORE1: 'btController' pc:0x40015c58 a0:0x80015e4d]
[TICK CORE1: 'btController' pc:0x4001600c a0:0x800162f5]
[TICK CORE1: 'btController' pc:0x40015c3b a0:0x80015e4d]
[TICK CORE1: 'btController' pc:0x40015dd7 a0:0x8001606d]
[TICK CORE1: 'btController' pc:0x4000c452 a0:0x80016018]
[TICK CORE1: 'btController' pc:0x40015c4f a0:0x800160b0]
[TICK CORE1: 'btController' pc:0x40015c64 a0:0x80015e4d]
[TICK CORE1: 'btController' pc:0x400160e1 a0:0x800162c9]
[TICK CORE1: 'btController' pc:0x40015c47 a0:0x800160b0]
[TICK CORE1: 'btController' pc:0x400158b5 a0:0x800d79c6]
I (977) wifi: state: auth -> assoc (0)
I (977) wifi: state: assoc -> run (10)
[TICK CORE1: 'btController' pc:0x400158b0 a0:0x800d79c6]
[TICK CORE1: 'btController' pc:0x400158b5 a0:0x800d79c6]
[TICK CORE1: 'btController' pc:0x400158b0 a0:0x800d79c6]
[TICK CORE1: 'btController' pc:0x400158b0 a0:0x800d79c6]
[TICK CORE1: 'btController' pc:0x400158b0 a0:0x800d79c6]
[TICK CORE1: 'btController' pc:0x400158b3 a0:0x800d79c6]
[TICK CORE1: 'btController' pc:0x400158b3 a0:0x800d79c6]
[TICK CORE1: 'btController' pc:0x400158b3 a0:0x800d79c6]
[TICK CORE1: 'btController' pc:0x400158b3 a0:0x800d79c6]
[TICK CORE1: 'btController' pc:0x400158b3 a0:0x800d79c6]
[TICK CORE1: 'btController' pc:0x400158b0 a0:0x800d79c6]
[TICK CORE1: 'btController' pc:0x400158b0 a0:0x800d79c6]
[TICK CORE1: 'btController' pc:0x400158b0 a0:0x800d79c6]
[TICK CORE1: 'btController' pc:0x400158b0 a0:0x800d79c6]
[TICK CORE1: 'btController' pc:0x400158b0 a0:0x800d79c6]
[TICK CORE1: 'btController' pc:0x400158b0 a0:0x800d79c6]
[TICK CORE1: 'btController' pc:0x400158b0 a0:0x800d79c6]
[TICK CORE1: 'btController' pc:0x400158b0 a0:0x800d79c6]
[TICK CORE1: 'btController' pc:0x400158b0 a0:0x800d79c6]
[TICK CORE1: 'btController' pc:0x400158b0 a0:0x800d79c6]
...
E (5757) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (5757) task_wdt:  - IDLE1 (CPU 1)
E (5757) task_wdt: Tasks currently running:
E (5757) task_wdt: CPU 0: IDLE0
E (5757) task_wdt: CPU 1: btController
... more ticks and more watchdog issues
```

hb-kaertech, 2020-03-17:
This was fixed by 66a01ef490edb33ca21dc182e02b187d9990a3ff / https://github.com/espressif/esp32-bt-lib/commit/659f0a055ac7f8c0abb310660e607245793dc56b .
Unfortunately, a typo in a06f025e61577f123a01d476c1d73faeb272b518 (#7419 instead of #4719) did not tag this task.

Thank you!

closed 2020-03-17 by hb-kaertech

Alvin1Zhang, 2020-03-18:
@hb-kaertech Thanks for the notes.