espressif/esp-idf #4721: Can it do WPS_TYPE_PIN but from the esp side without the need to enter anything on the router ?

minanagehsalalma, 2020-02-05:
Just like a normal wps pin connect app

negativekelvin, 2020-02-05:
https://github.com/espressif/esp-idf/blob/aa087667dffb8f024a5bf162cef4d5c0ffb152ab/components/wpa_supplicant/src/esp_supplicant/esp_wps.c#L2042

minanagehsalalma, 2020-02-05:
> https://github.com/espressif/esp-idf/blob/aa087667dffb8f024a5bf162cef4d5c0ffb152ab/components/wpa_supplicant/src/esp_supplicant/esp_wps.c#L2042

@negativekelvin can we expect it to get supported in any time ?

negativekelvin, 2020-02-06:
I'm not sure what the advantage in entering a pin instead of just entering the wifi password

minanagehsalalma, 2020-02-06:
> I'm not sure what the advantage in entering a pin instead of just entering the wifi password

It just faster... and in case of connecting to a more than one router  wps pin would be easier. ..

Here is what i am thinking of 
You connect to the esp ap and open a page that you select the network from the scanned list and then enter the wps pin in and click connect .

What do you think ?

nachiketkukade, 2020-04-09:
As par WPS Specs, entering a PIN is to be done on a Registrar, which we don't support.(already mentioned).

Furthermore, the proposal that the PIN to be entered by connecting to the ESP-SoftAP-Registrar(with what security?) for configuring the ESP-STA-Enrollee doesn't fit any WPS use cases. The WPS use cases are below -

1. Configuring a Standalone AP (with/without external Registrar)
2. Adding Enrollee device to network with help of Registrar (built-in in AP or External).

So in essence this will be a complicated task for the simple purpose of configuring a device.
As @negativekelvin pointed out, entering a Wi-Fi password or other provisioning methods are simpler and efficient for this purpose.

minanagehsalalma, 2020-04-09:
> physical access to the Router is a must.

Nah not for wps pin.

I don't know what to say but wps pin is a basic part of wps functionality ...

> a complicated task

Idk why you think adding a Registrar would be complex ?

nachiketkukade, 2020-04-09:
Ok let's go over this one more time.

The first use case where we're configuring the out-of-box AP, we can read a PIN from Router's sticker and enter it for setting up the router. Here the AP is the enrollee while STA is registrar.

The 2nd use case where we're adding a new device to the network, we will read a PIN from the Device and enter it on the Router Webpage or on its keypad (if it has one). Here the device is the enrollee and the Router has a built-in registrar. The Webpage needs to have User Authentication over TLS, so it is assumed we already have a secured connection with the Router. This means we already had physical access to Router during the first use case.

Now what we're talking is the reverse. ESP as Enrollee, but read PIN from Router and enter it on ESP. This doesn't fit into a WPS use case, we can't have Enrollee + built-in Registrar, since accepting PIN is done by a Registrar.

Do you agree? If I've missed anything let me know.

minanagehsalalma, 2020-04-09:
Hmm...the 2nd can be done using the first one too as you can have the router pin already saved in your device by storing it after reading it from the sticker... also you can find it in the router page which means you don't need a physical access.... and in case of the sticker is damaged and you don't have an already connected device or don't want to get up and go to the router you can use a wps pin generator which is available online and for all platforms..

In all the case that i mentioned the only one that needs physical access is the first case only.. and the pin is just 8 digits the most of people can simply memorize.

As you can see there is mostly no need for physical access. .. and If not yet convinced.

You can take a look on these 
https://play.google.com/store/apps/details?id=com.tester.wpswpatester

https://play.google.com/store/apps/details?id=com.xti.wifiwarden
And see the use cases of them as the use cases of this suggestion.

nachiketkukade, 2023-03-10:
To give an update, WPS registrar is supported now, please see commit 924c709efe71b2dfeea1991657f26d035f9e7b17. 

As for the above comment, these Apps and other WPS PIN generators available on the Internet are bruteforcing or guessing the sticker PIN of the AP and forcing the AP to give its default credentials. The Smartphone in this case exploits one of the registrar use cases and can use the credentials to connect itself. How it can be implemented using ESP-IDF is up to the developers.

closed 2023-03-10 by nachiketkukade

minanagehsalalma, 2023-03-13:
> To give an update, WPS registrar is supported now, please see commit [924c709](https://github.com/espressif/esp-idf/commit/924c709efe71b2dfeea1991657f26d035f9e7b17).

@nachiketkukade  thanks alot, keep the good work on 🔥🔥

so regrading this part and the comment below it 

  ```
  if (os_strncmp((char *)cfg->pin, "00000000", 8) != 0) {

    if ((os_strncmp((char *)cfg->pin, "00000000", 8) == 0) || !wps_pin_str_valid((char *)cfg->pin)) {
        unsigned int spin = 0;

        cfg->dev_pw_id = DEV_PW_DEFAULT;
        cfg->pin_len = 8;
        if (wps_generate_pin(&spin) < 0) {
            return -1;
	}
```

there is a well known exploit in wps pin called [null pin](https://github.com/t6x/reaver-wps-fork-t6x/wiki/Introducing-a-new-way-to-crack-WPS:-Option--p-with-an-Arbitrary-String#:~:text=PIN%20value%20is%20%22-,NULL,-%22%20(none%2C%20an%20%22empty)) which is just an empty pin ... so to get it to work we need to modify that part, right ?

nachiketkukade, 2023-03-17:
Yes, you can try modifying this part. If there are no failures ahead it should work