espressif/esp-idf #4724: A fatal error occurred: MD5 of file does not match data in flash!

raviibot, 2020-02-06:
hi,
 I have custom board with ESP32D2WD  module while flashing the code with UART cable in to the esp3d2wd module using eclipse IDE using esp-idf-version 3.3 I am getting the following error mostly.very rarely it is flashing I have checked the power supply, tried with different uart cables and  different baud rates still the problem is present. 
But when I am flashing the same hardware with  ardunio IDE I am not getting the error it is flashing well.

C:\Users\ravi.ibot029\Desktop\esp_task\spi_master>idf.py -p COM47 flash
Note: You are using Python 3.7.3. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Checking Python dependencies...
Python requirements from C:\esp-idf_3.3\requirements.txt are satisfied.
Running ninja in directory C:\Users\ravi.ibot029\Desktop\esp_task\spi_master\build
Executing "ninja all"...
[1/3] Performing build step for 'bootloader'
ninja: no work to do.
Running esptool.py in directory C:\Users\ravi.ibot029\Desktop\esp_task\spi_master\build
Executing "C:\Users\ravi.ibot029\.espressif\python_env\idf3.3_py3.7_env\Scripts\python.exe C:\esp-idf_3.3\components/esptool_py/esptool/esptool.py -p COM47 -b 460800 --after hard_reset write_flash @flash_project_args"...
esptool.py -p COM47 -b 460800 --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin 0x10000 spi_master.bin
esptool.py v2.8
Serial port COM47
Connecting.....
Detecting chip type... ESP32
Chip is ESP32D2WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, Embedded Flash, Coding Scheme None
Crystal is 40MHz
MAC: 24:0a:c4:19:65:74
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 25600 bytes to 15252...
Wrote 25600 bytes (15252 compressed) at 0x00001000 in 0.4 seconds (effective 569.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 82...
Wrote 3072 bytes (82 compressed) at 0x00008000 in 0.0 seconds (effective 1525.4 kbit/s)...
Hash of data verified.
Compressed 177328 bytes to 81434...
Wrote 177328 bytes (81434 compressed) at 0x00010000 in 2.0 seconds (effective 726.3 kbit/s)...
File  md5: 83f1fc8c755f40a086b04eb3aaa14c00
Flash md5: 9d8dbeb0977a624f1121231435182dab
MD5 of 0xFF is e9fb180d264d3e2b83dd69e11be4e5e4

A fatal error occurred: MD5 of file does not match data in flash!
esptool.py failed with exit code 2

igrr, 2020-04-08:
Hi @raviibot, do you have anything connected to GPIO12? Can you please check the voltage on GPIO12 pin when ESP32 chip is being reset? As per the datasheet, this pin should be pulled high on reset, for the ESP32-D2WD chip.

Alvin1Zhang, 2020-05-26:
Will close due to short of feedback/updates, feel free to reopen with more details. Thanks. @raviibot

closed 2020-05-26 by Alvin1Zhang