espressif/esp-idf #4725: UDP Multicast example "create_multicast_ipv6_socket" function error.

ThomasGeorgiadis, 2020-02-06:
INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ +] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ +] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ +] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Development Kit:      [ESP32-Wrover-Kit|
- Kit version (for WroverKit): [v3]
- Module or chip used:  [ESP32-WROOM-32|
- IDF version : v4.1-dev-1961-g73384c1b5
- Build System: [Make|
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     |Linux|

- Using an IDE?: [No|
- Power Supply:         [USB|

## Problem Description

I am using the UDP Multicast example to create an ipv6 socket. When i compile the code i get the following error . Is there any solution or is it a bug from the idf?
## Code

[image]

## Debug Logs

[image]

freakyxue, 2020-03-23:
hi @ThomasGeorgiadis

LwIP for IDF 4.0 has been updated to 2.1.2.

typedef struct ipv6_mreq {
  struct in6_addr ipv6mr_multiaddr; /*  IPv6 multicast addr */
  unsigned int    ipv6mr_interface; /*  interface index, or 0 */
} ipv6_mreq;
 ipv6mr_interface member of IPv6 Multicast is an int variable, not an address

ThomasGeor, 2020-03-23:
Thanks for your help @freakyxue . I had resolved it quite a long time ago and forgot to update the issue. It's exactly as you say,although i couldn't find it in the documentation. I needed to see the code thoroughly.

freakyxue, 2020-03-23:
hi @ThomasGeorgiadis
 can  I close this issue?

ThomasGeor, 2020-03-23:
Sure.

Alvin1Zhang, 2020-03-23:
@freakyxue Thanks for support. @ThomasGeorgiadis Thanks for reporting, will close now. Thanks.

closed 2020-03-23 by Alvin1Zhang