espressif/esp-idf #4726: ESP SPI Master with two devices on one bus does not work

Wielebny666, 2020-02-06:
INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [+] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [+] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [+] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 3.3.1
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 5.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes] VSCode
- Power Supply:         [USB]

## Problem Description

I connected two devices to the SPI port. AS3933 and CC1101. If I initiate AS3933, communication works correctly. 
After adding the second CC1101 device, there is no communication with AS3933. 
After physical disconnection, CC1101 still has no communication.

### Code to reproduce this issue
```
void app_main()
{
	uint8_t test;
	as3933_create();
	as3933_read_byte(5, &test)
    ESP_LOGD(TAG, "RFID Init %s", test == 0x96 ? "OK" : "Fail");
	cc1101_create();
	as3933_read_byte(5, &test) //return 0!!
    ESP_LOGD(TAG, "RFID Init %s", test == 0x96 ? "OK" : "Fail");
}
```

```
static spi_device_handle_t as3933;
esp_err_t as3933_create()
{
    esp_err_t error = ESP_OK;

    spi_bus_config_t buscfg = {
        .miso_io_num = GPIO_NUM_19,
        .mosi_io_num = GPIO_NUM_23,
        .sclk_io_num = GPIO_NUM_18,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1};

    spi_device_interface_config_t devcfg = {
        .clock_speed_hz = SPI_MASTER_FREQ_8M,
        .mode = 1,                                    //SPI mode 1
        .spics_io_num = GPIO_NUM_5,                   //CS pin
        .flags = SPI_DEVICE_POSITIVE_CS,
        .queue_size = 1, //We want to be able to queue 7 transactions at a time
        .command_bits = 8};

    //Initialize the SPI bus
    error = spi_bus_initialize(VSPI_HOST, &buscfg, 0);
    error = spi_bus_add_device(VSPI_HOST, &devcfg, &as3933);
    return error;
}

esp_err_t as3933_read_byte(uint8_t addr, uint8_t *byte) {
    esp_err_t error = ESP_OK;
    spi_transaction_t t;
    memset(&t, 0, sizeof(t));
    t.length = 8;
    t.cmd = addr | 0x40;
    t.rx_buffer = byte;

    error = spi_device_polling_transmit(as3933, &t);
    return error;
}
```

```
static spi_device_handle_t cc1101_handle;
esp_err_t cc1101_create() {
      esp_err_t error = ESP_OK;
      spi_device_interface_config_t devcfg = {
          .clock_speed_hz = SPI_MASTER_FREQ_8M,
          .mode = 3,                                    //SPI mode 1
          .spics_io_num = GPIO_NUM_15,                   //CS pin
          .queue_size = 1,                              //We want to be able to queue 7 transactions at a time
          .command_bits = 8};

      error = spi_bus_add_device(VSPI_HOST, &devcfg, &cc1101_handle);
      return error;
}
```

## Debug Logs

D (348) AS3933_HAL: as3933_create
D (418) spi: SPI3 use iomux pins.
D (428) spi_master: eff: 2000, limit: 80000k(/0), 0 dummy, -1 delay
D (438) spi_master: SPI3: New device added to CS0, effective clock: 2000kHz
D (438) AS3933_HAL: as3933_cmd?[0m
D (448) spi_master: SPI device changed from 3 to 0?[0m
D (458) AS3933_HAL: as3933_read_byte
D (468) app_main: RFID Init OK
D (2468) CC1101_HAL: cc1101_create
D (2468) spi_master: eff: 2000, limit: 80000k(/0), 0 dummy, -1 delay
D (2468) spi_master: SPI3: New device added to CS1, effective clock: 2000kHz
D (2478) AS3933_HAL: as3933_read_byte
D (2488) app_main: RFID Init Fail

alemariusnexus, 2020-02-07:
Could this maybe be related to my issue #4641? I still haven't heard anything back from that one. Although with your code, both devices seem to be full-duplex.

Wielebny666, 2020-02-10:
I changed the order of adding the device and it works correctly. 
First, I add a device with a low-cs level.

Wielebny666, 2020-05-24:
Up

Wielebny666, 2020-06-16:
Solution in this threat https://github.com/espressif/esp-idf/issues/4641

Wielebny666, 2020-07-09:
In master is this same problem
https://github.com/espressif/esp-idf/blob/625bd5eb1806809ff3cc010ee20d1f750aa778a1/components/soc/src/esp32/include/hal/spi_ll.h#L294

Icarus113, 2020-09-28:
Your issue is resolved by commit 27a6f266, check this issue when it's sync to master. Thanks~

KaeLL, 2020-09-28:
Interesting. I'm having an issue which I can't tell if it's related, but it seems to be in some degree through the use of polling transactions. I get a LoadProhibited crash right when I'm adding a second device to VSPI bus, and the addr2line output points exactly to [this](https://github.com/espressif/esp-idf/blob/5ef1b390026270503634ac3ec9f1ec2e364e23b2/components/driver/spi_master.c#L601) line with discriminator 2. Similar to your case, it does seems to be something related to when the all but the first device are added to the spi bus, as I'm also adding the 2nd device after I started using the first one.
@Icarus113 do you have any idea about this being related to OP's issue?