espressif/esp-idf #4730: [C++] "Warning: pointer of type 'void *' used in arithmetic" in mesh_util.h

mcilloni, 2020-02-07:
## Environment

- Development Kit: ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v4.0-rc-12-ga61fd1e42 
- Build System: all
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description
g++ raises a slew of `"warning: pointer of type 'void *' used in arithmetic"` warnings whenever `components/bt/esp_ble_mesh/mesh_common/include/mesh_util.h` is included (directly or not) by C++ code. 

```
In file included from /home/marco/esp-homekit-sdk/esp-idf/components/bt/esp_ble_mesh/mesh_common/include/mesh_util.h:19,
                 from /home/marco/esp-homekit-sdk/esp-idf/components/bt/esp_ble_mesh/mesh_core/include/mesh_access.h:16,
                 from /home/marco/esp-homekit-sdk/esp-idf/components/bt/esp_ble_mesh/api/esp_ble_mesh_defs.h:21,
                 from ../src/gweventhandlers/blemesh_event_handler.cpp:13:
/home/marco/esp-homekit-sdk/esp-idf/components/bt/esp_ble_mesh/mesh_common/include/mesh_util.h: In function 'void sys_memcpy_swap(void*, const void*, size_t)':
/home/marco/esp-homekit-sdk/esp-idf/components/bt/esp_ble_mesh/mesh_common/include/mesh_util.h:403:34: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     __ASSERT(((src < dst && (src + length) <= dst) ||
                              ~~~~^~~~~~~~
/home/marco/esp-homekit-sdk/esp-idf/components/bt/esp_ble_mesh/mesh_common/include/mesh_trace.h:82:15: note: in definition of macro '__ASSERT'
         if (!(test)) {                                  \
               ^~~~
/home/marco/esp-homekit-sdk/esp-idf/components/bt/esp_ble_mesh/mesh_common/include/mesh_util.h:404:34: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
               (src > dst && (dst + length) <= src)),
                              ~~~~^~~~~~~~
/home/marco/esp-homekit-sdk/esp-idf/components/bt/esp_ble_mesh/mesh_common/include/mesh_trace.h:82:15: note: in definition of macro '__ASSERT'
         if (!(test)) {                                  \
               ^~~~
In file included from /home/marco/esp-homekit-sdk/esp-idf/components/bt/esp_ble_mesh/mesh_core/include/mesh_access.h:16,
                 from /home/marco/esp-homekit-sdk/esp-idf/components/bt/esp_ble_mesh/api/esp_ble_mesh_defs.h:21,
                 from ../src/gweventhandlers/blemesh_event_handler.cpp:13:
/home/marco/esp-homekit-sdk/esp-idf/components/bt/esp_ble_mesh/mesh_common/include/mesh_util.h:407:9: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     src += length - 1;
     ~~~~^~~~~~~~~~~~~
cc1plus: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
/home/marco/esp-homekit-sdk/esp-idf/components/bt/esp_ble_mesh/mesh_common/include/mesh_util.h:410:22: warning: ISO C++ forbids incrementing a pointer of type 'void*' [-Wpointer-arith]
         *((u8_t *)dst++) = *((u8_t *)src--);
                      ^~
/home/marco/esp-homekit-sdk/esp-idf/components/bt/esp_ble_mesh/mesh_common/include/mesh_util.h:410:41: warning: ISO C++ forbids decrementing a pointer of type 'const void*' [-Wpointer-arith]
         *((u8_t *)dst++) = *((u8_t *)src--);
```

This is caused by the improper use of arithmetic operations on `void*` pointers in [sys_memcpy_swap](https://github.com/espressif/esp-idf/blob/master/components/bt/esp_ble_mesh/mesh_common/include/mesh_util.h#L401):

```
static inline void sys_memcpy_swap(void *dst, const void *src, size_t length)
{
    __ASSERT(((src < dst && (src + length) <= dst) ||
              (src > dst && (dst + length) <= src)),
             "Source and destination buffers must not overlap");

    src += length - 1;

    for (; length > 0; length--) {
        *((u8_t *)dst++) = *((u8_t *)src--);
    }
}
```

The error is that pointer arithmetic operations may only involve pointers to object types, but void is by definition an incomplete type. This only works because GNU C goes against the standard and defines (for convenience's sake) `sizeof(void) == 1` in these cases, as if `void*` was actually `char*`; this is not tolerated, however,  by standard C++ (i.e. when passing `-std=c++17` to g++). 

### Expected Behavior
No warning is printed when including headers including mesh_util.h while using standard C++.

### Actual Behavior
Misleading warnings are printed when including headers including mesh_util.h while using standard C++.

### Steps to reproduce

Include mesh_util.h in C++ code and compile with -std=c++17

### Code to reproduce this issue

```cpp
#include <blemesh/blemesh_data_api.h>

extern "C"
void app_main() {}

```

Campou, 2020-02-12:
Hi @mcilloni 

Thanks for reporting this, we will fix it.

Regards

Alvin1Zhang, 2020-05-26:
@mcilloni Thanks for reporting, the fix detail is here https://github.com/espressif/esp-idf/commit/326ccfca88dd339fea84e87649ec73706540db9d, feel free to reopen if the issue still happens. Thanks.

closed 2020-05-26 by Alvin1Zhang