espressif/esp-idf #4731: Add doc: Installation bash script

cristovao-trevisan, 2020-02-07:
Hi, to make it easier to my coworkers, I've created an installation bash script. If you are interested I can create a PR to add this to the documentation.

This is the prototype (probably need to add some env var for the version and some detection for the profile file, as I'm using zsh):
```sh
echo "\n\nCloning IDF v3.3.1\n\n"
mkdir -p ~/esp
cd ~/esp
git clone --branch v3.3.1 --recursive git@github.com:espressif/esp-idf.git

echo "Installing components\n\n"
cd esp-idf
./install.sh

echo "Setting up environment\n\n"
IDF_PATH=$HOME/esp/esp-idf
echo "export IDF_PATH=$IDF_PATH" >> ~/.zprofile
echo ". $IDF_PATH/add_path.sh > /dev/null" >> ~/.zprofile

echo "Executing export.sh"
. ./export.sh

echo "\n\nDONE\n"
```

igrr, 2020-02-08:
Hi @cristovao-trevisan, thanks for raising this request.
There a few things that may need to be configurable for the script to be widely useful:
- Version/branch of ESP-IDF to download
- Whether to clone it Github or download the zip archive of the git repository (for folks who have unreliable connection to Github)
- Whether to use an HTTPS or SSH URL if cloning from Github
- Where to put the ESP-IDF directory
- Whether to set custom IDF_TOOLS_PATH or not
- Whether to add setting of IDF_PATH and export.sh to some login script. If yes, to which one (bashrc, profile, zprofile, etc). Or maybe another option, add a function or an alias to the profile file, which will execute the export script. One think I am quite certain is that the IDF environment should *not* be activated by default in the profile script, unless the account is used exclusively for IDF development. This is because the export.sh script activates a Python virtual environment, dedicated to ESP-IDF packages. This will likely cause problems for other use cases involving Python. For instance, `python -m venv` will no longer work since a virtual environment will already be active.

All in all, i think such script could be useful when deployed in a known environment. But it might not be advisable for users to blindly assume that the script will do the right thing for their system. Possibly an interactive script could solve this problem by asking a series of questions and then adjusting the installation accordingly? It has been my perception that Linux users are often allergic to such interactive scripts and "installers", although admit I might be wrong and this isn't wide spread.

Finally, although not directly related, there is another way to create a very reproducible environment for ESP-IDF: the `espressif/idf` Docker image (described here: https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-docker-image.html). There are some caveats regarding interactive usage, but these apply to Linux and macOS only. This image potentially could be used for day-to-day development on Linux.

cristovao-trevisan, 2020-02-12:
Hi @igrr , and thank you for this very detailed response.

I've seen the new release came a couple days ago, and with it the vscode extension. As the extension already does the setup with all the options on a really nice interface, I will recommend my coworkers to use it instead.

Since I find my solution, I'll be closing this issue, but if there is more interest I can work on the script.
This library is being very useful for me and I plan to solve some issues instead of creating them in the near future :)

closed 2020-02-12 by cristovao-trevisan