espressif/esp-idf #4732: Specify API stability level

KaeLL, 2020-02-07:
In the documentation, specify whether APIs are stable, and if so, to which extent.
I know there are some functions kept for backwards compatibility that end up getting removed, but is there a repository wide policy on this? What about the order of `enum` constants, and the order of `struct` parameters, etc? Are they stable or not, and how much?

igrr, 2020-02-08:
Hi @KaeLL, good point that this should be documented.
Until this information is added to the docs, here are the answers to your questions:

1. Within one major version of IDF (e.g. 3.x), the policy is to not remove functions, type definitions, enum constants, defines, header files, and to not change function signatures. There are some exceptions to this rule:
   - A definition may be removed when a function/type/constant could never have been used successfully. For example, when related functionality was never implemented or was broken.
   - Functions and types defined in `examples` subdirectory are not considered part of the API, and may be changed at any time. This also applies to the components included with the examples (such as esp_modem, led_strip and others).

   At the same time, we may deprecate functions and types (using a `deprecated` attribute) and header files (using a preprocessor warning). The functions/types/header files should still work after being deprecated. Deprecated definitions may be removed in the next major version.

2. We don't have a rule about keeping the order of enum constants and of structure members. We don't make changes to the ordering in patch versions (e.g. 3.3.x), but we don't guarantee that the order will be the same between minor versions. So if you are, for instance, distributing a pre-compiled library for ESP-IDF, this library may need to be recompiled for different minor versions.

We'll add this information to https://docs.espressif.com/projects/esp-idf/en/latest/versions.html#versioning-scheme. Please let me know if there is anything else you think should be mentioned.

AxelLin, 2021-03-17:
> 2. We don't have a rule about keeping the order of enum constants and of structure members. We don't make changes to the ordering in patch versions (e.g. 3.3.x), but we don't guarantee that the [...]

This could be a potential bug if you don't guarantee that the order of enum between versions.
The user application may save the enum value as user setting in NVS.
And it even can cause breakage within esp-idf components. e.g. a9b0336ac5f5c20876da868ffa7adc0d38971ef5

KaeLL, 2021-03-17:
Well, that's new.
To be honest, the argument for stable `struct`s declarators' order is just because I like my short initializer lists.  
In the end, this is just another example of [Hyrum's Law](https://www.hyrumslaw.com/).