espressif/esp-idf #4733: OTA Causes Core Meditation Error and HTTP_CLIENT: esp_transport_read returned:-1 and errno:9

jonhunterking66, 2020-02-07:
## Environment

- Development Kit:      [ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):
    v4.1-dev-2126-gf91080637
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Using an IDE?:  VS Code 
- Power Supply:      USB

## Problem Description

Attempting to OTA through Google drive causes core meditation errors. I am doing this based off of a proven example so I am confident this is not a code issue.  

### Expected Behavior

OTA would complete successfully 

### Actual Behavior

Core meditation errors and W (10175) HTTP_CLIENT: esp_transport_read returned:-1 and errno:9 

### Steps to reproduce

1. Use master branch of esp-idf
2. Change menuconfig settings for SSID and Password to your network, select 2 ota partitiona and 4MB flash. 
3. Run example code shared below

### Code to reproduce this issue

I am using an OTA example line for line. It can be found here: https://github.com/Mair/esp32-course/tree/master/_18_OTA/_18_1_OTA_using_google_drive

## Debug Logs

```
W (10175) HTTP_CLIENT: esp_transport_read returned:-1 and errno:9 
W (10175) HTTP_CLIENT: esp_transport_read returned:-1 and errno:9 
W (10175) HTTP_CLIENT: esp_transport_read returned:-1 and errno:9 
W (10185) HTTP_CLIENT: esp_transport_read returned:-1 and errno:9 
W (10195) HTTP_CLIENT: esp_transport_read returned:-1 and errno:9 
W (10205) HTTP_CLIENT: esp_transport_read returned:-1 and errno:9 
W (10205) HTTP_CLIENT: esp_transport_read returned:-1 and errno:9 
W (10215) HTTP_CLIENT: esp_transport_read returned:-1 and errno:9 
W (10225) HTTP_CLIENT: esp_transport_read returned:-1 and errno:9 
W (10225) HTTP_CLIENT: esp_transport_read returned:-1 and errno:9 
W (10235) HTTP_CLIENT: esp_transport_read returned:-1 and errno:9 
W (10245) HTTP_CLIENT: esp_transport_read returned:-1 and errno:9 
W (10245) HTTP_CLIENT: esp_transport_read returned:-1 and errno:9 
W (10255) HTTP_CLIENT: esp_transport_read returned:-1 and errno:9 
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40149635  PS      : 0x00060330  A0      : 0x80144ce8  A1      : 0x3ffbefd0  
0x40149635: mbedtls_ssl_read at /home/jonhunterking/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:8273

A2      : 0x00080000  A3      : 0x3ffc791c  A4      : 0x00000116  A5      : 0x3ffc7bf0  
A6      : 0x00000001  A7      : 0xff000000  A8      : 0x80144f12  A9      : 0x3ffbefb0  
A10     : 0x00000269  A11     : 0x00000004  A12     : 0x3ffbeee4  A13     : 0x3ffae920  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000800cc  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

ELF file SHA256: 0b93f2ed83bbc0e16ed6b2b7abf1e122e0fa7f5f1f49e6039b40a0beba0f8792

Backtrace: 0x40149632:0x3ffbefd0 0x40144ce5:0x3ffbeff0 0x4013e997:0x3ffbf010 0x40164ee5:0x3ffbf030 0x401398d8:0x3ffbf050 0x400f8745:0x3ffbf090 0x400f87ed:0x3ffbf1e0 0x400f8938:0x3ffbf200 0x400f8c47:0x3ffbf230 0x400d5f26:0x3ffbf260 0x40092915:0x3ffbf2e0
0x40149632: mbedtls_ssl_read at /home/jonhunterking/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:8267 (discriminator 1)

0x40144ce5: esp_mbedtls_read at /home/jonhunterking/esp/esp-idf/components/esp-tls/esp_tls_mbedtls.c:132

0x4013e997: esp_tls_conn_read at /home/jonhunterking/esp/esp-idf/components/esp-tls/esp_tls.h:453
 (inlined by) ssl_read at /home/jonhunterking/esp/esp-idf/components/tcp_transport/transport_ssl.c:157

0x40164ee5: esp_transport_read at /home/jonhunterking/esp/esp-idf/components/tcp_transport/transport.c:182

0x401398d8: esp_http_client_read at /home/jonhunterking/esp/esp-idf/components/esp_http_client/esp_http_client.c:854

0x400f8745: _http_handle_response_code at /home/jonhunterking/esp/esp-idf/components/esp_https_ota/src/esp_https_ota.c:81

0x400f87ed: _http_connect at /home/jonhunterking/esp/esp-idf/components/esp_https_ota/src/esp_https_ota.c:115

0x400f8938: esp_https_ota_begin at /home/jonhunterking/esp/esp-idf/components/esp_https_ota/src/esp_https_ota.c:180

0x400f8c47: esp_https_ota at /home/jonhunterking/esp/esp-idf/components/esp_https_ota/src/esp_https_ota.c:400

0x400d5f26: run_ota at /home/jonhunterking/Github/OtO_FW/build/../main/main.c:46

0x40092915: vPortTaskWrapper at /home/jonhunterking/esp/esp-idf/components/freertos/xtensa/port.c:143
```

jonhunterking66, 2020-02-07:
Update, this is not an issue on version release/v4.0

AxelLin, 2020-02-07:
@jonhunterking66 
Check if the mentioned patch helps.
https://www.esp32.com/viewtopic.php?f=13&t=13772

closed 2020-02-10 by jonhunterking66