espressif/esp-idf #4735: Enable WebAssembly Micro Runtime for ESP

xwang98, 2020-02-08:
**Is your feature request related to a problem? Please describe.**

WebAssembly Micro Runtime (WAMR) is a [bytecode alliance](https://bytecodealliance.org/) project for standalone WebAssembly runtime targeting embedded environment. The WAMR GitHub location is https://github.com/bytecodealliance/wasm-micro-runtime.

There are a few features which make the WAMR a perfect WebAssembly runtime for ESP:
- small footprint
- support both interpreter and ahead of time compilation
- application framework supports multiple WAMS applications and inter-communication
- Event driven programming API set ready for programming applications, and yet support for extension.
- remote application management support

**Describe the solution you'd like**

We will like to check interest from epressif and the ESP community for enabling the WAMR for ESP and establishing collaboration between the two projects.

**Additional context**

igrr, 2020-02-08:
Thanks for the pointer to the WAMR project, @xwang98! It seems that the port for ESP32 is in progress by @nickzoic. I'll follow up in https://github.com/bytecodealliance/wasm-micro-runtime/issues/134 in case some help is needed.

xwang98, 2020-02-09:
@igrr thank you! 

We happen to see the project https://github.com/espressif/llvm-project. It should be helpful if we want to support AoT complation for ESP. I wonder what is the status of this project?

nickzoic, 2020-02-09:
Wow, thanks Ivan, very much appreciated!

xwang98, 2020-04-08:
@igrr we recently greatly improved the WAMR interpreter perormance and did performance and memory comparision with wasm3. You can check it out here:
https://github.com/bytecodealliance/wasm-micro-runtime/wiki/Performance

Another news is that we have enabled AoT support for Xtensa. The test was done on Zephyr/ESP32.  Comparing to the interpreter mode, the AoT performance is amazingly fast. 

@nickzoic we have re-organized the platform porting for WAMR. Now the porting should be much simpler than ever. I wonder if you still have plan to port WAMR for ESP32?

It would be a really nice thing to have WAMR enabled for ESP32 community.

xwang98, 2020-04-08:
Here is the new guideline for how to port WAMR for a platform:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/master/doc/port_wamr.md

1c3t3a, 2021-12-16:
I've implemented the aforementioned API for esp-idf, but struggle with the memprotect implementation. Is it possible that someone looks over the PR (mentioned above)?

igrr, 2021-12-16:
@1c3t3a I don't think there's a way to implement a generic `mprotect` function on an ESP32. Does leaving it unimplemented cause any issue in WAMR? If you are seeing some issue, please post the details either in the PR or here and we'll try to help.

1c3t3a, 2021-12-19:
Alright, thanks for the answer! I also got another problem. I would like to check the stack upper bound address. I found a few methods for acquiring `pxStack` from the TCB, but I wonder if there is a cleaner solution to it?

wallento, 2021-12-31:
With https://github.com/bytecodealliance/wasm-micro-runtime/pull/927 now works for esp32 and esp32c3.
I am struggling in creating a "component" out of wamr. Is there any comparable project you can refer me to?
I was thinking about having my own simple idf+wamr project like this:
```
cmake_minimum_required(VERSION 3.5)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)

set (COMPONENTS ${IDF_TARGET} freertos esptool_py wamr)
list(APPEND EXTRA_COMPONENT_DIRS "$ENV{WAMR_PATH}/esp-idf")

project(wamr-esp32c3)
```
I am unfortunately in a bad trial-and-error loop with cmake..

wallento, 2022-01-01:
Got it! It is in the PR now: https://github.com/bytecodealliance/wasm-micro-runtime/pull/927
Quick steps to test are described here: https://github.com/wallento/wasm-micro-runtime/blob/esp-idf-fixes/build-scripts/esp-idf/README.md
Most easily (You need to have ESP-IDF setup, `IDF_PATH` set and the export file sourced):
```
git clone git@github.com:bytecodealliance/wasm-micro-runtime.git
cd wasm-micro-runtime/
export WAMR_PATH=$PWD
git fetch origin pull/927/head:esp-idf
git checkout esp-idf
cd build-scripts/esp-idf/example/wamr-hello/
idf.py set-target esp32c3
idf.py build
idf.py flash
idf.py monitor
```

xoesq, 2022-03-20:
Hi and thanks to everyone!

@wallento Do you know how to use the app-framework timers for esp-idf and wamr as a component?
Thank you very much.

igrr, 2022-09-29:
I'm going to close this issue, since WAMR can now be used with ESP-IDF ([docs](https://github.com/bytecodealliance/wasm-micro-runtime/blob/78414b627c1d8309134391c1ff863a89f0db4764/doc/build_wamr.md#esp-idf)), as well as an IDF component ([docs](https://github.com/bytecodealliance/wasm-micro-runtime/blob/78414b627c1d8309134391c1ff863a89f0db4764/build-scripts/esp-idf/README.md)).

closed 2022-09-29 by igrr

xwang98, 2022-09-29:
Thanks to everyone!

goodluckrhj, 2022-10-21:
> I'm going to close this issue, since WAMR can now be used with ESP-IDF [...]

Hi, I'm new to both ESP32 and WAMR. I have some questions about

- what is the suggested embedded OS to run "pub/sub, micro serivce, message, app manager"?
- is it possible to OTA when wamr acts as a component on esp32?

goodluckrhj, 2022-10-21:
> I'm going to close this issue, since WAMR can now be used with ESP-IDF [...]

Hi, I'm new to both ESP32 and WAMR. I have some questions about

- what is the suggested embedded OS to run "pub/sub, micro serivce, message, app manager"?
- is it possible to OTA when wamr acts as a component on esp32 with all features above implemented?