espressif/esp-idf #4736: http_request example build error

DasVoelkel, 2020-02-08:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):  "version": "3.30300.190916" according to platformios json of the package
- Build System:        platformio
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:    Linux

- Using an IDE?: Yes  VSCode with platformio
- Power Supply:         USB

## Problem Description
When trying to compile this:

https://github.com/espressif/esp-idf/blob/c38bf3534eab4e598177d11f97f57afeb664df33/examples/protocols/http_request/main/http_request_example_main.c#L97

the selected line causes an error:
```
sorry, unimplemented: non-trivial designated initializers not supported
     };
     ^
lib/mg_wifi/mg_web_tasks.cpp:27:5: sorry, unimplemented: non-trivial designated initializers not supported
lib/mg_wifi/mg_web_tasks.cpp:27:5: warning: missing initializer for member 'addrinfo::ai_socktype' [-Wmissing-field-initializers]
lib/mg_wifi/mg_web_tasks.cpp:27:5: warning: missing initializer for member 'addrinfo::ai_protocol' [-Wmissing-field-initializers]
lib/mg_wifi/mg_web_tasks.cpp:27:5: warning: missing initializer for member 'addrinfo::ai_addrlen' [-Wmissing-field-initializers]
lib/mg_wifi/mg_web_tasks.cpp:27:5: warning: missing initializer for member 'addrinfo::ai_addr' [-Wmissing-field-initializers]
lib/mg_wifi/mg_web_tasks.cpp:27:5: warning: missing initializer for member 'addrinfo::ai_canonname' [-Wmissing-field-initializers]
lib/mg_wifi/mg_web_tasks.cpp:27:5: warning: missing initializer for member 'addrinfo::ai_next' [-Wmissing-field-initializers]
lib/mg_wifi/mg_web_tasks.cpp: At global scope:
lib/mg_wifi/mg_web_tasks.cpp:21:1: warning: 'void mg::module_wifi::http_get_task(void*)' defined but not used [-Wunused-function]
 http_get_task(void *pvParameters)
```

### Expected Behavior

No error in examples.

igrr, 2020-02-08:
Hi @DasVoelkel,
As written, the IDF example you are referring to is valid C code (as indicated by the file extension), but not valid C++ code.
Most of the time taking C code and compiling it with a C++ compiler will work as you'd expect. However there are a few exceptions. Designated initializers in C99 allow arbitrary order of structure fields, while in C++ designated initializers are not supported at all until C++20. GCC implements a limited support for designated initializers in C++ mode as an extension. It requires that all fields must be initialized, and the order of fields must match the order in which they are declared in the structure. In this case the constraint doesn't hold, which is what the compiler is telling. To resolve the issue, you need to change the initialization code to something that C++ supports, for example:
```c++
const struct addrinfo hints = {);  // this zeroes out the structure fields
hints.ai_family = AF_INET;
hints.ai_socktype = SOCK_STREAM;
```

DasVoelkel, 2020-02-09:
Ah! ok thank your for the explaination! 
I didn't know that, until now it seems i was just lucky and dodged the problem

closed 2020-02-09 by DasVoelkel