espressif/esp-idf #4741: Compilation errors in kconfig files

luquinhas98, 2020-02-10:
Making menuconfig on the hello_world project gives me this strange error. I have tried to make clean the tools/kconfig directory and get the same. I'm running the command on 32-bit MSYS2.

> $ make menuconfig
> 
> make[1]: Entering directory '/home/lucas/esp/esp-idf/tools/kconfig'
> /usr/bin/gcc -c  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -MMD -MP -I "." -I "/home/lucas/e       sp/esp-idf/tools/kconfig"  /home/lucas/esp/esp-idf/tools/kconfig/mconf.c -o mconf.o
> flex -L -Pzconf -ozconf.lex.c /home/lucas/esp/esp-idf/tools/kconfig/zconf.l
> bison -t -l -p zconf -o zconf.tab.c /home/lucas/esp/esp-idf/tools/kconfig/zconf.y
> sed -E "s/\\x0D$//" /home/lucas/esp/esp-idf/tools/kconfig/zconf.gperf | gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$'        -p -t
> /usr/bin/gcc -I /home/lucas/esp/esp-idf/tools/kconfig -c  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -       DLOCALE -MMD -MP -I "." -I "/home/lucas/esp/esp-idf/tools/kconfig"  zconf.tab.c -o zconf.tab.o
> In file included from zconf.tab.c:2457:0:
> zconf.lex.c:853:2: error: expected identifier or ‘(’ before ‘do’
>         49,   49,   49,   49,   49,   49,   49,   49,   49,   49
>   ^
> zconf.lex.c:873:2: error: expected identifier or ‘(’ before ‘while’
>         11,  -52,  -52,  -52,  -52,  -52,  -52,  -52,  -52,  -52,
>   ^
> zconf.lex.c:1033:27: error: expected identifier or ‘(’ before ‘?’ token
>         12,   12,   12,   12,   12,   12,   12,   12,   12,   12,
>                            ^
> zconf.lex.c:1123:2: error: expected identifier or ‘(’ before ‘{’ token
>   }
>   ^
> zconf.lex.c:1141:2: error: expected identifier or ‘(’ before ‘{’ token
> 
>   ^
> zconf.lex.c:1993:2: warning: data definition has no type or storage class
>   int yy_is_jam;
>   ^
> In file included from zconf.tab.c:2457:0:
> zconf.lex.c:49:16: warning: type defaults to ‘int’ in declaration of ‘zconftext’ [-Wimplicit-int]
>  #endif
>                 ^
> zconf.lex.c:1195:20: note: in expansion of macro ‘yytext’
> 
>                     ^
> zconf.lex.c:1993:3: note: in expansion of macro ‘yytext_ptr’
>   int yy_is_jam;
>    ^~~~~~~~~~
> zconf.lex.c:49:16: error: conflicting types for ‘zconftext’
>  #endif
>                 ^
> zconf.lex.c:1195:20: note: in expansion of macro ‘yytext’
> 
>                     ^
> zconf.lex.c:1993:3: note: in expansion of macro ‘yytext_ptr’
>   int yy_is_jam;
>    ^~~~~~~~~~
> zconf.lex.c:49:16: note: previous declaration of ‘zconftext’ was here
>  #endif
>                 ^
> zconf.lex.c:1187:14: note: in expansion of macro ‘yytext’
> 
>               ^
> In file included from zconf.tab.c:2457:0:
> zconf.lex.c:1993:17: error: ‘yy_bp’ undeclared here (not in a function)
>   int yy_is_jam;
>                  ^
> In file included from zconf.tab.c:2457:0:
> zconf.lex.c:39:16: warning: data definition has no type or storage class
>  #ifdef yy_create_buffer
>                 ^
> zconf.lex.c:1995:2: note: in expansion of macro ‘yyleng’
> 
>   ^
> zconf.lex.c:39:16: warning: type defaults to ‘int’ in declaration of ‘zconfleng’ [-Wimplicit-int]
>  #ifdef yy_create_buffer
>                 ^
> zconf.lex.c:1995:2: note: in expansion of macro ‘yyleng’
> 
[... 188 lines cut ...]
> zconf.lex.c: In function ‘yy_try_NUL_trans’:
> zconf.lex.c:3987:20: warning: initialization makes pointer from integer without a cast [-Wint-conversion]
> zconf.lex.c: In function ‘yyunput’:
> zconf.lex.c:4033:11: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
> zconf.lex.c:4089:15: warning: assignment makes integer from pointer without a cast [-Wint-conversion]
> zconf.lex.c:4093:15: warning: assignment makes integer from pointer without a cast [-Wint-conversion]
> zconf.lex.c: In function ‘zconfrestart’:
> zconf.lex.c:4267:28: error: expected ‘)’ before ‘{’ token
> zconf.tab.c:2463:0: error: expected declaration or statement at end of input
> In file included from zconf.tab.c:2457:0:
> zconf.tab.c: At top level:
> zconf.lex.c:1981:24: warning: ‘yy_fatal_error’ used but never defined
>    }
>                         ^
> In file included from zconf.tab.c:2457:0:
> zconf.lex.c:2207:13: warning: ‘zconf_endhelp’ used but never defined
> 
>              ^~~~~~~~~~~
> zconf.lex.c:2209:13: warning: ‘zconf_endfile’ used but never defined
> 
>              ^
> make[1]: *** [Makefile:179: zconf.tab.o] Error 1
> make[1]: Leaving directory '/home/lucas/esp/esp-idf/tools/kconfig'
> make: *** No rule to make target '/home/lucas/esp/esp-idf/tools/kconfig/conf-idf', needed by '/home/lucas/esp/esp-idf/tools/kconfig       /mconf-idf'.  Stop.
> )

dobairoland, 2020-02-10:
Hi @luquinhas98. Please give us more information about your system: ESP-IDF version, OS name and version, and the outputs of `flex --version` and `bison --version`.

Please try `make -C /home/lucas/esp/esp-idf/tools/kconfig clean` and see if it helps.

dobairoland, 2020-02-18:
Hi @luquinhas98. Are you still experiencing this problem?

You can also try the ZIP-ed MSYS environment where everything should work:
https://docs.espressif.com/projects/esp-idf/en/latest/get-started-legacy/windows-setup.html#toolchain-setup (for v4.0 or later versions)

luquinhas98, 2020-02-21:
Hello, I tried the prepackaged MSYS environment and everything works now. Thank you.

closed 2020-02-21 by luquinhas98