espressif/esp-idf #4742: Lost write responses with Nimble, but not with Bluedroid

zacwbond, 2020-02-10:
## Environment

- Development Kit:      none (custom HW)
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.1-dev-2047-g600d542f5
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): gcc8_2_0-esp-2019r2-linux-amd64
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         external 3.3V

## Problem Description

I have my ESP configured as a central, and it talks to an external peripheral remote control device that uses a long connection interval.  When it needs an update from the central, we use the following sequence:

1. Peripheral sends an indicate to the central.
2. Central issues a write-with-response to the peripheral.
3. Peripheral receives the write, and sends the response.

The remote is built of an NRF52 BLE, and the ESP32 is the central.

I recently updated our code to use the Nimble stack instead of the Bluedroid stack, and after the update I get frequent disconnects because the peripheral does not see some of the write-with-response messages sent in step 2.  The central then times out and closes the connection after awaiting the response to the write and not getting it.

I hooked up an NRF BLE sniffer to examine what's going on.  Using the sniffer, I examined communication with both the Nimble and the Bluedroid stacks.  In both cases, the peripheral device is completely unchanged, and in both cases, our connection parameters are as follows:

`min_int = 60, max_int = 120, conn_int = 120, latency = 90, timeout = 3000`

**With Bluedroid**
For every indicate received from the peripheral, I see that the the central sends a write-with-response.  I then always see the response to that write come either 150 or 300 ms after.  

But in cases where the central receives a new notify before receiving a prior write response, it will send a new write-with-response.  Later on, I will see two subsequent write responses, one for each write-with-response sent.

**With Nimble**
For every indicate received from the peripheral, I see that the central sends a write-with-response.  

But in cases where the central receives a new notify before receiving a prior write response, it will send a new write-with-response.  Later on, the connection will time out because the peripheral only sends a single response for the two writes-with-response that it has now received.

I don't know why changing the ESP central's stack would impact the behavior of the NRF peripheral whose code hasn't changed, but is Nimble resulting in a timing or some other difference that could be causing the failure?  I've attached two Wireshark logs for review.  

[BLE_Success.pcapng.log](https://github.com/espressif/esp-idf/files/4181927/BLE_Success.pcapng.log)
[BLE_Failure.pcapng.log](https://github.com/espressif/esp-idf/files/4181928/BLE_Failure.pcapng.log)

prasad-alatkar, 2020-02-12:
Hi @zacwbond, from what I understood from your issue description :

- Peripheral sends notification to get update from central. I suppose you are using `ble_gattc_notify_custom` API to do so. 
- Central issues write with response to characteristic value (`ble_gattc_write_flat`) after receiving notification.

From the sniffer logs, I could see average interval between successive write-requests as ~300msec, which is more than sufficient. 

So, with given connection parameters and above information, I tried to mimic the app behavior in `blecent` and `blehr` app (`examples/bluetooth/nimble`) and I did not face this issue. 

Basically at peripheral side, after sending notification, please check in GAP event if the notification is sent successfully (in case of NimBLE `BLE_GAP_EVENT_NOTIFY_TX`), if failed due to no memory (`BLE_HS_ENOMEM`), you can provide some delay before you send another notification. I understand peripheral app is same for both Bluedroid and NimBLE stack, so there should not be any issue, but please confirm the above behavior once. At central side, Can you please confirm you are not explicitly calling `ble_gap_terminate(peer->conn_handle, BLE_ERR_REM_USER_CONN_TERM)` for any failed case ? 
If possible, can you share console logs with debug prints (menuconfig-->component config-->log output-->  Default log verbosity to Debug) ? If the above mentioned steps do not help you to mitigate this issue, is it possible to provide demo application based on `blecent` from examples ?

zacwbond, 2020-02-12:
Hi @prasad-alatkar 

You are correct in the description, except that the peripheral is not an ESP32 device, it's an NRF51.  The NRF peripheral device has code that prevents sending a new notify until the prior notify is sent.  

The central is the ESP32 device, and it is switching the central from Bluedroid to Nimble that seems to cause the problem.

I can confirm that my code on the central isn't calling `ble_gap_terminate()`.  I believe the connection ends up being terminated in `ble_gattc_timer()` when the write-with-response operation times out.

I will acquire some more debug.  Last time I used it I didn't see much additional information: writes are sent, and some get responses, while others eventually time out.

I also have looked at debug on the NRF side of things.  They have event callbacks similar to ESP, one of which is `BLE_GATTS_EVT_WRITE`.  That event is never seen.  It's as if some of the writes observed in the sniffer trace somehow do not make it to the peripheral (but only with Nimble).  However, the peripheral continues sending out the right sequence numbers, so clearly it must have gotten the write.

In your test, did you specifically exercise the case of sending a new write before the response to the most recent write has been received?  That seems to be the situation in which it fails for Nimble.

Looking at my Bluedroid log, it seems this case never happens.  Does Bluedroid's write queue avoid actually sending write requests until the response to the prior write in its queue was received?  If so, maybe Nimble should be doing the same thing?

prasad-alatkar, 2020-02-13:
Hi @zacwbond , As per spec v4.2, Vol 3, Part F, section 3.3.2

> Many attribute protocol PDUs use a sequential request-response protocol.
Once a client sends a request to a server, that client shall send no other
request to the same server until a response PDU has been received.

So, application should not send a new write before the response to the most recent write has been received. However please share device side debug logs for this issue so as to understand it in better manner.

zacwbond, 2020-03-06:
I implemented a write queue in my application code and it has solved this problem, so I am closing the issue.  Thanks for the help!

closed 2020-03-06 by zacwbond