espressif/esp-idf #4743: vTaskDelay multicore bug (forever loop in vTaskDelay)

surfskidude, 2020-02-10:
## Environment

Development Kit: ESP32-WROVER-B
Module or chip used: WINGONEER T-Koala ESP32
IDF version: v4.1-dev-2055-gf5b82c5b1
Build System: CMake
Compiler version: crosstool-NG esp-2019r2) 8.2.0
Operating System: Linux
Power Supply: USB

## Problem Description

I have been working on finding a bug for the last two days and I now believe there is a bug in vTaskDelay. The code does not crash, but my task and possibly the network task get stuck in vTaskDelay. I can ping the board, but network is not working. Also my high priority debug thread is running.

Using debug printouts via a task with higher priority, I have found that my timer task does not return from vTaskDelay. I also get the following prinouts:

E (56667) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (56667) task_wdt:  - IDLE1 (CPU 1)
E (56667) task_wdt: Tasks currently running:
E (56667) task_wdt: CPU 0: IDLE0
E (56667) task_wdt: CPU 1: timer

I am porting another application to ESP, which already has a timer implementation and the timer internally uses vTaskDelay with variable timeouts.

The bug is only triggered if I activate another task, which initiates a client HTTP connection with a response timeout. I believe both my timer task and the lwIP network task goes into a foreverloop in vTaskDelay.

As a test, I have redesigned my timer to instead use xTimerCreate/xTimerStart and I can no longer reproduce the error.

I do not believe this is a stack issue and I have the following enabled:
$ grep ASSERT sdkconfig
CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_ENABLE=y
# CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_SILENT is not set
# CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_DISABLE is not set
CONFIG_FREERTOS_ASSERT_ON_UNTESTED_FUNCTION=y
CONFIG_FREERTOS_ASSERT_FAIL_ABORT=y
# CONFIG_FREERTOS_ASSERT_FAIL_PRINT_CONTINUE is not set
# CONFIG_FREERTOS_ASSERT_DISABLE is not set
# CONFIG_OPENSSL_ASSERT_DO_NOTHING is not set
CONFIG_OPENSSL_ASSERT_EXIT=y
CONFIG_OPTIMIZATION_ASSERTIONS_ENABLED=y
# CONFIG_OPTIMIZATION_ASSERTIONS_SILENT is not set
# CONFIG_OPTIMIZATION_ASSERTIONS_DISABLED is not set

Dazza0, 2020-02-10:
> I am porting another application to ESP, which already has a timer implementation and the timer internally uses vTaskDelay with variable timeouts.

Just to rule out the obvious. Check that the timer task isn't in a loop continuously calling `vTaskDelay(0)`.

> The bug is only triggered if I activate another task, which initiates a client HTTP connection with a response timeout. I believe both my timer task and the lwIP network task goes into a foreverloop in vTaskDelay.

What are the priorities and core affinities of the timer and HTTP task?

When the task WDT triggers, you can usually tell where a task is stuck from the backtrace. A backtrace will be printed if you configure the task WDT to panic on a timeout (via the `CONFIG_ESP_TASK_WDT_PANIC` option).

surfskidude, 2020-02-11:
Yes, I triple checked that I do not run into the condition: vTaskDelay(0)

Thanks for the CONFIG_ESP_TASK_WDT_PANIC tip. I'll try that soon.

I had the code compiled with -02. Going back do -g seems to remove the problem. Are there any known bugs in the Tensilica GCC version?

I am still investigation. I believe my own code works with -02, but not ESP-IDF libs. However, I had to add the following to this very basic function:

#ifdef ESP_PLATFORM
#pragma GCC push_options
#pragma GCC optimize ("O0")
#endif
DoubleLink*
DoubleList_removeFirst(DoubleList* o)
{
   if(o->next != (DoubleLink*)o)
   {
      DoubleLink* l = o->next;
      DoubleLink_unlink(l);
      return l;
   }
   return 0;
}
#ifdef ESP_PLATFORM
#pragma GCC pop_options
#endif

I found out by instrumenting the code that the C code line "if(o->next != (DoubleLink*)o)" produced assembler code with an infinite loop.

surfskidude, 2020-02-11:
After more testing, no code can be compiled with -O2. I had a crash inside the included Lua VM. The build appears to work after compiling everything with -g. The compiler is without doubt broken.

X-Ryl669, 2020-02-25:
Or you have a stack corruption somewhere and what you think your code is doing is not what the cpu is seeing.
You should enable stack canary in the config. I already had such strange errors in the past (and still do), and usually, they are all due to stack corruption.

surfskidude, 2020-02-26:
I doubt it is a stack issues since I have been running the code extensively since last email. I have an ESP32 that has been running for 10 days now without a problem.

X-Ryl669, 2020-02-26:
I never experienced compilation issue with the cross compiler from Espressif (I have even built my own to have  the `-fstack-usage` feature which wasn't included in Espressif's version). I had once a linking issue with the previous version (the linker produced invalid binary), but since 8.2, the compiler seems rock solid to me.

surfskidude, 2020-02-26:
BTW, I forgot to mention that I have always had stack checked enabled (check using canary bytes).

surfskidude, 2020-02-26:
The crashes with optimization appears random so I wonder if it could be related to pSRAM? -- the code is probably faster with optimization? I have my own allocator, which operates on a buffer declared as follows:
EXT_RAM_ATTR static char poolBuf[2*1024*1024]; 
I have attached my sdkconfig file.

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/4256729/sdkconfig.zip)

Dazza0, 2020-03-09:
@surfskidude 

> if it could be related to pSRAM?
Might be possible. We've had PSRAM bugs before and used some compiler workaround. Are you disabling the cache at any point in your application (e.g., some form of non-volatile storage doing a flash write).

I see in your configuration that you have `CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY` enabled. I'm wondering if it's possible that FreeRTOS static variables are being placed in PSRAM. Could you try turning off this option and see if the issue persists?

surfskidude, 2020-03-09:
No I am not disabling the cache and all options are based on the defaults from the esp-idf examples, except for CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY. When I turn this option off, I get the following compile error: region `dram0_0_seg' overflowed by 1972040 bytes
This is due to the pool I allocate as explained in my comment above.
EXT_RAM_ATTR static char poolBuf[0x200000‬];

Dazza0, 2020-03-10:
> EXT_RAM_ATTR static char poolBuf[0x200000‬];

What is being allocated in this allocated in this buffer? Is this being used in tandem with the ESP-IDF's heap allocator or is it a replacement?

> Is` the problem related to the following: #2892

It might be. But if it was a PSRAM issue, I would expect your application to crash in random places (i.e., lots of Load or Store prohibited). However, if your program appears to be stuck at random places, that suggests that the issue is either a FreeRTOS issue or a compiler issue (although your SDKCONFIG shows that `CONFIG_COMPILER_OPTIMIZATION_LEVEL_DEBUG=y`).

Could you enable the [core dump feature](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html#esp32-core-dump) (dumping to flash should be easiest), and collect a few crashes worth of core dumps. I want to see the backtraces of each task at the point where the Task WDT times out.

surfskidude, 2020-03-16:
Just to let you know that I have not had a chance to run this and also to make sure you full understand that the problem only occurs with optimization enabled.  The code works just fine with: CONFIG_COMPILER_OPTIMIZATION_LEVEL_DEBUG=y. In fact, I have had an ESP32 running for 25 days now without any issues. The code is available here (compiled with debug): https://github.com/RealTimeLogic/LspAppMgr-ESP32

surfskidude, 2020-06-21:
I just tested all of the above in the latest commit (625bd5e) and I am no longer able to reproduce the issues.

closed 2020-06-21 by surfskidude