espressif/esp-idf #4744: ESP-IDF Tools Installer v2.2 does not work due to two different issues

vmsh0, 2020-02-10:
## Environment

Windows 10 with latest updates applied. (Other environment items were not applicable.)

## Problem Description

I cannot install ESP-IDF Tools and ESP-IDF using the official installer. I would like to start developing for the ESP32 but I cannot install the tools. I would prefer not to install the environment manually.

I encounter two issues:

## Issue 1

### Expected Behavior

ESP-IDF is copied to destination directory after extraction and setp continues.

### Actual Behavior

Error message "Failed to copy ESP-IDF to the destination directory: E:\espidf\3.3" appears and setup halts.

### Steps to reproduce

1. Download and run `esp-idf-tools-setup-2.2.exe`
2. Choose option to download ESP-IDF (tested with 3.3 release and 3.3 branch)

## Debug Logs

[log1.txt](https://github.com/espressif/esp-idf/files/4182406/log1.txt)

Relevant part:

> 2020-02-10 19:00:44.329   --------
> 2020-02-10 19:00:44.332   Moving C:\Users\r\AppData\Local\Temp\is-EG1TH.tmp\esp-idf-v3.3.1 to E:\espidf\3.3
> 2020-02-10 19:00:44.332   Message box (OK):
>                           Failed to copy ESP-IDF to the destination directory: E:\espidf\3.3
> 2020-02-10 19:01:53.031   User chose OK.
> 2020-02-10 19:01:53.031   Message box (Yes/No):
>                           Installation log has been created, it may contain more information about the problem.
>                           Display the installation log now?
> 2020-02-10 19:01:53.962   User chose Yes.
> 2020-02-10 19:01:53.989   CurStepChanged raised an exception.

### Workaround

Go to the `.espressif\releases` directory, manually extract and copy esp-idf-v3.3.1.zip to ESP-IDF destination directory, and re-run installer.

## Issue 2

### Expected Behavior

ESP-IDF Tools are set-up.

### Actual Behavior

Error message "Installation has failed with exit code 1" appears and setup halts.

### Steps to reproduce

1. Download and run `esp-idf-tools-setup-2.2.exe`
2. Choose option to download ESP-IDF (tested with 3.3 release and 3.3 branch)
3. When installation fails, apply workaround from Issue 1
4. Re-run `esp-idf-tools-setup-2.2.exe`
5. Choose option to use the previously extracted ESP-IDF

## Debug Logs

[log2.txt](https://github.com/espressif/esp-idf/files/4182407/log2.txt)

Relevant part:

> 2020-02-10 19:10:02.444   Installing Python environment:C:\Users\r\AppData\Local\Programs\Python\Python37\python.exe E:\espidf\3.3\tools\idf_tools.py install-python-env
> 2020-02-10 19:10:03.415   Done, exit code=1
> 2020-02-10 19:10:03.415   --------
> 2020-02-10 19:10:03.415   Running command: C:\Users\r\AppData\Local\Programs\Python\Python37\python.exe E:\espidf\3.3\tools\idf_tools.py install-python-env
>                           usage: virtualenv [--version] [-v | -q] [--discovery {builtin}] [-p py] [--creator {builtin,cpython3-win,venv}] [--seeder {app-data,pip}] [--no-seed] [--activators comma_separated_list] [--clear] [--system-site-packages] [--copies]
>                                             [--download | --no-download] [--extra-search-dir d [d ...]] [--pip version] [--setuptools version] [--wheel version] [--no-pip] [--no-setuptools] [--no-wheel] [--clear-app-data] [--prompt prompt] [-h]
>                                             [dest]
>                           virtualenv: error: unrecognized arguments: --no-site-packages
>                           Creating a new Python environment in E:\espidf\.espressif\python_env\idf3.3_py3.7_env
>                           Traceback (most recent call last):
>                             File "E:\espidf\3.3\tools\idf_tools.py", line 1349, in <module>
>                               main(sys.argv[1:])
>                             File "E:\espidf\3.3\tools\idf_tools.py", line 1345, in main
>                               action_func(args)
>                             File "E:\espidf\3.3\tools\idf_tools.py", line 1156, in action_install_python_env
>                               stdout=sys.stdout, stderr=sys.stderr)
>                             File "C:\Users\r\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 347, in check_call
>                               raise CalledProcessError(retcode, cmd)
>                           subprocess.CalledProcessError: Command '['C:\\Users\\r\\AppData\\Local\\Programs\\Python\\Python37\\python.exe', '-m', 'virtualenv', '--no-site-packages', 'E:\\espidf\\.espressif\\python_env\\idf3.3_py3.7_env']' returned non-zero exit status 2.
>                           
> 2020-02-10 19:10:03.415   --------
> 2020-02-10 19:10:03.415   Message box (OK):
>                           Installation has failed with exit code 1
> 2020-02-10 19:10:21.217   User chose OK.
> 2020-02-10 19:10:21.235   Message box (Yes/No):
>                           Installation log has been created, it may contain more information about the problem.
>                           Display the installation log now?
> 2020-02-10 19:10:36.417   User chose Yes.
> 2020-02-10 19:10:36.460   CurStepChanged raised an exception.

igrr, 2020-02-10:
Thanks for reporting! The fix for Python environment installation is in review and will be released soon (along with a new version of the installer).

As a temporary workaround, you may downgrade `virtualenv` package as follows:

```
C:\Users\r\AppData\Local\Programs\Python\Python37\python.exe -m pip install --upgrade virtualenv=16.7.9
```

After this, retry the installation.

vmsh0, 2020-02-10:
Hi @igrr,
I confirm the workaround works. Thanks!

What about Issue 1? Not as inconvenient but I've seen many reports for that as well.

igrr, 2020-02-10:
@vmsh0 sure, checking that issue as well. Can you please link to the reports you've seen? To update them if it gets fixed.

igrr, 2020-02-10:
One more detail, did `E:\espidf\3.3` directory exist prior to running the installer? If yes, was it empty?

pavestru, 2020-02-10:
> As a temporary workaround, you may downgrade `virtualenv` package as follows:
> 
> ```
> C:\Users\r\AppData\Local\Programs\Python\Python37\python.exe -m pip install --upgrade virtualenv==16.7.9
> ```

Or remove `--no-site-packages` flag from `virtualenv` inside `idf_tools.py`. The flag has been [deprecated](https://github.com/pypa/virtualenv/blame/legacy/docs/reference.rst#L100) for quite some time (6 years).

igrr, 2020-02-10:
@pavestru yes, that's the fix which is going to be merged. For the users of the Windows tools installer the problem is that it's not trivial to modify idf_tools.py since it is extracted by the installer (at least when installing v3.3.1). For other OSes, modifying idf_tools.py can also be used as a workaround for now.

igrr, 2020-02-10:
Okay, I was able to reproduce the first issue as well. It happens because we are trying to use "RenameFile" function to move the extracted ESP-IDF directory into place, but this function doesn't work when the destination is on a different drive. This will also be fixed shortly.

vmsh0, 2020-02-10:
Hi @igrr,
Thank you for looking into the issue. I suspected it had something to do with different drives. I'm happy you'll be able to push a fix!

I had found issue #4128 and also one more, which I can't find anymore. (And then many help requests on esp32.com.)

zhongziming1999, 2020-02-11:
我也遇到了和楼主同样的问题，异常1和异常-2，非常困扰，等待乐鑫回复解决方法

armandr, 2020-02-11:
> @pavestru yes, that's the fix which is going to be merged. For the users of the Windows tools installer the problem is that it's not trivial to modify idf_tools.py since it is extracted by the [...]

would running this from CMD also work? it seems to be the last step of the install.

`python -m virtualenv "C:\\Users\\user\\.espressif\\python_env\\idf4.0_py3.7_env"`

igrr, 2020-02-11:
@armandr Yes, running this from CMD should work, provided that `python` refers to the same Python installation you are selecting in the Installer (in case you have more than one).

igrr, 2020-02-11:
The fix for the first issue has been merged in https://github.com/espressif/esp-idf/commit/7a18f02acd7005f7c56e62175a8d1968a1a9019d, backports to release/v4.1 and release/v4.0 are coming soon.

igrr, 2020-02-11:
If you are experiencing this issue (on Windows), please try this version of the installer: https://dl.espressif.com/dl/esp-idf-tools-setup-2.3.exe

MitchBradley, 2020-02-11:
I tried the installer version listed above and it worked - the --no-site-packages problem disappeared, and the installer finished, creating the proper shortcuts in the Start Menu etc.

vmsh0, 2020-02-11:
I confirm that the file copy issue ("Issue 1") has been resolved for me.

The installer also successfully installs the master branch (w/ Python 3.8), the 3.3.1 release version, but not the 4.0 branch version. Is 3.3.1 expected to work at all? I see the `--no-sites-package` fix hasn't been backported yet, so I find this surprising.

Also, what is the difference between "release version" and "branch version"? I though the first downloaded the SDK as a full zip package and the second cloned it from the Github branch, but it seems to me that the second one downloads a zip package as well.

igrr, 2020-02-11:
> The installer also successfully installs the master branch (w/ Python 3.8), the 3.3.1 release version, but not the 4.0 branch version. Is 3.3.1 expected to work at all? I see the --no-sites-package fix hasn't been backported yet, so I find this surprising.

Once the fix is backported to release/v3.3 and release/v4.0, these branches will work as well. The backport should land tomorrow, hopefully. For the v3.3.1 and v4.0 versions, the installer includes a workaround (uses a bundled version of idf_tools.py). This is because we can't change the release tags...

> Also, what is the difference between "release version" and "branch version"? I though the first downloaded the SDK as a full zip package and the second cloned it from the Github branch, but it seems to me that the second one downloads a zip package as well.

The difference between release branches and release versions is the difference between branches and tags. The installation flow for release branches is indeed to download the latest release tag for the branch (if one exists) and then use it as a reference repository when cloning the branch. Most of the time, this is the quickest way to download a release branch of IDF, and the final result is the same as if you would clone the branch from scratch.

closed 2020-02-12 by espressif-bot

andalera, 2020-02-14:
Thanks for the 2.3 version!  I'm up and running now!  I can confirm it resolves the error:
virtualenv: error: unrecognized arguments: --no-site-packages"
Cleaned everything out, and let it install python 3.7, git, and esp-idf release/4.0
All is well now.

Zeni241, 2020-02-14:
With https://dl.espressif.com/dl/esp-idf-tools-setup-2.3.exe installation was smooth. I used python 3.8, git 2.23.0, window .1.
Example hello_world running OK.

gsuberland, 2020-02-22:
Ran into this issue too, the 2.3 version works fine.

I can see that [windows-setup.rst](https://github.com/espressif/esp-idf/blob/master/docs/en/get-started/windows-setup.rst) has been updated to link to the 2.3 installer, but the [online docs](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html) don't seem to have reflected that change yet. Might be worth looking into that so more folks don't run into this problem.

igrr, 2020-02-22:
This is an unfortunate timing as we are transitioning to a new documentation host, still needs some work for the docs to go live: https://github.com/espressif/esp-idf/blob/master/README.md#setting-up-esp-idf

markwest, 2020-02-22:
I am running into the following issue with version 2.2 and 2.3 of the installer. Not sure if this is the correct thread for this issue.
subprocess.CalledProcessError: Command '['C:\\Users\\user\\AppData\\Local\\Programs\\Python\\Python38\\python.exe', '-m', 'virtualenv', '--no-site-packages', 'C:\\Users\\user\\.espressif\\python_env\\idf4.0_py3.8_env']' returned non-zero exit status 2.

igrr, 2020-02-22:
@markwest which IDF version are you using with the 2.3 installer? If you are using an existing IDF directory (not downloading a new one using the installer), please pull the latest version of your branch (master, release/v4.1, release/v4.0). The fixes for this issue have been merged into the respective branches.
Alternatively, you can use the installer to download a fresh copy of IDF, in this case the installation should succeed.

markwest, 2020-02-22:
@igrr I went back to the latest version of virtualenv (20.0.5), removed the existing IDF directory, and ran version 2.3 of the installer again. I did not encounter any issues with the install. Also, I successfully flashed and monitored the hello_world example. Thanks for your help.

xuhongv, 2020-02-23:
I choosed to do it with the installer from https://dl.espressif.com/dl/esp-idf-tools-setup-2.3.exe
I started esp-idf_tools-setup-2.2.exe as admin
Available Python version: Python 3.7 (64-bit)
Available Git versions: Install Git 2.21.0
Available ESP-IDF versions:  ESP8266-master

But error show : 
```
Installing Python environment:C:\Users\Python37\python.exe "D:\ESP8266_RTOS_SDK\tools\idf_tools.py" --idf-path D:\ESP8266_RTOS_SDK install-python-env
Done, exit code=1
```

markwest, 2020-02-23:
@xuhongv It sounds like your setup is similar to mine. I was getting the same error. In my case, I did not use an existing ESP-IDF framework. Instead, I allowed the installer to download a new one, which worked.

ralphcastro504, 2020-02-23:
ran with the same issue I installed v2.3 and then had exit code 2, did the process and then had exit code 4. Decided to follow @markwest recommendation, did everything from new and the installation went clean. 

Thanks all.

sheldon123z, 2020-02-27:
> Or remove `--no-site-packages` flag from `virtualenv` inside `idf_tools.py`. The flag has been [deprecated](https://github.com/pypa/virtualenv/blame/legacy/docs/reference.rst#L100) for quite some [...]

Hi, I met the same problem, and this method actually worked. after removing the flag, I am able to re-run the installer successfully on Windows 10.

paulzhn, 2020-02-29:
It seems like that latest version of virtualenv has removed the `--no-site-packages` flag and the created virtualenv doesn't inherit system site-packages by default. Instead, the flag `--system-site-packages` is added to inherit system site-packages.

zzeroo, 2020-03-04:
Just for the record

```diff
diff --git a/tools/idf_tools.py b/tools/idf_tools.py
index 158b5a764..32a7c57a1 100755
--- a/tools/idf_tools.py
+++ b/tools/idf_tools.py
@@ -1180,7 +1180,7 @@ def action_install_python_env(args):
             subprocess.check_call([sys.executable, '-m', 'pip', 'install', '--user', 'virtualenv'],
                                   stdout=sys.stdout, stderr=sys.stderr)

-        subprocess.check_call([sys.executable, '-m', 'virtualenv', '--no-site-packages', idf_python_env_path],
+        subprocess.check_call([sys.executable, '-m', 'virtualenv', idf_python_env_path],
                               stdout=sys.stdout, stderr=sys.stderr)
     run_args = [virtualenv_python, '-m', 'pip', 'install', '--no-warn-script-location']
     requirements_txt = os.path.join(global_idf_path, 'requirements.txt')
```

markhu, 2020-03-08:
This happened on MacOS also. The above workarounds worked.

bl4krat, 2020-04-12:
I had "issue 1" with version 2.2 of the installer too (installing to D:).  2.3 fixed it. But the "getting started website still points to the buggy version 2.2"

https://docs.espressif.com/projects/esp-idf/en/stable/get-started/windows-setup.html

efsieitch, 2020-05-02:
Did not notice the 2.3 version of tools setup. I ran the ver. 2.2 and ran into the Issue #2. I removed the '--no-site-packages' flag from 'idf_tools.py' and execute it again to make it work perfectly.

I realized the "stable" version of documentation still refering to the 2.2 version. (https://docs.espressif.com/projects/esp-idf/en/stable/get-started/windows-setup.html)
To see the 2.3 version link, change the doc. version in the lower-left part of the page to "lastest" or enter this link (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html)

kifantidis, 2020-05-04:
Hello I would like to share my experience with you in case someone falls into the same problem as I did. 
I've been trying to install esp-idf stable version v4.0 with https://dl.espressif.com/dl/esp-idf-tools-setup-2.3.exe but I've received many fails until I've got it working. 
Things I did in order for the installation to succeed was:
1. Install python 3.7 from esp-idf-tools-setup-2.3
2. I had git already installed
3. Install esp-idf v4.0 (not release) from esp-idf-tools-setup-2.3
4. Install .espressif's tools from esp-idf-tools-setup-2.3
This way I've got installation working BUT! The esp-idf v4.0 was not the one that should have been. So next I deleted esp-idf folder and downloaded the latest esp-idf v4.0 with git. 

I don't know if this problem came up because I didn't download the release/v4.0 version within esp-idf-tools-setup-2.3 installation. 
After this if you want to work with any IDE you have to add tool's paths in "System environment variables".
I hope this will help anyone who get the same problem as me. 

Regards, Kostas Ifantidis

projectgus, 2020-05-05:
@kifantidis,

Thanks for sharing this. Can you please describe the failures that you experienced before you got everything working?

Angus

kifantidis, 2020-05-05:
The thing that I've noticed is that the timeout feature in esp_tls_conn_new(); function timeout configuration was missing. That was causing the socket to stay inside in infinite loop. Timeout also wasn't working for esp_https_ota_begin(); function.

projectgus, 2020-05-05:
Hi @kifantidis , OK, thanks for noting this. This is probably not related to the ESP-IDF Tools installer, although it might related to the exact ESP-IDF version. We would like to help investigate it. Could you please open a new issue and fill out the details asked for in the new issue template?

dipen-1337lab, 2020-05-08:
Namaste,

I too ran into these two different issues.

As suggested by many, used the latest **esp-idf-tools-setup-2.3.exe** installer along with **ESP-IDF-Ver-4.0**.

Installation completed successfully and the final installation logs on the **ESP-IDF Command Prompt** are as below:

Using Python in C:\Program Files (x86)\Python38-32\
Python 3.8.2
Using Git in C:\Program Files\Git\cmd\
git version 2.26.2.windows.1
Setting IDF_PATH: D:\espIdfVer40

Adding ESP-IDF tools to PATH...
    D:\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin
    D:\.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin
    D:\.espressif\tools\cmake\3.13.4\bin
    D:\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin
    D:\.espressif\tools\mconf\v4.6.0.0-idf-20190628\
    D:\.espressif\tools\ninja\1.9.0\
    D:\.espressif\tools\idf-exe\1.0.1\
    D:\.espressif\tools\ccache\3.7\
    D:\.espressif\python_env\idf4.0_py3.8_env\Scripts
    D:\espIdfVer40\tools

Checking if Python packages are up to date...
Python requirements from D:\espIdfVer40\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

D:\espIdfVer40>

ORParga, 2020-05-16:
I ran **esp-idf-tools-setup-2.3.exe** with [Python 3.8.3 - May 13, 2020 x84-64](https://www.python.org/downloads/windows/) and **ESP-IDF-Ver-4.0** in Windows10 64bits 

and I tested **hello world** example

Everything OK.

har-in-air, 2021-01-06:
I just tried running  esp-idf-tools-setup-2.3.exe on Windows 10 x64 with previously installed miniconda python  v3.8.5, Git  2.30.0, selected esp-idf-4.1, and I get the following error (from the end of the setup log):

2021-01-06 16:25:44.893   Installation process succeeded.
2021-01-06 16:25:44.899   Setting PATH for this process: C:\Users\hari\miniconda3\;C:\Program Files\Git\cmd\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\AMD\ATI.ACE\Core-Static;C:\Program Files\Git\cmd;C:\Users\hari\AppData\Local\Microsoft\WindowsApps;;C:\Modeltech_pe_edu_10.4a\win32pe_edu;C:\Users\Hari\Anlogic\TD4.6.4\bin\
2021-01-06 16:25:44.899   PYTHONPATH=
2021-01-06 16:25:44.900   idf_tools.py exists in IDF directory
2021-01-06 16:25:44.900   idf_tools.py command: C:\Users\hari\miniconda3\python.exe "C:\Users\hari\esp32\esp-idf-v4.1\tools\idf_tools.py" --idf-path C:\Users\hari\esp32\esp-idf-v4.1
2021-01-06 16:25:44.900   Installing tools:C:\Users\hari\miniconda3\python.exe "C:\Users\hari\esp32\esp-idf-v4.1\tools\idf_tools.py" --idf-path C:\Users\hari\esp32\esp-idf-v4.1 install
2021-01-06 16:25:45.904   Done, exit code=1
2021-01-06 16:25:45.904   --------
2021-01-06 16:25:45.904   Running command: C:\Users\hari\miniconda3\python.exe "C:\Users\hari\esp32\esp-idf-v4.1\tools\idf_tools.py" --idf-path C:\Users\hari\esp32\esp-idf-v4.1 install
                          Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache
                          Installing xtensa-esp32-elf@esp-2020r2-8.2.0
                          Downloading xtensa-esp32-elf-gcc8_2_0-esp-2020r2-win64.zip to C:\Users\hari\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp-2020r2-win64.zip.tmp
                          WARNING: Download failure <urlopen error unknown url type: https>
                          WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2020r2-win64.zip to C:\Users\hari\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp-2020r2-win64.zip.tmp
                          Downloading xtensa-esp32-elf-gcc8_2_0-esp-2020r2-win64.zip to C:\Users\hari\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp-2020r2-win64.zip.tmp
                          WARNING: Download failure <urlopen error unknown url type: https>
                          WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2020r2-win64.zip to C:\Users\hari\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp-2020r2-win64.zip.tmp
                          Downloading xtensa-esp32-elf-gcc8_2_0-esp-2020r2-win64.zip to C:\Users\hari\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp-2020r2-win64.zip.tmp
                          WARNING: Download failure <urlopen error unknown url type: https>
                          WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2020r2-win64.zip to C:\Users\hari\.espressif\dist\xtensa-esp32-elf-gcc8_2_0-esp-2020r2-win64.zip.tmp
                          ERROR: Failed to download, and retry count has expired
                          Traceback (most recent call last):
                            File "C:\Users\hari\esp32\esp-idf-v4.1\tools\idf_tools.py", line 1387, in <module>
                              main(sys.argv[1:])
                            File "C:\Users\hari\esp32\esp-idf-v4.1\tools\idf_tools.py", line 1383, in main
                              action_func(args)
                            File "C:\Users\hari\esp32\esp-idf-v4.1\tools\idf_tools.py", line 1167, in action_install
                              tool_obj.download(tool_version)
                            File "C:\Users\hari\esp32\esp-idf-v4.1\tools\idf_tools.py", line 584, in download
                              raise DownloadError()
                          __main__.DownloadError

I googled this "Download failure: .... https"  error, and on  StackOverflow found the accepted fix is to install python openssl, which I did with "conda install  openssl".  Tried installation again, same error. Rebooted and tried again, same error.

krzychb, 2021-01-07:
@har-in-air this report looks like a duplicate of https://github.com/espressif/esp-idf/issues/6367. 

Please check solution in https://github.com/espressif/esp-idf/issues/6367. In case it does not help open a new ticket. This ticket is for older version of windows tools installer and closed.

har-in-air, 2021-01-08:
@krzychb Thanks for the info.
I was able to get the Windows 10 install and example project build working with v2.3.3 tools installer when I used the "install Python" option, instead of pointing to my existing miniconda Python 3.8.5 installation.