espressif/esp-idf #4746: Not able to get WIFI_EVENT_STA_DISCONNECTED event if WiFi(STA) + PPPoS

AxelLin, 2020-02-11:
Module or chip used: ESP32-WROOM-32
IDF version: v4.1-dev-2196-g64654c044479
Build System: CMake
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Operating System: Linux
Power Supply: USB

The PPPoS has unexpected impact on WiFi (STA) interface.
Test by bring up WIFI STA and PPPoS at the same time.
Move test device far away from the AP until not able to get ping response.
In wifi only case, I can get WIFI_EVENT_STA_DISCONNECTED event and when I move the device cose to the AP I can re-connect to AP.
In wifi + pppos case, I never get WIFI_EVENT_STA_DISCONNECTED event.
Thus not able to re-connect AP any more.

Below is the log with WiFi + PPPoS:
WiFi (STA) + PPPoS case, never get WIFI_EVENT_STA_DISCONNECTED event.
Thus not able to reconnect to AP.

I (206646) wifi: state: run -> init (22c0)
I (206647) wifi: pm stop, total sleep time: 806822 us / 203750738 us

I (206647) wifi: new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1

Below is the log with WiFi (STA) only for comparison:

I (663985) wifi: state: run -> init (6c0)
I (663986) wifi: pm stop, total sleep time: 805310 us / 661638099 us

I (663987) wifi: new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
I (663995) Got WIFI_EVENT_STA_DISCONNECTED
I (664115) wifi: new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
I (664116) wifi: state: init -> auth (b0)
I (665117) wifi: state: auth -> init (200)
I (665118) wifi: new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
I (665118) wifi: new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
I (665123) wifi: state: init -> auth (b0)
I (665139) wifi: state: auth -> assoc (0)
I (665151) wifi: state: assoc -> run (10)
I (665224) wifi: connected with TEST_AP, aid = 6, channel 3, BW20, bssid = ac:84:c6:32:14:77
I (665225) wifi: security type: 4, phy: bgn, rssi: -82
I (665227) wifi: pm start, type: 0

I (665237) wifi: AP's beacon interval = 102400 us, DTIM period = 1
I (665244) Got WIFI_EVENT_STA_CONNECTED

david-cermak, 2020-02-11:
Hi @AxelLin 

Thank you for the report. This time it is a deadlock, making the ppp-netif default interface after the STA lost IP, but calling the "user" version (remote call to tcpip thread) while already on lwip context.

Posting the fix here: 
[0001-ESP-NETIF-Set-default-interface-for-ppp-netif-must-b.patch.txt](https://github.com/espressif/esp-idf/files/4184891/0001-ESP-NETIF-Set-default-interface-for-ppp-netif-must-b.patch.txt)
But will test it a bit more before pushing these commits.

Thanks again for all the recent testing you've provided and posting the relevant issue. Sorry for the trouble, the pppos client example and the related component clearly need more testing.

AxelLin, 2020-02-11:
Hi @david-cermak 

Thanks a lot for the quick fix.
I just tested the patch and it works.

Thank you!

umer-ilyas, 2020-02-11:
thanks @david-cermak i am having almost same issue, will test your patch.

AxelLin, 2020-02-13:
Hi @david-cermak 

I'm checking your change again but it seems incorrect in some places.
With your change, esp_netif_ppp_set_default_netif() must be called from lwip thread.
But in esp_netif_start() and esp_netif_stop() , it calls esp_netif_update_default_netif() in non-lwip thread if esp_netif->is_ppp_netif.
So I think your change still violate the lwip's threading model.

david-cermak, 2020-02-13:
@AxelLin You are correct, this was just a quick fix, an updated version is here
[0001-ESP-NETIF-Set-default-interface-for-ppp-netif-must-b.patch.txt](https://github.com/espressif/esp-idf/files/4199079/0001-ESP-NETIF-Set-default-interface-for-ppp-netif-must-b.patch.txt)

but please wait for the final commit closing this issue, this is still a WIP version.

AxelLin, 2020-02-13:
Hi @david-cermak 

I just checked your new patch, but the issue is still there.

e.g.
esp_err_t esp_netif_start(esp_netif_t *esp_netif)
{
    if (esp_netif->is_ppp_netif) {
        // No need to start PPP interface in lwip thread
        esp_err_t ret = esp_netif_start_ppp(esp_netif->lwip_ppp_ctx);
        if (ret == ESP_OK) {
            /* Axel: Below call is not safe because this is not called from lwip task context . */
            esp_netif_update_default_netif(esp_netif, ESP_NETIF_STARTED);
        }
        return ret;
    }
    return esp_netif_lwip_ipc_call(esp_netif_start_api, esp_netif, NULL);
}

The same issue in esp_netif_stop.

AxelLin, 2020-02-13:
Hi @david-cermak 

BTW, there is another routing problem.

This routing problem can be easily tesed in STA+PPPoS use case.
Take esp_netif_stop() as example:
The esp_netif_stop_ppp() does not set the inteface down.
so calling esp_netif_update_default_netif() will still choose ppp as default route even it's stop.
Add netifapi_netif_set_down(esp_netif->lwip_netif); before calling esp_netif_update_default_netif() fixes the issue so it can set previous default interface.
Similarly in esp_netif_start(), needs to call netifapi_netif_set_up(esp_netif->lwip_netif); before esp_netif_update_default_netif(esp_netif, ESP_NETIF_STARTED);

ps. I use netifapi_netif_set_up/down because it's not in lwip task context. and this needs to enable LWIP_NETIF_API in lwipopts.h.
Anyway, just want to point out the issue.

AxelLin, 2020-02-13:
> Hi @david-cermak
> 
> I just checked your new patch, but the issue is still there.
Oops. I misread the code.
The new patch is fine.

AxelLin, 2020-02-29:
Hi @david-cermak 

Just wondering how is the status.

AxelLin, 2020-03-07:
> Hi @AxelLin
> 
> Thank you for the report. This time it is a deadlock, making the ppp-netif default interface after the STA lost IP, but calling the "user" version (remote call to tcpip thread) while already on lwip context.

Hi @david-cermak,

I don't expect a dead-lock issue needs such long time to fix.
I'm wondering if you hit other issue?
Just want to remind such dead-lock issue can block finding other issue.
e.g. as I mentioned the routing issue.
So had better get the bug fix first so we can move on.

Alvin1Zhang, 2020-03-10:
@AxelLin Thanks for reporting and very sorry for the slow turnaround, currently we are still investigating deep into the issue, and will share updates once any progress. Thanks.

AxelLin, 2020-03-10:
Hi @Alvin1Zhang 

Thanks for the update.
I have tested @david-cermak 's patch 0001-ESP-NETIF-Set-default-interface-for-ppp-netif-must-b.patch.txt
for quite long period and it works well regarding fix the dead-lock issue.

Regarding the routing issue, I had sent the code showing how I address the issue in lwIP way.
Maybe you guys have better idea to address the routing issue in esp way.

closed 2020-03-20 by espressif-bot

Torarqui, 2020-03-25:
I'm having this same issue, but on IDF 4.0, can you provide a patch to that version? Thanks

Torarqui, 2020-03-26:
@david-cermak  I see. Thanks for the quick response, I'll try use the lwip API directly and will open a separate issue if needed.

umer-ilyas, 2020-04-28:
do we have a **WiFi(STA) + PPPoS** example of idf-v4.1

can anybody provide this @AxelLin