espressif/esp-idf #4748: make export.sh less verbose

DiegoActionItem, 2020-02-11:
I currently have a hack (a modified `export.sh`) which defines an internal echo command:

```
function esp_echo() {
    if [[ -n "${IDF_EXPORT_VERBOSE}" ]]
    then
   echo $1 $2 $3
    fi
}
```

and all references to "echo" have been redefined to be "esp_echo()" - but still `esp-idf/tools/check_python_dependencies.py` is verbose (see last line of that script). My workaround is :

```
 +    if [[ -n "${IDF_EXPORT_VERBOSE}" ]]
 +    then
 +        python ${IDF_PATH}/tools/check_python_dependencies.py || return 1
 +    else
 +        python ${IDF_PATH}/tools/check_python_dependencies.py | grep -v "are satisfied" || return 1
 +    fi
```

I am hoping for a better solution... I think teaching the python script about the new ENV variable. Can we find a better alternative? (I can try to do a PR "soon" but as this requires legal thingies, a series of trivial patches here will be easier IMHO).

```

igrr, 2020-02-11:
@DiegoActionItem Can you try the following command as an alternative to export.sh? (IDF_PATH must be set already):
```
eval $($IDF_PATH/tools/idf_tools.py --quiet export)
```

DiegoActionItem, 2020-02-12:
@igrr  nice! Works as expected

Can you update the official documentation? I am keeping that snippet in my `~/.bash_profile` and it makes my interactive logins quite noisy.

igrr, 2020-02-12:
No, sorry, we'd rather have the version in the docs which prints sufficient information for us to debug possible issues.

There are some docs on idf_tools.py, in case you are interested: https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-tools.html#idf-tools-py-script

closed 2020-02-12 by DiegoActionItem