espressif/esp-idf #4749: intialising wifi_config_t without init list makes wifi not connect

DasVoelkel, 2020-02-11:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):  "version": "3.30300.190916" according to platformios json of the package
- Build System:        platformio
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:    Linux

- Using an IDE?: Yes  VSCode with platformio
- Power Supply:         USB

## Problem Description
https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/station/main/station_example_main.c

Works perfectly fine for my wifi.
When i try to compile it to cpp, the initializer list won't work.
Which is the same as here and not fault of the library:
https://github.com/espressif/esp-idf/issues/4736

But when I try to init the value like this, from a char* given to the function to start a wifi with a specific name:

```
wifi_config_t wifi_config = {
    .sta = {
        .ssid = {0},
        .password = {0},
        .scan_method = (wifi_scan_method_t)0,
        .bssid_set = false,
        .bssid = {0},
        .channel = 0,
        .listen_interval = 0,
        .sort_method = (wifi_sort_method_t)0,
        .threshold = {0, (wifi_auth_mode_t)0}},
};

void wifi_init_sta(char *ssid_, char *pw_)
{
    s_wifi_event_group = xEventGroupCreate();

    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

wifi_config_t second_config;

    strncpy( (char*)second_config.sta.ssid,ssid_,32);
    strncpy( (char*)second_config.sta.password,pw_,64);

    for (int i = 0; i < strlen(ssid_); i++)
        wifi_config.sta.ssid[i] = (uint8_t)ssid_[i];

    for (int i = 0; i < strlen(pw_); i++)
        wifi_config.sta.password[i] = (uint8_t)pw_[i];

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &second_config));
    ESP_ERROR_CHECK(esp_wifi_start());

    printf("[DEBUG] config 1 :\n");
for (int i = 0; i < sizeof wifi_config; i++) {
      if(i == 96)
    printf("\n\n");
    printf(" %02x", ((uint8_t*)&wifi_config)[i]);
  
}
printf("\n");

printf("[DEBUG] wifi config 2:\n");
for (int i = 0; i < sizeof second_config; i++) {
      if(i == 96)
    printf("\n\n");
    printf(" %02x", ((uint8_t*)&second_config)[i]);
}
printf("\n");

}
```

Wifi will say that the AP can not be found, although the  data is the same.
Hexdumping the whole thing reveals:

```
[DEBUG] config 1 :
 57 6f 68 6e 7a 69 6d 6d 65 72 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 63 6c 6f 75 64 79 63 61 72 74 6f 6f 6e 32 30 33 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] wifi config 2:
 57 6f 68 6e 7a 69 6d 6d 65 72 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 63 6c 6f 75 64 79 63 61 72 74 6f 6f 6e 32 30 33 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

 a6 00 00 00 43 01 40 3f 23 00 06 00 cd cd 00 00 a8 39 fb 3f 03 00 00 00 fc 0e fc 3f

```
There is data at the end which config 1 is missing.

Initializing the config like config 2 seems incredibly ugly and is the only way i found to make this work.

### Expected Behavior

For all bytes in the wifi_config_t object to be set the same, and therefor wifi connecting to the network as expected

DasVoelkel, 2020-02-11:
dumping both structs before ssid and pw are written to them creates:
```
[DEBUG] config 1 :
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] wifi config 2:
 61 70 69 6e 01 00 00 00 a0 1a fb 3f a0 1a fb 3f 02 00 00 00 c0 13 0d 80 d0 1d fb 3f 80 77 12 40 03 00 00 00 10 1e fb 3f e0 1d fb 3f 0c 00 00 00 60 80 42 3f e0 1d fb 3f 0c 00 00 00 01 00 00 00 00 00 00 00 b2 45 0d 80 30 1e fb 3f a4 80 f4 3f 00 00 00 00 10 1e fb 3f e0 1d fb 3f 0c 00 00 00

 80 f8 01 80 0c 00 00 00 0c 00 00 00 cd cd 00 00 a6 00 00 00 43 01 40 3f 23 00 06 00
```

It seems like the garbage at the end makes the second config work over the first?

igrr, 2020-02-11:
```
wifi_config_t second_config;
```

This structure is not initialised, and will contain garbage values.
You need to zero it out before usage, in either of the two ways:

```c++
wifi_config_t second_config = {};
```

or 

```c++
wifi_config_t second_config;
memset(&second_config, 0, sizeof(second_config);
```

DasVoelkel, 2020-02-11:
ah ok i mixed up the configs when testing... Sorry to bother

closed 2020-02-11 by DasVoelkel