espressif/esp-idf #4755: NVS crashes on nvs_page.cpp - Page::copyItems - assert(end<=ENTRY_COUNT);

egoncalves-cutitronics, 2020-02-12:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v3]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.3
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a5
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes (Eclipse)]
- Power Supply:         [USB]

## Problem Description

NVS crashes on file nvs_page.cpp - function Page::copyItems - statement assert(end<=ENTRY_COUNT);

I have made a program that stresses the NVS system by calling many times the function nvs_set_u32(). In total around 7 Kbytes are written to the NVS system.
This programs makes NVS crash in:
file nvs_page.cpp
function Page::copyItems
statement assert(end<=ENTRY_COUNT);

The message shown is:

==========================================

assertion "end <= ENTRY_COUNT" failed: file "C:/esp/esp-idf/components/nvs_flash/src/nvs_page.cpp", line 437, function: esp_err_t nvs::Page::copyItems(nvs::Page&)
abort() was called at PC 0x400d3053 on core 0

ELF file SHA256: b81f491a6da62d355bf07a9ea5eee6db31978fe608754eea176a25651f191bed

Backtrace: 0x40086fa0:0x3ffbb4c0 0x400871e9:0x3ffbb4e0 0x400d3053:0x3ffbb500 0x400e2fb3:0x3ffbb530 0x400e2114:0x3ffbb570 0x400e1d23:0x3ffbb590 0x400e11a3:0x3ffbb600 0x400e9e74:0x3ffbb640 0x400e7794:0x3ffbb690 0x400e4d35:0x3ffbb720 0x400e50a3:0x3ffbb750 0x400e51b3:0x3ffbb770 0x400d2bb9:0x3ffbb790 0x400d158c:0x3ffbb7b0 0x40089e39:0x3ffbb7d0

Rebooting...
I (122) cpu_start: Pro cpu up.
I (122) cpu_start: Application information:
I (122) cpu_start: Project name:     unit_test_test
I (125) cpu_start: App version:      touchsensor_testing-56-g8ddbcf7

==========================================

### Code to reproduce this issue

```cpp

#include <esp_err.h>

esp_err_t LOG_Format ( char *partition_label )
{
    esp_err_t err;

    err = nvs_flash_init_partition ( partition_label );
    if ( err != ESP_OK )
	return err;

    err = nvs_flash_erase_partition ( partition_label );
    if ( err != ESP_OK )
	return err;

    err = nvs_open_from_partition ( partition_label, "LOG"/*namespace*/, NVS_READWRITE, &partition_handle );
    if ( err != ESP_OK )
	return err;

    err = nvs_open_from_partition ( partition_label, "LOG"/*namespace*/, NVS_READWRITE, &partition_handle );
    if ( err != ESP_OK )
	return err;

    err = nvs_set_u32 ( partition_handle, "Current Index", 0 );
    if ( err != ESP_OK )
	return err;

    return ESP_OK;
}

esp_err_t LOG_Open ( char *partition_label )
{
    esp_err_t err;

    err = nvs_open_from_partition ( partition_label, "LOG"/*namespace*/, NVS_READWRITE, &partition_handle );
    if ( err != ESP_OK )
	return err;

    return ESP_OK;
}

esp_err_t LOG_Write ( char *Text )
{
    esp_err_t err;
    static char log_str[200];
    char current_index_str [10+1]; // key is an unit32_t so 4 bytes which gives 10 digits
    uint32_t current_index;

	snprintf ( log_str, sizeof log_str, "%04d/%02d/%02d %02d:%02d:%02d,%s", 2020, 2, 12, 14, 39, 0, Text );

    err = nvs_get_u32 ( partition_handle, "Current Index", &current_index );
    if ( err != ESP_OK )
	return err;

    sprintf ( current_index_str, "%d", current_index );
    err = nvs_set_str ( partition_handle, current_index_str, log_str );
    if ( err != ESP_OK )
	return err;

    current_index++;

    err = nvs_set_u32 ( partition_handle, "Current Index", current_index );
    if ( err != ESP_OK )
	return err;

    err = nvs_commit ( partition_handle );
    if ( err != ESP_OK )
	return err;

    return ESP_OK;
}

	static char *partition_label = "nvs";

    err = LOG_Format ( partition_label );
    TEST_ASSERT_EQUAL ( ESP_OK, err );

    err = LOG_Open ( partition_label );
    TEST_ASSERT_EQUAL ( ESP_OK, err );

	for ( i=0 ; i<200 ; i++ )
	{
	    char Text [10];
	    sprintf ( Text, "LOG1 %d", i );
	    err = LOG_Write ( Text );
	}

0xjakob, 2020-02-13:
Hi @egoncalves-cutitronics, thanks for posting this issue! We're gonna have a look at it.
Quick question: why do you open the partition three times? Doesn't it return an error? Not sure whether it's related to the crash though.

egoncalves-cutitronics, 2020-02-13:
My mistake. The code is part of a UNITY test so it tries to stress everything, maybe a bit too much... :)

0xjakob, 2020-02-13:
Ok, good. I also don't expect the several openings to be related to that problem.

0xjakob, 2020-02-13:
Ok, good. I also don't expect the several openings to be related to that problem. I could reproduce the problem, it seems to happen while one of the last pages is requested. We'll continue to have a look into the problem. It also happens on the current master, so updating doesn't seem to help for now.

0xjakob, 2020-02-14:
Hello @egoncalves-cutitronics, could you please try to swap `nvs_flash_init_partition` and `nvs_flash_erase_partition`? It seems like one shouldn't erase a partition while it's initialized. Please let me know if that fixes the problem.
Of course, the documentation is not really clear about that and we'll probably build in a check as well...

egoncalves-cutitronics, 2020-02-14:
Hi
thanks for the suggestion, but after changing the code I still have the same problem.

0xjakob, 2020-02-14:
OK, just to confirm: you do `nvs_flash_erase_partition()` before anything else in your code?
If that's the case and the problem still persists, please post you code again as it is right now, if possible. Thanks!

egoncalves-cutitronics, 2020-02-14:
Yes.
Now I have corrected the code above as well.
I call LOG_Format() that erases the partition,  initialises and opens it.
After this is just a matter of performing many writes to the partition until it crashes.

0xjakob, 2020-02-14:
The crash is still the same? I just ran the code on two different devices and didn't see any issue. Even when writing beyond the capacity of the nvs partition.

0xjakob, 2020-02-14:
The only thing I changed is nvs includes and that I used this main function:
```
void app_main(void) {
	static char *partition_label = "nvs";

        LOG_Format ( partition_label );

    printf("start stress test\n");
	for (int i=0 ; i<400 ; i++ )
	{
	    char Text [20];
	    sprintf ( Text, "LOG1 %d", i );
	    LOG_Write ( Text );
	}
    printf("stop stress test\n");
}
```

egoncalves-cutitronics, 2020-02-14:
Yes. I tried also several times and I always got the crash.
I have even cleaned the project to be sure that the compilation of the changed code was done.
I'm using version 3.3, not the latets 4.0.
The NVS partition is quite big, almost 1Mbytes, the partition_layout.csv file defines NVS as:
nvs	            data	       nvs	        0x9000	     0xF4000

0xjakob, 2020-02-14:
OK, need to check. 
There is definitely a problem when erasing the flash while the same partition is initialized. This will be fixed. It had the same symptoms which you described above and the bugfix fixed your code (altered by me as decribed above) with the default partition size on the latest master branch.

0xjakob, 2020-02-17:
@egoncalves-cutitronics I can't produce any crash with version v3.3 and and the large partition either. Is it possible for you to share the exact error message and backtrace again? And maybe the original code you used (like do you do something else before calling the provided functions, e.g.)? Thanks!

egoncalves-cutitronics, 2020-02-18:
The trace is:
==============================================================
Running Testing LOG 1...
assertion "end <= ENTRY_COUNT" failed: file "C:/esp/esp-idf/components/nvs_flash/src/nvs_page.cpp", line 437, function: esp_err_t nvs::Page::copyItems(nvs::Page&)
abort() was called at PC 0x400d2feb on core 0

ELF file SHA256: ed6f7d8ac10a766c2ca5bfa8bf260705486afc48b5c35382d8cee3ee44c0e317

Backtrace: 0x40086fa0:0x3ffbb4f0 0x400871e9:0x3ffbb510 0x400d2feb:0x3ffbb530 0x400e2f07:0x3ffbb560 0x400e2098:0x3ffbb5a0 0x400e1ca7:0x3ffbb5c0 0x400e0ba1:0x3ffbb630 0x400e1099:0x3ffbb680 0x400e9cd2:0x3ffbb6a0 0x400e75fc:0x3ffbb6f0 0x400e4c89:0x3ffbb720 0x400e4ff7:0x3ffbb750 0x400e5107:0x3ffbb770 0x400d2b65:0x3ffbb790 0x400d1538:0x3ffbb7b0 0x40089e39:0x3ffbb7d0

Rebooting...
ets Jun 8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4188
load:0x40078000,len:11100
ho 0 tail 12 room 4
load:0x40080400,len:6084
entry 0x400806e0
I (155) cpu_start: Pro cpu up.
I (155) cpu_start: Application information:
I (155) cpu_start: Project name: unit_test_test
... (this is what one gets after reset)
==============================================================

0xjakob, 2020-02-18:
Could you download and post the contents of the flash?
You can use `esptool.py` like this (or adapt the data if you changed it eventually):
`components/esptool_py/esptool/esptool.py  read_flash 0x9000 0xF4000 nvs_dump.bin`
Then we can upload it to our testing system and have a look. 
Thank you.

egoncalves-cutitronics, 2020-02-18:
I tried that command before:

C:\esp\esp-idf\components\esptool_py\esptool\esptool.py read_flash 0x9000 0xF4000 C:\temp\nvs_dump.bin

but it doesn't seem to work...
No file is produced in the C:\temp\ directory
There is a cmd window that shows up for a brief moment just to disappear without allowing any reading!
Is there a change that can be made to the .py script in order to have a pause?

igrr, 2020-02-18:
Due to file associations on Windows the Python file may open in some other program (not the currently active Python interpreter). If you try adding `python ` in front of this command, it might fix the issue.

egoncalves-cutitronics, 2020-02-18:
Success :)

C:\esp\esp-idf\tools>python C:\esp\esp-idf\components\esptool_py\esptool\esptool.py --port COM13 read_flash 0x9000 0xF4000 c:\temp\nvs_dump.bin
esptool.py v2.8-dev
Serial port COM13
Connecting.....
Detecting chip type... ESP32
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: bc:dd:c2:da:21:60
Uploading stub...
Running stub...
Stub running...
999424 (100 %)
999424 (100 %)
Read 999424 bytes at 0x9000 in 92.6 seconds (86.3 kbit/s)...
Hard resetting via RTS pin...

I changed the extension from .bin to .txt since git didn't accept the .bin

[nvs_dump.txt](https://github.com/espressif/esp-idf/files/4220317/nvs_dump.txt)

0xjakob, 2020-02-18:
The flash partition contains old garbage data and was definitely not overwritten with with a pattern. I can also reproduce this, if I try this on my board. `nvs_flash_erase_partition()` doesn't seem to work correctly.
@egoncalves-cutitronics can you try this, if possible:
`idf.py erase_flash`? This erases the whole flash of your ESP32! so please check that you don't have other important data like another nvs partition or so on your board.
Does the application work again afterwards?

0xjakob, 2020-02-19:
If you can run your tests on another board for comparison, it would be great, too.

egoncalves-cutitronics, 2020-02-19:
Hi
I have erased the flash completely confirming the operation by dumping it.
However when running the application I still get the same problem...

0xjakob, 2020-02-19:
How many times do you write now? Is it still the same number as before? You said "After this is just a matter of performing many writes to the partition until it crashes.", so I assume you increased the number of calling `LOG_Write()`, is that correct? Does the crash happen each time you run the app or periodically/randomly?

0xjakob, 2020-02-19:
And it would be still interesting to see how the application behaves on another ESP32 board to rule out flash hardware issues. Do you know roughly about how many times the flash was written already or how many times the application ran?

egoncalves-cutitronics, 2020-02-20:
It crashes if I perform some calls to LOG_Write(). It breaks after 170 calls although that I would say depends of the amount of data written.
I use the demo board and it also breaks the same away with the same number of calls.

0xjakob, 2020-02-20:
@egoncalves-cutitronics, you may want to remove the posted flash partition again. I inspected it once more and as it turns out, it has the old partition layout with the small NVS partition (0x9000 0x6000), followed by the PHY init and the app partition. It can even run the app and I see that the problem persists on my device with your code. 
Is it possible to isolate the unit test and post the code so that we have the exact same source code to reproduce it on our devices? Thanks!

egoncalves-cutitronics, 2020-02-21:
sorry, but could you explain what you pretend me to do?
I cannot understand the sentence: "you may want to remove the posted flash partition again"

0xjakob, 2020-02-24:
@egoncalves-cutitronics  It's just that with the "nva-partition" flash-copy, all the other unit tests are visible if you flash it. If that's a problem, than you can remove it. If not, then leave it.
That aside, is it possible to create a single test case where the problem occurs which we can run on our devices? Again, we can't reproduce the crash on our devices, yet, if we flash your posted flash-image and run the code, then indeed the crash occurs. It may be related to the other unit test cases.
It also looks like the nvs partition size is still the small one which is the default nvs partition size.

Alvin1Zhang, 2020-05-26:
Will close due to short of feedback/updates, feel free to reopen with more details. Thanks.

closed 2020-05-26 by Alvin1Zhang