espressif/esp-idf #4756: make: *** No rule to make target 'menu'.  Stop.

varunsavita, 2020-02-12:
1:- Hello_world code Path -
**varun@LAPTOP-RFBF50C5 MINGW32 ~/esp-idf/esp-idf/examples/hello_world_back**
$ ls
CMakeLists.txt  loadable_elf_example_test.py  main  Makefile  README.md  sdkconfig.ci

2:- IDF_PATH 
$ printenv IDF_PATH
**c:/MINGW32/home/varun/esp-idf/esp-idf**

3:- Trying to install 
$ **python -m pip install -r $IDF_PATH/requirements.txt**
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'c:/MINGW32/home/varun/esp-idf/esp-idf/requirements.txt'

4:- Try to run make menuconfig 
varun@LAPTOP-RFBF50C5 MINGW32 ~/esp-idf/esp-idf/examples/hello_world_back
$ make menuconfig
Makefile:8: c:/MINGW32/home/varun/esp-idf/esp-idf/make/project.mk: No such file or direc                                    tory
make: *** No rule to make target 'c:/MINGW32/home/varun/esp-idf/esp-idf/make/project.mk'                                    .  Stop.

igrr, 2020-02-12:
Can you please try setting IDF_PATH to `$HOME/esp-idf/esp-idf`?

varunsavita, 2020-02-13:
I got same response.
varun@LAPTOP-RFBF50C5 MINGW32 ~/esp-idf/esp-idf/examples/hello_world_back
$ export IDF_PATH="$home/esp-idf/esp-idf"

varun@LAPTOP-RFBF50C5 MINGW32 ~/esp-idf/esp-idf/examples/hello_world_back
$ printenv IDF_PATH
/esp-idf/esp-idf

varun@LAPTOP-RFBF50C5 MINGW32 ~/esp-idf/esp-idf/examples/hello_world_back
$ cd esp-idf/esp-idf/examples/hello_world_back/
-bash: cd: esp-idf/esp-idf/examples/hello_world_back/: No such file or directory

dobairoland, 2020-02-13:
@varunsavita Please note the `$HOME` should be written with all capital letters. You can see that `IDF_PATH` is `/esp-idf/esp-idf` because `$home` is empty.

varunsavita, 2020-02-13:
Thanks,
It has worked
 I selected port then send a **make flash**

make[1]: *** [/home/varun/esp-idf/esp-idf/make/component_wrapper.mk:292: asio/asio/src/asio.o] Error 1
make: *** [/home/varun/esp-idf/esp-idf/make/project.mk:604: component-asio-build] Error 2

**Partition**
 MINGW32 ~/esp-idf/esp-idf/components/partition_table
$ cat partitions_singleapp.csv
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,      data, nvs,     ,        0x6000,
phy_init, data, phy,     ,        0x1000,
factory,  app,  factory, ,        1M,

Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: **Toolchain version is not supported**: crosstool-ng-1.22.0-80-g6c4433a5

I want to see output of hello_world
I use **make monitor**
 MINGW32 ~/esp-idf/esp-idf/examples/hello_world_back
$ make monitor
Makefile:8: /make/project.mk: No such file or directory
make: *** No rule to make target '/make/project.mk'.  Stop.

varunsavita, 2020-02-14:
just try to flash Hello_world program
first, do, make build then make flash and also open another window  for monitor - make monitor
 I got Error 
when i put - make flash 
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
/home/varun/esp-idf/esp-idf/make/component_wrapper.mk:103: /home/varun/esp-idf/esp-idf/components/bootloader/subproject/components/micro-ecc/component.mk: No such file or directory
make[2]: *** No rule to make target '/home/varun/esp-idf/esp-idf/components/bootloader/subproject/components/micro-ecc/component.mk'.  Stop.
make[1]: *** [/home/varun/esp-idf/esp-idf/make/project.mk:468: component-micro-ecc-build] Error 2
make: *** [/home/varun/esp-idf/esp-idf/components/bootloader/Makefile.projbuild:41: /home/varun/esp-idf/esp-idf/examples/hello_world_back/build/bootloader/bootloader.bin] Error 2

another window - make monitor  error
$ make monitor
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in t                                            he configuration menu
MONITOR
*** 'make monitor' target requires an app to be compiled and flashed first.
*** Run 'make flash monitor' to build, flash and monitor
*** Or alternatively 'make simple_monitor' to view the serial port as-is.
make: *** [/home/varun/esp-idf/esp-idf/components/esptool_py/Makefile.projbuild:                                            105: monitor] Error 1

varunsavita, 2020-02-17:
@dobairoland @igrr  Please guide me where is the issue. because the outcome is nill form me.

dobairoland, 2020-02-17:
Hi @varunsavita. I see that you've getting error after error.

There is ZIPed environment you can download and use (everything is working there): https://docs.espressif.com/projects/esp-idf/en/latest/get-started-legacy/windows-setup.html#toolchain-setup (I assume that you are using the master branch or v4.0 of ESP-IDF. There is another ZIP for older IDF versions).

Please note that the MSYS environment is deprecated and will not be supported in the future. If you are a new IDF user then I recommend to use https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html#esp-idf-tools-installer, the `install.bat` and `export.bat` scripts in Windows command line. 

Please go throught the getting started guide in order to set-up your environment: https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html

varunsavita, 2020-02-17:
Thank you, I'll try

varunsavita, 2020-02-24:
**I see that you've getting error after error.

There is ZIPed environment you can download and use (everything is working there): https://docs.espressif.com/projects/esp-idf/en/latest/get-started-legacy/windows-setup.html#toolchain-setup (I assume that you are using the master branch or v4.0 of ESP-IDF. There is another ZIP for older IDF versions).**

 @dobairoland  I downloaded 
open -- 
 git clone -b v4.0 --recursive https://github.com/espressif/esp-idf.git

varun@LAPTOP-RFBF50C5 MSYS ~/esp/esp-idf
$ ./install.sh
Installing ESP-IDF tools
/usr/bin/env: ‘python’: No such file or directory
Not to understand where is my fault.

igrr, 2020-02-24:
@varunsavita If you are using Windows, it is recommended to install everything using the IDF Tools Installer (make sure you use version v2.3 of it): https://dl.espressif.com/dl/esp-idf-tools-setup-2.3.exe

varunsavita, 2020-02-24:
@igrr  in window  I installed IDF Command Prompt and 
run the example code and saw output but I need an editor that's why I want to install Eclipse or VS code.

igrr, 2020-02-24:
@varunsavita Okay, can you please confirm you have the command prompt environment working? If yes, I will close this issue. If you have a specific issue configuring VS Code, could you please post it along with the details in https://github.com/espressif/vscode-esp-idf-extension?

varunsavita, 2020-02-24:
Yes, Thank you for supporting.

closed 2020-02-24 by varunsavita