espressif/esp-idf #4757: install.sh fails on virtualenv, argument --no-site-packages no recognized

Faebu93, 2020-02-13:
## Environment
- IDF version (run ``git describe --tags`` to find it): release/V4.0 (release branch)
- Build System:         N/A
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): N/A
- Operating System:     Linux
- Using an IDE?: VS Code, installing from the Espressif IDF extension (but also tried direct execution of install.sh with different versions of Python)
- Python version: Python 3.8.1
- Virtualenv: python -m virtualenv --version > virtualenv 20.0.1 from 
 home/user/.local/lib/python3.8/site-packages/virtualenv/__init__.py

## Problem Description
Installing ESP-IDF from the extension, the setup of the virtualenv fails, since --no-site-packages has long been deprecated and seems to be not accepted anymore in the newest versions (see script output below). It is anyway standard behaviour I understand. So install.sh fails more or less silently if run in the VS Code extension, when trying to compile it just gives some unrelated errors (of course, the tool chain is not properly setup). Manually running install.sh produces the same error.
Just removing --no-site-packages in line 1183 of install.sh seems to do the trick for me (not tested extensively yet).
Maybe completely remove it or first check if it is supported and accordingly run the command?
Many thanks and best!

### Expected Behavior
install.sh completes without failure

### Actual Behavior
install.sh fails to setup virtualenv, see output below

### Steps to reproduce

1. Probably only with newest virtualenv version?
2. run install.sh

## Debug Logs
```bash
./install.sh
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2019r2-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190313 (already installed)
Installing Python environment and packages
Creating a new Python environment in /home/user/.espressif/python_env/idf4.0_py3.8_env
usage: virtualenv [--version] [-v | -q] [--discovery {builtin}] [-p py] [--creator {builtin,cpython3-posix,venv}] [--seeder {app-data,pip}] [--no-seed] [--activators comma_separated_list] [--clear] [--system-site-packages]
                  [--symlinks | --copies] [--download | --no-download] [--extra-search-dir d [d ...]] [--pip version] [--setuptools version] [--wheel version] [--no-pip] [--no-setuptools] [--no-wheel] [--clear-app-data] [--prompt prompt]
                  [-h]
                  [dest]
virtualenv: error: unrecognized arguments: --no-site-packages
Traceback (most recent call last):
  File "/home/user/Documents/_esp-idf/tools/idf_tools.py", line 1372, in <module>
    main(sys.argv[1:])
  File "/home/user/Documents/_esp-idf/tools/idf_tools.py", line 1368, in main
    action_func(args)
  File "/home/user/Documents/_esp-idf/tools/idf_tools.py", line 1183, in action_install_python_env
    subprocess.check_call([sys.executable, '-m', 'virtualenv', '--no-site-packages', idf_python_env_path],
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python', '-m', 'virtualenv', '--no-site-packages', '/home/user/.espressif/python_env/idf4.0_py3.8_env']' returned non-zero exit status 2.

igrr, 2020-02-13:
Hi @Faebu93, this has been fixed in release/v4.0 branch by commit https://github.com/espressif/esp-idf/commit/8cb3cc515477e777144acb3b58f5156c4ad69124, could you please update the branch and try again?

Faebu93, 2020-02-13:
Hum, strange, I had chosen release/v4.0 branch, downloaded 1 hour ago, but will try again. Thanks!

Faebu93, 2020-02-13:
So I cloned the repository directly from Git, now it is working smoothly. Thanks!
Maybe the extension was still downloading an outdated zip file...
Cheers!

closed 2020-02-13 by Faebu93