espressif/esp-idf #4758: EC cyphers cannot be disabled

DiegoActionItem, 2020-02-13:
In v3.3.1, I was able to disable ECC. In v4.0.0, I get this error:

```
/home/diego/src/esp-idf-4.0/components/bootloader_support/src/idf/secure_boot_signatures.c: In function 'esp_secure_boot_verify_signature_block':
/home/diego/src/esp-idf-4.0/components/bootloader_support/src/idf/secure_boot_signatures.c:96:5: error: unknown type name 'mbedtls_ecdsa_context'; did you mean 'mbedtls_rsa_context'?
     mbedtls_ecdsa_context ecdsa_context;
     ^~~~~~~~~~~~~~~~~~~~~
     mbedtls_rsa_context
/home/diego/src/esp-idf-4.0/components/bootloader_support/src/idf/secure_boot_signatures.c:97:5: error: implicit declaration of function 'mbedtls_ecdsa_init'; did you mean 'mbedtls_rsa_init'? [-Werror=implicit-function-declaration]
     mbedtls_ecdsa_init(&ecdsa_context);
     ^~~~~~~~~~~~~~~~~~
```

I think a few `#ifdef CONFIG_MBEDTLS_ECP_C` can be added to that code.

negativekelvin, 2020-02-13:
https://github.com/espressif/esp-idf/commit/abd6d40796ea409765f972ad4fed4bc82db64d9a#diff-483c520ee5d302f1c694fbc5eaf0dcdd

projectgus, 2020-04-01:
Hi @DiegoActionItem ,

Sorry for the long delay in getting back to you and many thanks for being patient.

As @negativekelvin points out, this bug was fixed on the master branch for v4.1 but the fix wasn't backported to the release/v4.0 branch. I've just submitted a backport for it, when it merges then this issue will be automatically closed. The fix will be in the v4.0.1 bugfix release, or you can check out the release/v4.0 branch once it's available.

DiegoActionItem, 2020-04-01:
epic, nice work.

closed 2020-04-01 by DiegoActionItem