espressif/esp-idf #4759: make menuconfig

varunsavita, 2020-02-13:
I was  trying to flash from eclipse.
Where can I see an output. I used **make menuconfig**
$ make monitor
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-80-g6c4433a5
Expected to see version: esp-2019r2
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 5.2.0
Expected to see version(s): 8.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Python requirements from G:/IcapoTech_workspace/msys32/home/varun/esp-idf/esp-idf/requirements.txt are satisfied.
MONITOR
*** 'make monitor' target requires an app to be compiled and flashed first.
*** Run 'make flash monitor' to build, flash and monitor
*** Or alternatively 'make simple_monitor' to view the serial port as-is.
make: *** [/home/varun/esp-idf/esp-idf/components/esptool_py/Makefile.projbuild:144: monitor] Error 1

[image]

DiegoActionItem, 2020-02-13:
1) You need to compile the app
2) You are using the wrong compiler.

Follow documentation (you are probably missing export.sh).

varunsavita, 2020-02-13:
I missed something but don't know where is problem 

varun@LAPTOP-RFBF50C5 MINGW32 /
$ install.bat
-bash: install.bat: command not found

varun@LAPTOP-RFBF50C5 MINGW32 /
$ install.ps1
-bash: install.ps1: command not found

varun@LAPTOP-RFBF50C5 MINGW32 /
$ export IDF_TOOLS_PATH="$HOME/esp-idf/esp-idf"

varun@LAPTOP-RFBF50C5 MINGW32 /
$ install.ps1
-bash: install.ps1: command not found

DiegoActionItem, 2020-02-13:
‏You are on a mingw32 environment, what happens when you run: `./install.sh` 

Or instead from a `cmd` shell run the corresponding batch file?

varunsavita, 2020-02-13:
Thanks  for response
 I was trying to flash
19:40:13 **** Build of configuration Default for project HelloWorld ****
python /home/varun/esp-idf/esp-idf/tools/windows/eclipse_make.py flash 
G:\msys32\mingw32\bin\python.exe: can't open file '/home/varun/esp-idf/esp-idf/tools/windows/eclipse_make.py': [Errno 2] No such file or directory
"python /home/varun/esp-idf/esp-idf/tools/windows/eclipse_make.py flash" terminated with exit code 2. Build might be incomplete.

DiegoActionItem, 2020-02-13:
Now you are mixing environments, trying to run Windows's python and giving it "unix" (mingw) paths :)

Please post the whole command. and enclose it with \`\`\`, for example:

>\`\`\`
>some code
>\`\`\`

To get displayed as:
```
some code
```

varunsavita, 2020-02-13:
Ohh :(  
Actually i want to code in eclipse.
Not understand  what going on.
I was trying to run the code that's why i saw alot of things.
how to make it easy?

varunsavita, 2020-02-14:
Oh :(
Okay, now I will reinstall again.
I want to make it easy.

kolipakakondal, 2020-02-18:
Hello, If you're looking for eclipse support for CMake idf projects, please refer to https://github.com/espressif/idf-eclipse-plugin#espressif-idf-eclipse-plugins

This has inbuilt GUI based menuconfig, serial monitor and many more.

varunsavita, 2020-02-24:
@kolipakakondal 
No Toolchain found for Target ESP32

kolipakakondal, 2020-02-24:
@varunsavita Can you make sure you run this step https://github.com/espressif/idf-eclipse-plugin#installing-esp-idf-tools

This will take care of configuring the required paths in the CDT build environment along with installing the tools.

I hope that helps.

varunsavita, 2020-02-24:
[image]
Yes, I did

kolipakakondal, 2020-02-24:
@varunsavita  Did something went wrong during the "install tools" from eclipse? If everything goes well, it will set the CMake toolchain in the preferences.

Check if you're able to see toolchain in the preferences? https://github.com/espressif/idf-eclipse-plugin#configuring-toolchains

https://github.com/espressif/idf-eclipse-plugin#configuring-core-build-toolchains
https://github.com/espressif/idf-eclipse-plugin#configuring-cmake-toolchain

Below logs will help.
https://github.com/espressif/idf-eclipse-plugin#espressif-idf-tools-console
https://github.com/espressif/idf-eclipse-plugin#error-log

Alvin1Zhang, 2020-04-10:
@varunsavita Would you please share if any updates for the issue? Thanks.

Alvin1Zhang, 2020-05-26:
Close due to short of feedback/updates, feel free to reopen with more details. Thanks.

closed 2020-05-26 by Alvin1Zhang