espressif/esp-idf #4761: I2C transaction is ending with I2C_STATUS_TIMEOUT when running in repeated-start-mode

Nan000, 2020-02-13:
## Environment

- Development Kit: TTGO ESP32-WROVER-B
- Module or chip used: ESP32-WROVER-B
- IDF version: v4.0-rc
- Build System: CMake + idf.py
- Compiler version: xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Windows 10
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes, Eclipse
- Power Supply: USB

## Setup
TTGO ESP32-WROVER-B module connected to ICM-20600 motion sensor via I2C (10k pull-up resistors)

## Problem Description
I've tried to perform a single byte read on the "Who am I" register (0x75) of the ICM-20600 motion sensor which ended with an I2C time out error. My I2C sequence was as follows:

`Start Condition + Setup Write to ICM-20600 I2C ADDR + Register ADDR + Repeated Start Condition + Setup Read to ICM-20600 I2C ADDR + Single Byte Read + Stop Condition`

After checking the actual signals using a logic analyzer (see screenshot below) it can be seen that the communiation is initiated properly but isn't completed with an stop-condition at the end. Also, the clock timing seems unstable around the point in time the repeated start condition is generated.

Since from hardware perspective everything seems to be fine I've cross checked the setup by writing an equivalent Aduino code which didn't show the issue as shonw below. The standard Aduino I2C code doesn't use the repeated start condition which might be the general issue here.

If checked the ESP-IDF driver routine and found that it enter s the following I2C_STATUS_TIMEOUT if-branch:

[image]

### Expected Behavior
The I2C interface should read the register content in repeated-start-mode and return ESP_OK.

### Code to reproduce this issue
####  C code using ESP-IDF SDK
[image]

#### C++ code using Arduino
```cpp
#include <Wire.h>

#define SDA 21
#define SCL 22

#define I2C_ADDR         0x68
#define I2C_REG_TO_READ  0x75

TwoWire I2C = TwoWire(0);

int data;

void setup(){
  Serial.begin(115200);
  I2C.begin(SDA, SCL, 100000); // SDA pin 21, SCL pin 22 [TTGO]
}

void loop(){

  I2C.beginTransmission(I2C_ADDR);
  I2C.write(I2C_REG_TO_READ);
  I2C.endTransmission();

  I2C.requestFrom(I2C_ADDR, 1);

    if (I2C.available() == 1)
    {
      data = I2C.read();
    }

    delay(1000);
}
```

#### Signal trace produced by ESP-IDF SDK
It can be seen that after the byte has been send from the slave a short signal gab happens and after that the master generates some additional clock cycles with an different clock frequency (the software was setup to run SCL at 100 kHz)

[image]

#### Signal trace produced by Arduino
[image]

stickbreaker, 2020-02-14:
@Nan000 
The i2c sequence for your IDF version is malformed. the last byte of READ needs to have the master send a NAK to tell the Slave device to terminate the read operations so line 71 of your example should be:
```c++
ESP_ERROR_CHECK( i2c_master_read_byte( p_i2c_command, &data_8u, NAK);
```

I think your understanding of ACK and NAK is backwards,  ACK is logic 0, NAK is logic 1.

The weird waveform is because the slave device is still in READ mode, it is driving SDA for the next byte of data.  After the 15ms timeout the SCL pulses are a bus recovery operation triggered by the timeout. 

Chuck.

stickbreaker, 2020-02-14:
@Nan000 Also, it you want to duplicate the ReSTART operation in Arduino Code use:
```c++
void loop(){

  I2C.beginTransmission(I2C_ADDR);
  I2C.write(I2C_REG_TO_READ);
  uint8_t err = I2C.endTransmission(FALSE); // no STOP, next cmd will be ReSTART

  if(err != 0 ) 
    Serial.printf("Error selecting register error=%d(%s)\n", err, I2C.getErrorText(err));

  uint8_t count = I2C.requestFrom(I2C_ADDR, 1);

  if(count != 1) { // returned count will match requested count unless error.
    Serial.printf("Error read %d bytes, error= %d(%s)\n", I2C.lastError(), I2C.getErrorText(I2C.lastError()));
  }

  if (I2C.available() == 1)
  {
      data = I2C.read();
  }

  delay(1000);
}
```

Chuck.

koobest, 2020-02-19:
Hi,@Nan000
Has your problem been solved？ESP32 support  repeated-start mode. The code using IDF is implemented like this：
```
static esp_err_t i2c_master_restart_read(i2c_port_t i2c_num, uint8_t *data_wr, size_t size)
{
    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, (ESP_SLAVE_ADDR << 1) | WRITE_BIT, ACK_CHECK_EN);
    i2c_master_write_byte(cmd, 0x22, ACK_CHECK_EN);
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, (ESP_SLAVE_ADDR << 1) | READ_BIT, ACK_CHECK_EN);
    i2c_master_read(cmd, data_wr, size, I2C_MASTER_LAST_NACK);
    i2c_master_stop(cmd);
    esp_err_t ret = i2c_master_cmd_begin(i2c_num, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);
    return ret;
}
```

thanks !!

Nan000, 2020-02-19:
Hi all,

thanks a lot for all the helpful comments. I didn't realize that function `i2c_master_read_byte(i2c_cmd_handle_t cmd_handle, uint8_t* data, i2c_ack_type_t ack)` was expecting an enum of type `i2c_ack_type_t` and not an `bool` like the other required function calls before. **The problem is solved now.**

_Code Fix_
[image]

_I2C Trace after code fix_
[image]

closed 2020-02-19 by Nan000