espressif/esp-idf #4762: BLE Nimble GATT write requests from a client are getting cut off at ~267 bytes

MirChamb3r, 2020-02-13:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32
- IDF version (run ``git describe --tags`` to find it):
    v4.0-beta1
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    crosstool-ng-1.22.0-80-g6c4433a
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

I'm having an issue when trying to do a write to a GATT server I have set up with Nimble. I have the MTU set to 512 and can do a read with 512 bytes from my client no problem. However, when I try to write anything larger than 267 bytes Nimble only shows getting the first ~267 bytes (sometimes less) and nothing else.

### Expected Behavior
I would expect to be able to write MTU bytes over BLE, same as reading.

### Actual Behavior
Anything over 267 bytes will get cut off when written.

### Steps to reproduce

1. Create a GATT server using Nimble with a characteristic defined
2. Set the MTU to 512
3. Try to write 267+ bytes to the Characteristic (using NRF connect or some other BLE app)
4. Printing out the received write request shows only the first 267 bytes

### Code

If you need code, look at Nimble's bleprph app example. That's what I followed to create my server.

prasad-alatkar, 2020-02-14:
Hi @MirChamb3r , I have tested this with `bleprph` app and I am able to write data of length more than 267 bytes to characteristic value. Can you please provide console logs with debug prints (`make menuconfig --> Component config --> Log output --> Default log verbosity `) ? You can also print the received data & length in `ble_att_svr_rx_write` (`$IDF_PATH/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c`) with adding below lines:
```
    BLE_HS_LOG(DEBUG, "\n Received Write req len = %d\n", OS_MBUF_PKTLEN((*rxom)));
    BLE_HS_LOG(DEBUG, "Write req data:\n");
    ble_hs_log_flat_buf((*rxom)->om_data, OS_MBUF_PKTLEN((*rxom)));
    BLE_HS_LOG(DEBUG, "\n");
```
Please let me know if you still face the issue.

MirChamb3r, 2020-02-14:
Thanks for the reply @prasad-alatkar . I tried what you suggested and these are the results:
Sent the following string from NRF connect app (297 bytes):
`Three of us are going to be in the office tomorrow and will be able to make it to the meeting tonight but I will be able to make it to the meeting tonight but I will be able to make it to the meeting tonight yes or no to the meeting tonight but I will be able to make it to the meeting tonight end`

NimBLE printed out the following:
```
Received Write req len = 299
Write req data:
0x10 0x00 0x54 0x68 0x72 0x65 0x65 0x20 0x6f 0x66 0x20 0x75 0x73 0x20 0x61 0x72 0x65 0x20 0x67 0x6f 0x69 0x6e 0x67 0x20 0x74 0x6f 0x20 0x62 0x65 0x20 0x69 0x6e 0x20 0x74 0x68 0x65 0x20 0x6f 0x66 0x66 0x69 0x63 0x65 0x20 0x74 0x6f 0x6d 0x6f 0x72 0x72 0x6f 0x77 0x20 0x61 0x6e 0x64 0x20 0x77 0x69 0x6c 0x6c 0x20 0x62 0x65 0x20 0x61 0x62 0x6c 0x65 0x20 0x74 0x6f 0x20 0x6d 0x61 0x6b 0x65 0x20 0x69 0x74 0x20 0x74 0x6f 0x20 0x74 0x68 0x65 0x20 0x6d 0x65 0x65 0x74 0x69 0x6e 0x67 0x20 0x74 0x6f 0x6e 0x69 0x67 0x68 0x74 0x20 0x62 0x75 0x74 0x20 0x49 0x20 0x77 0x69 0x6c 0x6c 0x20 0x62 0x65 0x20 0x61 0x62 0x6c 0x65 0x20 0x74 0x6f 0x20 0x6d 0x61 0x6b 0x65 0x20 0x69 0x74 0x20 0x74 0x6f 0x20 0x74 0x68 0x65 0x20 0x6d 0x65 0x65 0x74 0x69 0x6e 0x67 0x20 0x74 0x6f 0x6e 0x69 0x67 0x68 0x74 0x20 0x62 0x75 0x74 0x20 0x49 0x20 0x77 0x69 0x6c 0x6c 0x20 0x62 0x65 0x20 0x61 0x62 0x6c 0x65 0x20 0x74 0x6f 0x20 0x6d 0x61 0x6b 0x65 0x20 0x69 0x74 0x20 0x74 0x6f 0x20 0x74 0x68 0x65 0x20 0x6d 0x65 0x65 0x74 0x69 0x6e 0x67 0x20 0x74 0x6f 0x6e 0x69 0x67 0x68 0x74 0x20 0x79 0x65 0x73 0x20 0x6f 0x72 0x20 0x6e 0x6f 0x20 0x74 0x6f 0x20 0x74 0x68 0x65 0x20 0x6d 0x65 0x65 0x74 0x69 0x6e 0x67 0x20 0x74 0x6f 0x6e 0x69 0x67 0x68 0x74 0x20 0x62 0x75 0x74 0x20 0x49 0x20 0x77 0x69 0x6c 0x6c 0x20 0x62 0x65 0x20 0x61 0x62 0x6c 0x65 0x20 0x74 0x6f 0x20 0x6d 0x61 0x64 0x6f 0xfc 0x3f 0x00 0x08 0x35 0x00 0x14 0x7a 0xfc 0x3f 0x00 0x00 0x00 0x00 0x35 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x10 0x35 0x00 0x20 0x49 0x20 0x77
```
Which if you translate that back into a string, it ends up being:
`Three of us are going to be in the office tomorrow and will be able to make it to the meeting tonight but I will be able to make it to the meeting tonight but I will be able to make it to the meeting tonight yes or no to the meeting tonight but I will be able to madoü? 5 zü?    5        5  I w.`

So a tad off. If you count the bytes up to the last semi correct word 'mad' its 266 bytes total.
Also I printed out the om_len as well and it gets set to 267 every time even though the data received was 297 bytes. Forcing the om_len to the same length (297) gives back the same garble as above.

dhrishi, 2020-02-18:
@MirChamb3r `Set the MTU to 512` Is 512 bytes the negotiated MTU? Did you set it explicitly at both the central (nRF connect) and peripheral side (ESP32 with NimBLE)?

prasad-alatkar, 2020-02-18:
Hi @MirChamb3r apologies for late reply, having analyzed your response, below are few points/reasons that come to my mind for failure if any:
- As you are referring to `bleprph` example, please make sure that the length of destination buffer (`gatt_svr_sec_test_static_val`) where characteristic value is getting written is sufficient (> 298 for this case).
-  The `om_len` that you mention, is this `om->om_len` or `om_len = OS_MBUF_PKTLEN(om)` ? The total data gets carried in chain of mbufs, so `om->om_len` will give you length of data in that particular `mbuf`, however to get total length of data `OS_MBUF_PKTLEN(om)` should be used. 
- In your sdkconfig, modify `CONFIG_BT_NIMBLE_ACL_BUF_SIZE` to larger value, again this should not be an issue but it will help you increase size of data in each `mbuf`.
- As @dhrishi has already mentioned about ` Negotiated MTU `, please ensure you are getting mtu update event like `mtu update event; conn_handle=0 cid=4 mtu=512`. However mostly it should not be an issue IMHO.

- Regarding this: 

> Which if you translate that back into a string, it ends up being:

As you are using example based on `bleprph`, in `gatt_svr_chr_write` of `gatt_svr.c`, please add this line `ESP_LOGI("Data :", " In Text = %s ", (char *)dst);` & `ble_hs_log_flat_buf(dst, om_len);` to get the actual data, in my last reply I forgot on add that we should be using `ble_hs_mbuf_to_flat` API to convert data from mbufs to flat buffers. 
If you still find issue with above suggestions, please share diff of your code & sdkconfig.

MirChamb3r, 2020-02-18:
@dhrishi thanks for the response, yes I negotiate an mtu of 512 and get a log message to confirm it.
@prasad-alatkar thanks again for the response. So yup that has definitely fixed it! I didn't know there was another length I had to check against the om_len. I was only using the om_data and om_len in order to get the value from the client. This is how I currently fixed it:
`
uint16_t length = om->om_len;
if (length != OS_MBUF_PKTLEN(om)) {
        // Flatten the buffer to get the full data
        uint16_t fullLength = OS_MBUF_PKTLEN(om);
        uint8_t fullData[fullLength];
        if (ble_hs_mbuf_to_flat(om, &fullData, fullLength, nullptr) != 0) {
            ESP_LOGE(TAG, "Failed to get all write data");
            return;
        }
            
        writeValue(fullData, fullLength);
    } else {
        writeValue(om->om_data, length);
    }
}
`

I've tested this with 512-514, 1024-1026 bytes and it works perfectly so thanks for the help!

I will say that this is a little confusing on how it works though. I would expect to receive the full data in what's given to me in the callback and not have to check two different lengths. Couldn't the ble_att_svr_write() function handle this before sending it to the callback?

prasad-alatkar, 2020-02-19:
Hi @MirChamb3r , I would recommend you to go through `gatt_svr_chr_write` from `gatt_svr.c` in `bleprph` app. Basically you don't need to check two different lengths, just `om_len = OS_MBUF_PKTLEN(om);` should suffice. 
If you find this helpful, please help with closing the issue.

closed 2020-02-24 by MirChamb3r