espressif/esp-idf #4763: Undefined symbols 'os_msys_buf_free' and 'os_msys_buf_alloc'

agendek78, 2020-02-13:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): []
- Module or chip used:  [ESP32-PICO-D4]
- IDF version: v4.1-dev-2263-g605da33c3
- Build System:         [idf.py]
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes (eclipse)]
- Power Supply:      [external 3.3V]

## Problem Description

When trying to build 'examples\bluetooth\nimble\bleprph' example there is linking error.
c:/users/andy/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/bt/libbt.a(esp_nimble_hci.c.obj):(.literal.ble_buf_free+0x0): undefined reference to `os_msys_buf_free'
c:/users/andy/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/bt/libbt.a(esp_nimble_hci.c.obj):(.literal.ble_buf_alloc+0x8): undefined reference to `os_msys_buf_alloc'
c:/users/andy/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/bt/libbt.a(esp_nimble_hci.c.obj): in function `ble_buf_free':
D:/Projects/esp32/esp-idf/components/bt/host/nimble/esp-hci/src/esp_nimble_hci.c:317: undefined reference to `os_msys_buf_free'
c:/users/andy/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/Projects/esp32/esp-idf/components/bt/host/nimble/esp-hci/src/esp_nimble_hci.c:386: undefined reference to `os_msys_buf_alloc'
collect2.exe: error: ld returned 1 exit status

### Expected Behavior
Binary linked and ready to flash.

### Actual Behavior
Linking error.

### Steps to reproduce
Build example 'examples\bluetooth\nimble\bleprph'

dhrishi, 2020-02-14:
@agendek78 With commit `91c8f32`, the NimBLE submodule was also updated to include `os_msys_buf_alloc` and `os_msys_buf_free` APIs.

Looks like, you have the latest IDF but not the latest NimBLE submodule. Please update the same and give it a try.
```
$ cd /path/to/esp-idf
$ git submodule update --init --recursive
```

agendek78, 2020-02-14:
Sorry my fault! Thanks a lot.

closed 2020-02-14 by agendek78