espressif/esp-idf #4764: enc29j60 example failing

okasha55, 2020-02-14:
this is the serial output when i run the ethernet enc28j60 example
i haven't changed anything in the code

I (0) cpu_start: Starting scheduler on APP CPU.
I (386) enc28j60: revision: 0
E (386) enc28j60: enc28j60_verify_id(478): wrong chip ID
E (386) enc28j60: emac_enc28j60_init(852): vefiry chip ID failed
I (396) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (406) esp_eth: esp_eth_driver_install(194): init mac failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40085e0c
0x40085e0c: _esp_error_check_failed at /home/okasha/esp/esp-idf/components/esp32/panic.c:739

file: "../main/enc28j60_example_main.c" line 117
func: app_main
expression: esp_eth_driver_install(&eth_config, &eth_handle)

ELF file SHA256: 8ca8cda62156293abc0712223bd9b59fd78f39b314e4c8e12ae089feeb2dca9b

Backtrace: 0x4008585d:0x3ffb5790 0x40085e0f:0x3ffb57b0 0x400d5426:0x3ffb57d0 0x400d233e:0x3ffb58a0 0x4008893d:0x3ffb58c0
0x4008585d: invoke_abort at /home/okasha/esp/esp-idf/components/esp32/panic.c:157

0x40085e0f: _esp_error_check_failed at /home/okasha/esp/esp-idf/components/esp32/panic.c:740

0x400d5426: app_main at /home/okasha/esp/enc28j60/build/../main/enc28j60_example_main.c:117 (discriminator 1)

0x400d233e: main_task at /home/okasha/esp/esp-idf/components/esp32/cpu_start.c:563

0x4008893d: vPortTaskWrapper at /home/okasha/esp/esp-idf/components/freertos/xtensa/port.c:143

Rebooting...
ets Jun  8 2016 00:22:57

suda-morris, 2020-02-14:
please check your wire connection

nx518, 2020-02-14:
I got wrong Chip ID, when my frequency was too high in combination with jumper cables. Try it with 5MHz.

okasha55, 2020-02-15:
thanks alot.
wow didn't notice that jumper lengths would matter 
i used shorter jumper wires instead of the long flat cable i used and it worked
i actually checked the connection using a multimeter so the only factor is the cable length

closed 2020-02-15 by okasha55

prajapak, 2021-04-04:
I have the same issue. What length is longer or shorter? I am using about 6inches jumper available in market. Please advise.

okasha55, 2021-04-05:
2-3 inches jumpers was fine with me and the PCB connection was flawless.

> I have the same issue. What length is longer or shorter? I am using about 6inches jumper available in market. Please advise.

prajapak, 2021-04-08:
@okasha55 thanks for reply. I used reduced 20cm jumper to 10cm but I still see same issue. I used 5MHz. These are the pins I used and confirmed same GPIOs are configured via menuconfig.

GPIO | ENC28J60
-- | --
GPIO19 | SPI_CLK
GPIO23 | SPI_MOSI
GPIO25 | SPI_MISO
GPIO22 | SPI_CS
GPIO4 | Interrupt

prajapak, 2021-04-08:
Got it working. I had to change CLK to SCK pin. This board has lot of mislabels.