espressif/esp-idf #4765: wifi manager custom endpoint handler function

humansthatmake, 2020-02-14:
Hi,

I'm trying to add a custom endpoint on the provisioning manager example, so I followed the documentation on the [programming guide](https://docs.espressif.com/projects/esp-idf/en/stable/api-reference/provisioning/wifi_provisioning.html#additional-endpoints) and I ended up with this:

```void app_main()
{
    /* Initialize NVS partition */
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        /* NVS partition was truncated
         * and needs to be erased */
        ESP_ERROR_CHECK(nvs_flash_erase());

        /* Retry nvs_flash_init */
        ESP_ERROR_CHECK(nvs_flash_init());
    }

    /* Initialize TCP/IP and the event loop */
    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));
    wifi_event_group = xEventGroupCreate();

    /* Initialize Wi-Fi */
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    /* Configuration for the provisioning manager */
    wifi_prov_mgr_config_t config = {
        /* What is the Provisioning Scheme that we want ?
         * wifi_prov_scheme_softap or wifi_prov_scheme_ble */
        .scheme = wifi_prov_scheme_ble,

        /* Any default scheme specific event handler that you would
         * like to choose. Since our example application requires
         * neither BT nor BLE, we can choose to release the associated
         * memory once provisioning is complete, or not needed
         * (in case when device is already provisioned). Choosing
         * appropriate scheme specific event handler allows the manager
         * to take care of this automatically. This can be set to
         * WIFI_PROV_EVENT_HANDLER_NONE when using wifi_prov_scheme_softap*/
        .scheme_event_handler = WIFI_PROV_SCHEME_BLE_EVENT_HANDLER_FREE_BTDM,

        /* Do we want an application specific handler be executed on
         * various provisioning related events */
        .app_event_handler = {
            .event_cb = prov_event_handler,
            .user_data = NULL
        }
    };

    /* Initialize provisioning manager with the
     * configuration parameters set above */
    ESP_ERROR_CHECK(wifi_prov_mgr_init(config));

    //CUSTOM ENDPOINT CREATED BY HTM
    wifi_prov_mgr_endpoint_create("custom-endpoint");

    bool provisioned = false;
    /* Let's find out if the device is provisioned */
    ESP_ERROR_CHECK(wifi_prov_mgr_is_provisioned(&provisioned));

    /* If device is not yet provisioned start provisioning service */
    if (!provisioned) {
        ESP_LOGI(TAG, "Starting provisioning");

        /* What is the Device Service Name that we want
         * This translates to :
         *     - Wi-Fi SSID when scheme is wifi_prov_scheme_softap
         *     - device name when scheme is wifi_prov_scheme_ble
         */
        char service_name[12];
        get_device_service_name(service_name, sizeof(service_name));

        /* What is the security level that we want (0 or 1):
         *      - WIFI_PROV_SECURITY_0 is simply plain text communication.
         *      - WIFI_PROV_SECURITY_1 is secure communication which consists of secure handshake
         *          using X25519 key exchange and proof of possession (pop) and AES-CTR
         *          for encryption/decryption of messages.
         */
        wifi_prov_security_t security = WIFI_PROV_SECURITY_1;

        /* Do we want a proof-of-possession (ignored if Security 0 is selected):
         *      - this should be a string with length > 0
         *      - NULL if not used
         */
        const char *pop = "abcd1234";

        /* What is the service key (could be NULL)
         * This translates to :
         *     - Wi-Fi password when scheme is wifi_prov_scheme_softap
         *     - simply ignored when scheme is wifi_prov_scheme_ble
         */
        const char *service_key = NULL;

        /* This step is only useful when scheme is wifi_prov_scheme_ble. This will
         * set a custom 128 bit UUID which will be included in the BLE advertisement
         * and will correspond to the primary GATT service that provides provisioning
         * endpoints as GATT characteristics. Each GATT characteristic will be
         * formed using the primary service UUID as base, with different auto assigned
         * 12th and 13th bytes (assume counting starts from 0th byte). The client side
         * applications must identify the endpoints by reading the User Characteristic
         * Description descriptor (0x2901) for each characteristic, which contains the
         * endpoint name of the characteristic */
        uint8_t custom_service_uuid[] = {
            /* LSB <---------------------------------------
             * ---------------------------------------> MSB */
            0x21, 0x43, 0x65, 0x87, 0x09, 0xba, 0xdc, 0xfe,
            0xef, 0xcd, 0xab, 0x90, 0x78, 0x56, 0x34, 0x12
        };
        wifi_prov_scheme_ble_set_service_uuid(custom_service_uuid);

        /* Start provisioning service */
        ESP_ERROR_CHECK(wifi_prov_mgr_start_provisioning(security, pop, service_name, service_key));

        /* Uncomment the following to wait for the provisioning to finish and then release
         * the resources of the manager. Since in this case de-initialization is triggered
         * by the configured prov_event_handler(), we don't need to call the following */
        // wifi_prov_mgr_wait();
        // wifi_prov_mgr_deinit();

        //CUSTOM ENDPOINT REGISTERED BY HTM
        wifi_prov_mgr_endpoint_register("custom-endpoint", custom_ep_handler, custom_ep_data);

    } else {
        ESP_LOGI(TAG, "Already provisioned, starting Wi-Fi STA");

        /* We don't need the manager as device is already provisioned,
         * so let's release it's resources */
        wifi_prov_mgr_deinit();

        /* Start Wi-Fi station */
        wifi_init_sta();
    }

    /* Wait for Wi-Fi connection */
    xEventGroupWaitBits(wifi_event_group, WIFI_CONNECTED_EVENT, false, true, portMAX_DELAY);

    /* Start main application now */
    while (1) {
        ESP_LOGI(TAG, "Hello World!");
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}```

This way we can call a custom-endpoint. But now we need to make an event handler function, so I initialized the function like this.

```static void custom_ep_handler( void *event, void *custom_ep_data)
{}```

Unfortunately, now I don't know how to add the content of the function as I don't know which events should I be expecting. 

I recently saw this [tutorial](https://medium.com/the-esp-journal/esp32-device-provisioning-configuring-custom-data-2e9c17aa4d51) from Kevar Sovani but it's not clear as well for me on how to write the proper handler function.

Is there a stub code for this?

shahpiyushv, 2020-02-14:
@humansthatmake , the query isn't clear. If you are asking for how to write your own custom handler, here is a sample for that:

```
esp_err_t custom_prov_data_handler(uint32_t session_id, const uint8_t *inbuf, ssize_t inlen,
                                          uint8_t **outbuf, ssize_t *outlen, void *priv_data)
{
    if (inbuf) {
        ESP_LOGI(TAG, "Received data: %.*s", inlen, (char *)inbuf);
    }
    char response[] = "SUCCESS";
    *outbuf = (uint8_t *)strdup(response);
    if (*outbuf == NULL) {
        ESP_LOGE(TAG, "System out of memory");
        return ESP_ERR_NO_MEM;
    }
    *outlen = strlen(response) + 1; /* +1 for NULL terminating byte */

    return ESP_OK;
}
```

If you want to handle the Wi-Fi provisioning manager events, there is a sample: https://github.com/espressif/esp-idf/blob/release/v3.3/examples/provisioning/manager/main/app_main.c#L60

humansthatmake, 2020-02-15:
Thank you for the enlightenment @shahpiyushv .

I'm actually using that provisioning manager example and I also read the documentation on "Additional Endpoints", so The function handler you posted is one of the things I was looking for because I saw no reference to it.
So, my question now is, how do you make use of the `custom_ep_data` parameter from `wifi_prov_mgr_endpoint_register` on this function handler? 

`wifi_prov_mgr_endpoint_register("custom-endpoint", custom_ep_handler, **custom_ep_data**);`

Also how does protocomm come into play here?

"See [protocomm](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/provisioning/protocomm.html) for the function signature of an endpoint handler"

shahpiyushv, 2020-05-26:
@humansthatmake , sorry I missed out on this. custom_ep_data is actually any private data that you want to access in your handler. You get it as the priv_data argument in your callback function. This is optional.

Protocomm comes into play because it abstracts out the actual communication protocol (BLE/HTTP+SoftAP). So, irrespective of what transport you use for the provisioning, the same callback will get invoked.

Meanwhile, we have added the custom endpoint usage in the standard [esp-idf provisioning example](https://github.com/espressif/esp-idf/blob/master/examples/provisioning/wifi_prov_mgr/main/app_main.c) itself.

jlacarin, 2020-06-26:
Hi @shahpiyushv,

I've been trying to setup a custom endpoint on the Provisionning manager as well, but it's been difficult to say the least.

As far as I understand it, `wifi_prov_mgr_endpoint_create` and `wifi_prov_mgr_endpoint_register` are supposed to add a new BLE characteristic with a UUID following a similar pattern to the pre-defined ones (like `prov-scan`, `prov-session`, etc.), but it doesn't seem to have added to the list of characteristics when I check in an app like Lightblue.

I've tried to run the latest version of the esp-idf Provisionning example you linked in your last message, and I still can't get this characteristic to appear.

Both `wifi_prov_mgr_endpoint_create` and `wifi_prov_mgr_endpoint_register` seem to complete without failing and `prov_ctx->endpoint_uuid_used` seems to have the expected value (`FF54`). I've tried sending data to the custom endpoint using both the iOS and Android versions of the Provisionning libraries, just in case I misunderstood the way it's supposed to work, without any luck as well.

So I'm a little bit confused about what I've been doing wrong or if I misunderstood how it works.

shahpiyushv, 2020-06-26:
@jlacarin flashing the wifi_prov_mgr example I had mentioned does indeed show me the custom-data characteristics in the nRF connect app (Screenshot attached)
[image].

I will ask our phone app developers to provide information on how to post data to this. However, have you checked the new [Android](https://github.com/espressif/esp-idf-provisioning-android) and [iOS](https://github.com/espressif/esp-idf-provisioning-ios) provisioning libraries? This sendData method for [ESPDevice](https://espressif.github.io/esp-idf-provisioning-ios/Classes/ESPDevice.html) would probably be what you are looking for.

jlacarin, 2020-06-26:
It's good to see it working on your screenshot, it reassures me after a day of not getting it to work!

Which version of the esp-idf have you been running to get this result? I've tried with 3.3, 4.0, 4.1, 4.2 and the most recent `master` with no luck :( Maybe I have a configuration problem somewhere...

I've been working with the Android and iOS Provisionning libraries and have tried the sendData method already, I've been able to communicate with the other characteristics using it, just not the missing `custom-data`.

shahpiyushv, 2020-06-26:
@jlacarin , I am using the master branch. The example I mentioned (wifi_prov_mgr with custom data) was added in idf 4.2. I do not have any special configuration. Just using the defaults.

jlacarin, 2020-06-26:
Ok, I finally got it to work, thank you for your help!

At the end I'm still quite confused, even after getting a fresh copy of the esp-idf it still wouldn't work on the two ESP32 boards that I was trying.
The solution was to use a more recent (and fresher) chip that I have in another prototype.

Anyway, thanks for your help, I was able to use the sendData method in the iOS Provisionning tool to exchange some data back and forth between the chip and the phone, so now I can continue on my work :)

dhrishi, 2020-06-26:
@jlacarin If you were using the two ESP32 boards with the eariler code (without the new custom endpoint), then there is a possibility that nRF app cached the data. After BLE connection is done through the nRF app, click on the ellipsis symbol (⋮)  and then `Refresh services` to force it to refresh. Please see if this helps

jlacarin, 2020-06-26:
I'm not quite sure why this is happening, but one after the other, these two have both stopped letting me connect to their BLE in either the nRF app, Lightblue, or even the esp Provisionning library. Even completely wiping the board with `idf.py erase_flash`.

shahpiyushv, 2020-06-30:
@jlacarin , can you file a new issue for this specific problem you are facing? It would be easier to track and also be helpful if other people come searching for solutions for a similar problem.

Alvin1Zhang, 2020-07-08:
@humansthatmake @jlacarin Thanks for reporting and updates, will close now, for the new issue, please help file a new ticket, thanks.

closed 2020-07-08 by Alvin1Zhang