espressif/esp-idf #4766: esp_wifi_connect returns ESP_ERR_WIFI_SSID

isabelleletarte, 2020-02-14:
## Environment

- Development Kit:   |none]
- Module or chip used:  ESP32-WROOM-32
- IDF version v3.3-beta3-34-g8c57aa024
- Build System:         Make
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     macOS
- Using an IDE?:PyCharm
- Power Supply:  USB and Battery

## Problem Description
We have 400 units with ESP32 that try to connect to an AP. Half of them currently get stucks when first trying to connect to the AP. The error is that esp_wifi_connect returns ESP_ERR_WIFI_SSID and the original firmware is not checking for errors when calling this function. The firmare get stucks waiting for SYSTEM_EVENT_STA_DISCONNECTED or SYSTEM_EVENT_STA_START indefinitely. 

The units are closed so it would take us a lot of time to open them and flash a new firmware with an error check when calling the esp_wifi_connect. 

All the units were working fine when they were at the client location but now that they are at our offices (with the same network configuration) half of them are having this problem.

We tried going into nature were only our AP was available but we still get the same issue.

We know that flashing the devices we a new firmware that does an error check would fix this but we want to try to find a solution where we don't have to open all of them. Could there be an issue with network setting that could make the esp_wifi_connect return this error?

Our wifi code is pretty much the same as the station_example_main.c except that we wait for one of the bits to be set.
`xEventGroupWaitBits(_wifi_event_group, WIFI_GOT_IP_BIT | WIFI_CONNECT_FAIL_BIT, false, false, portMAX_DELAY);`

## Other items if possible
sdkconfig
`
#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#
CONFIG_IDF_TARGET="esp32"

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=y

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
CONFIG_APP_EXCLUDE_PROJECT_VER_VAR=
CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR=

#
# Bootloader config
#
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG=
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=
CONFIG_LOG_BOOTLOADER_LEVEL=3
CONFIG_BOOTLOADER_SPI_WP_PIN=7
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
CONFIG_BOOTLOADER_FACTORY_RESET=
CONFIG_BOOTLOADER_APP_TEST=
CONFIG_BOOTLOADER_WDT_ENABLE=y
CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE=
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
CONFIG_APP_ROLLBACK_ENABLE=

#
# Security features
#
[... 641 lines cut ...]
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_TCPIP_LWIP=y

#
# Unity unit testing library
#
CONFIG_UNITY_ENABLE_FLOAT=y
CONFIG_UNITY_ENABLE_DOUBLE=y
CONFIG_UNITY_ENABLE_COLOR=
CONFIG_UNITY_ENABLE_IDF_TEST_RUNNER=y
CONFIG_UNITY_ENABLE_FIXTURE=

#
# Virtual file system
#
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096
`

sagb2015, 2020-02-17:
Questions - 
1) Is the SSID same for both client and office location? 
2) Is the SSID same for success and failure cases when in office since you mention half the devices are connecting and other half or not.
3) How many APs with same SSID are in the office?
4) Can you provide the log for at-least one device ? (Since you know the reason for the failure, this may also be possible). 
5) Can application trigger OTA based on remotely issued command assuming WiFi connection is taken care of?

espxiehang, 2020-02-18:
Hi @isabelleletarte, can you give me the IDF version and commit ID you use?

espxiehang, 2020-02-18:
Hi @isabelleletarte ，please update your ESP-IDF version, I think this commit can solve this problem, please try.

  commit 7200b8bf778513d7a709be56b859823bb29958a3
  Author: zhangyanjiao <zhangyanjiao@espressif.com>
  Date:   Fri May 24 11:31:24 2019 +0800
 
     wifi bugfixs:
 
      1. fix the bug no disconnect event when STA recv disassoc after sending auth
      2. fix the bug full scan send multi events when recv deauth/disassoc

isabelleletarte, 2020-02-19:
@espxiehang  Thank you for your answer
Yes we will update the esp-idf version for the next firmware update but if we can't connect to wifi we can't do the firmware update without opening our clients products and we want to try to solve the issue without opening a lot of products.

isabelleletarte, 2020-02-19:
@sagb2015 

Is the SSID same for both client and office location? **Yes**
Is the SSID same for success and failure cases when in office since you mention half the devices are connecting and other half or not. **Yes**
How many APs with same SSID are in the office? **Only one**
Can you provide the log for at-least one device ? (Since you know the reason for the failure, this may also be possible). **Will include the logs in the next comment**
Can application trigger OTA based on remotely issued command assuming WiFi connection is taken care of? **Yes once we are connected to WiFi we can do a firmware update OTA.**

isabelleletarte, 2020-02-19:
Okay, there has been some development with this issue.

I've been able to open one of the product and I added some logs in the firmware to see what was happening.

It seems that for the devices that aren't connecting to the wifi when we call the esp_wifi_set_config function the configuration isn't written in the NVS.
The wifi config is pretty straightforward:
```
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
wifi_config_t wifi_config = {};
memcpy(wifi_config.sta.ssid, CONFIG_ESP_WIFI_SSID, sizeof(CONFIG_ESP_WIFI_SSID));
memcpy(wifi_config.sta.password, CONFIG_ESP_WIFI_PASSWORD, sizeof(CONFIG_ESP_WIFI_PASSWORD));
```
No error is trigger when calling any of the wifi function:
- esp_wifi_init(&cfg)

- esp_wifi_set_mode(WIFI_MODE_STA)

- esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config)

- esp_wifi_start()

- esp_wifi_set_ps(WIFI_PS_NONE)

**For the logs what I did is that :**

1. I print the wifi config element from the wifi_config_t that is sent in the esp_wifi_set_config (_current wifi sta_)
2. I get the wifi config in the nvs with esp_wifi_get_config(&config)and I print the results with _set sta_.

Also no error when initializing NVS at the beginning of the main.

_Expected behavior is:_
```
I (2521) mf_safe_boot: current wifi sta ssid : MF_Staff
I (2527) mf_safe_boot: current wifi sta password : mfprodesp32-11
I (2534) mf_safe_boot: current wifi sta bssid :
I (2539) mf_safe_boot: current wifi sta bssid set: 0
I (2545) mf_safe_boot: current wifi sta channel : 0
I (2550) mf_safe_boot: current wifi sta listen interval : 0
I (2556) mf_safe_boot: current wifi sta scan method : 0
I (2562) mf_safe_boot: current wifi sta sort method : 0
I (2568) mf_safe_boot: current wifi sta threshold : 0
I (2574) mf_safe_boot: current wifi sta threshold : 0
D (2627) nvs: nvs_set_blob sta.ssid 36
D (2632) nvs: nvs_set_blob sta.pswd 65
D (2636) nvs: nvs_set bssid.set 1 0
D (2638) nvs: nvs_set sta.lis_intval 2 3
D (2642) nvs: nvs_set sta.scan_method 1 0
D (2646) nvs: nvs_set sta.sort_method 1 0
D (2650) nvs: nvs_set sta.minrssi 1 -127
D (2654) nvs: nvs_set sta.minauth 1 0
D (2658) nvs: nvs_set_blob sta.apsw 2
D (2663) nvs: nvs_set_blob sta.apinfo 700
I (2626) mf_safe_boot: set config error ESP_OK
I (2670) mf_safe_boot: esp get sta list : ESP_OK
I (2676) mf_safe_boot: set wifi sta ssid : MF_Staff
I (2681) mf_safe_boot: set wifi sta password : mfprodesp32-11
I (2688) mf_safe_boot: set wifi sta bssid : �\���������
I (2694) mf_safe_boot: set wifi sta bssid set: 0
I (2699) mf_safe_boot: set wifi sta channel : 165
I (2705) mf_safe_boot: set wifi sta listen interval : 42405
I (2711) mf_safe_boot: set wifi sta scan method : 0
I (2716) mf_safe_boot: set wifi sta sort method : 0
I (2722) mf_safe_boot: set wifi sta threshold : -127
I (2728) mf_safe_boot: set wifi sta threshold : 0
```

_Behavior when the device is not connecting to wifi_
```
I (2516) mf_safe_boot: current wifi sta ssid : AKAN_Staff
I (2522) mf_safe_boot: current wifi sta password : akanprodesp32-99
I (2529) mf_safe_boot: current wifi sta bssid :
I (2534) mf_safe_boot: current wifi sta bssid set: 0
I (2540) mf_safe_boot: current wifi sta channel : 0
I (2545) mf_safe_boot: current wifi sta listen interval : 0
I (2551) mf_safe_boot: current wifi sta scan method : 0
I (2557) mf_safe_boot: current wifi sta sort method : 0
I (2563) mf_safe_boot: current wifi sta threshold : 0
I (2569) mf_safe_boot: current wifi sta threshold : 0
I (2621) mf_safe_boot: set config error ESP_OK
I (2627) mf_safe_boot: esp get sta list : ESP_OK
I (2632) mf_safe_boot: set wifi sta ssid :
I (2637) mf_safe_boot: set wifi sta password :
fy������ mf_safe_boot: set wifi sta bssid : K
I (2648) mf_safe_boot: set wifi sta bssid set: 0
I (2653) mf_safe_boot: set wifi sta channel : 165
I (2658) mf_safe_boot: set wifi sta listen interval : 42405
I (2665) mf_safe_boot: set wifi sta scan method : 0
I (2670) mf_safe_boot: set wifi sta sort method : 0
I (2676) mf_safe_boot: set wifi sta threshold : -127
I (2681) mf_safe_boot: set wifi sta threshold : 0
```

sagb2015, 2020-02-20:
@isabelleletarte Can you paste the complete log?  You can also enable wifi logs in component config -> Wi-Fi -> Enable debug logs.

liuzfesp, 2020-03-29:
Hi @isabelleletarte , how about the status of this issue? Is it possible the application pass the incorrect SSID/password to esp_wifi_set_config()?

Alvin1Zhang, 2020-05-06:
@isabelleletarte Would you please help share if more updates for the issue? Thanks.

Alvin1Zhang, 2020-05-26:
@isabelleletarte Thanks for reporting. Would you please help share if more updates for the issue? Thanks.

Alvin1Zhang, 2020-07-07:
@isabelleletarte Thanks for reporting, will close due to short of feedback and updates, feel free to reopen with more details. Thanks.

closed 2020-07-07 by Alvin1Zhang

wcauchois, 2020-10-27:
Hi @isabelleletarte, I'm also having an issue with some of my devices failing to connect to Wi-Fi. I don't have logs but while researching ESP Wi-Fi issues I came across yours. If set_config is failing to write credentials to flash it would probably explain what I'm seeing.

Did you ever figure out how to resolve your issue? I'd very much appreciate hearing how/if you ended up resolving your issue. Thanks!