espressif/esp-idf #4767: Random Interrupt wdt timeout occurs after leave idle with light-sleep

lsgunth, 2020-02-14:
## Environment

- Development Kit:     Custom Board
- Module or chip used:  ESP32-WROOM-32D
- IDF version:  v4.1-dev-1986-gca735340f93e
- Build System:         Make
- Compiler version:   xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?:  No
- Power Supply:     USB 5V -> 3.3V switching regulator

## Problem Description

Randomly, sometimes after a few minutes but usually within an hour, the device crashes and 
halts or reboots (depending on the config option). Sometimes I don't get a panic, but the last couple times I have and I've attached the traces with the trace back called out.

Both tracebacks indicate an issue around leave_idle() on CPU 1. In both cases CPU 0 looks to be idle. 

I had it happening very quickly but then I increased CONFIG_ESP_INT_WDT_TIMEOUT_MS to 1000 (from the default 300) and I thought that fixed it but I eventually saw it again. Decreasing it back down did not seem to make it more frequent.

It's hard to give much information with certainty seeing it takes so long to reproduce but I hope the tracebacks can give enough of a clue to someone who understands that code better than I do.
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/4207217/sdkconfig.txt)

### Steps to reproduce

I do not have simple steps to reproduce this. I can tell you that my project uses light sleep, but normally runs at 80MHz:

```
static const esp_pm_config_esp32_t pm_config = {
        .max_freq_mhz = CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ,
        .min_freq_mhz = CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ,
        .light_sleep_enable = true,
};
    ESP_ERROR_CHECK(esp_pm_configure(&pm_config));
```

And the only interrupt in my application is from a GPIO that looks like this:

```
static void IRAM_ATTR gpio_isr_handler(void *arg)
{
        vTaskNotifyGiveFromISR(ifmcu_task_hdl, NULL);
}
    gpio_wakeup_enable(PIN_IFMCU_BOOT, GPIO_INTR_LOW_LEVEL);
    gpio_intr_enable(PIN_IFMCU_BOOT);
    gpio_set_intr_type(PIN_IFMCU_BOOT, GPIO_INTR_NEGEDGE);

    gpio_isr_handler_add(PIN_IFMCU_BOOT, gpio_isr_handler, NULL);
```
I can't say for certain whether disabling this interrupt fixes the issue. Without it, the application will mostly be idle. And seeing it's hard to reproduce it's hard to say anything for certain.

I've ran my project with heap debugging on and haven't found any issues. 

## Debug Logs
```
[     48.733 ]  Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
[     49.131 ]  Core 1 register dump:
[     49.131 ]  PC      : 0x400825d4  PS      : 0x00060034  A0      : 0x80082daa  A1      : 0x3ffb0c40  
[     49.132 ]  A2      : 0x00000100  A3      : 0x3ff47028  A4      : 0x02ceca43  A5      : 0x3ffb3cc0  
[     49.134 ]  A6      : 0x00000000  A7      : 0x01304fdf  A8      : 0x3ff47024  A9      : 0x01304ff0  
[     49.135 ]  A10     : 0x00000000  A11     : 0x00000000  A12     : 0xefffffff  A13     : 0x3ff47030  
[     49.136 ]  A14     : 0x02e0679e  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000006  
[     49.137 ]  EXCVADDR: 0x00000000  LBEG    : 0x40097a12  LEND    : 0x40097a19  LCOUNT  : 0x00000000  
[     49.138 ]  Core 1 was running in ISR context:
[     49.138 ]  EPC1    : 0x400d2aec  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400825d4
[     49.139 ]  
[     49.139 ]  ELF file SHA256: 064b4bc0df11a88dab0be99dc34c1aa3b7634cbc43d8561f9824f7bbb4aee8cb
[     49.139 ]  
[     49.139 ]  Backtrace: 0x400825d1:0x3ffb0c40 0x40082da7:0x3ffb0c70 0x40082b19:0x3ffb0c90 0x40081d21:0x3ffb0cc0 0x4008312b:0x3ffb0ce0 0x400d2ae9:0x3ffbc740 0x400d353f:0x3ffbc760 0x40097ad0:0x3ffbc780
[... 62 lines cut ...]
[    713.631 ]  
[    713.631 ]  Core 0 register dump:
[    713.632 ]  PC      : 0x400d322c  PS      : 0x00060e34  A0      : 0x800d3c82  A1      : 0x3ffbc8b0  
[    713.632 ]  A2      : 0x3ffb3e20  A3      : 0x00060e20  A4      : 0x00060e20  A5      : 0x00060e20  
[    713.633 ]  A6      : 0x00000001  A7      : 0x00060023  A8      : 0x00000001  A9      : 0x3ffbc8a0  
[    713.633 ]  A10     : 0x00000003  A11     : 0x00060e23  A12     : 0x00060e20  A13     : 0x00060e23  
[    713.634 ]  A14     : 0x00000001  A15     : 0x3ffbe3e0  SAR     : 0x0000001d  EXCCAUSE: 0x00000006  
[    713.634 ]  EXCVADDR: 0x00000000  LBEG    : 0x40098092  LEND    : 0x40098099  LCOUNT  : 0x00000000  
[    713.635 ]  
[    713.635 ]  ELF file SHA256: 0deea7fc692e0915207b16e43464aa448f369cb948ed3dbaca295fad4278704a
[    713.636 ]  
[    713.636 ]  Backtrace: 0x400d3229:0x3ffbc8b0 0x400d3c7f:0x3ffbc8d0 0x40098150:0x3ffbc8f0

0x400d3229: esp_pm_impl_waiti at /home/logang/projects/stove/software/emu_v3/libs/esp-idf/components/esp32/pm_esp32.c:475
0x400d3c7f: esp_vApplicationIdleHook at /home/logang/projects/stove/software/emu_v3/libs/esp-idf/components/esp_common/src/freertos_hooks.c:63
0x40098150: prvIdleTask at /home/logang/projects/stove/software/emu_v3/libs/esp-idf/components/freertos/tasks.c:3382

[    713.636 ]  
[    713.636 ]  Rebooting...
[    713.636 ]  ets Jun  8 2016 00:22:57
[    713.636 ]  
[    713.636 ]  rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/4207219/sdkconfig.txt)

lsgunth, 2020-02-15:
Added a complete log with debug output: [log.txt](https://github.com/espressif/esp-idf/files/4207721/log.txt)

Shows 5 occurrences of the bug. The device is largely idle when it occurs. In 4 out of 5 cases it occured immediately after a line like:

```
esp_netif_lwip: if0x3ffc17c0 ip unchanged
```

For the most part, the gpio interrupt in my code was not firing for most of the runs.

Runs failed sometimes after 1 minute, sometimes after 40 minutes.

lsgunth, 2020-02-18:
Ok, I've removed almost everything from my application except for the Wifi connection and have found the bug depends on two things: enabling automatic light sleep and configuring with CONFIG_COMPILER_OPTIMIZATION_PERF. Without either there is no random reset. 

I'll try to reproduce it with one of the examples when I have time but another thing that might be exacerbating the problem is that I have my DHCP lease time set rather low for these devices, which at least means the core is waking up more frequently to talk to the DHCP server.

igrr, 2020-02-24:
@lsgunth Were you able to reproduce the issue with one of the examples? I have tried a few scenarios with light sleep and CONFIG_COMPILER_OPTIMIZATION_PERF=y, but so far haven't met this specific issue.

Also if you have a chance, could you update IDF to the latest master branch, and try the scenario which was crashing? We have a new esp_timer implementation now; if updating to the latest version resolves the issue, then it is likely that we need to look for the bug in the older implementation of esp_timer.

lsgunth, 2020-03-02:
Yes, I now have a simple example that reproduces the issue based on the smart_config example. I've attached the code, the sdkconfig default and a sample log that I ran all weekend and hit the bug several times. 

It takes longer to reproduce the bug than it does with my normal setup but it still will.

[smartconfig_main.c.txt](https://github.com/espressif/esp-idf/files/4277534/smartconfig_main.c.txt)
[sdkconfig.defaults.txt](https://github.com/espressif/esp-idf/files/4277535/sdkconfig.defaults.txt)
[log9.txt](https://github.com/espressif/esp-idf/files/4277537/log9.txt)

I'll rerun this with the latest master and report back.

lsgunth, 2020-03-02:
Also: I hadn't quite finished paring down the sdkconfig defaults. I suspect some of them are still not required to hit the bug.

lsgunth, 2020-03-06:
I ran it for most of the week rebased on master. So it looks to be fixed. Hopefully that change get's into the 4.1 branch at some point.

closed 2020-03-06 by lsgunth

AhmedSalem636, 2024-04-26:
Hello
Can anyone help telling me how this issue was fixed. Because I am facing the exact same issue. sometimes it halts, some other times it reboots.

thanks in advance.