espressif/esp-idf #4768: blink example compile error: 'rtc_gpio_desc' undeclared (first use in this function); did you mean 'rtc_io_desc'?

Jaydip-Chabhadiya, 2020-02-15:
When first make flash it's difficult for beginners to get started with error

We got error
error: 'rtc_gpio_desc' undeclared (first use in this function); did you mean 'rtc_io_desc'?

Directory: esp-idf/examples/get-started/blink

Change file
sdkconfig
Line 258:
# CONFIG_RTCIO_SUPPORT_RTC_GPIO_DESC is not set
To
CONFIG_RTCIO_SUPPORT_RTC_GPIO_DESC=y

Jaydip-Chabhadiya, 2020-02-15:
Above method working for me so please add it to future release

Jaydip-Chabhadiya, 2020-02-15:
I have modified code to use it as Arduino using below steps:
1) cd ~/esp/esp-idf/examples/get-started/blink/
2) mkdir -p components && \                                                        
cd components && \
git clone https://github.com/espressif/arduino-esp32.git arduino && \
cd arduino && \
git submodule update --init --recursive && \
cd ../.. && \
make menuconfig -> Arduino Configuration -> Autostart Arduino setup and loop on boot

Then 

make flash

Error: 
/home/anon/esp/esp-idf/examples/get-started/hello_world/components/arduino/cores/esp32/esp32-hal-gpio.c:90:24: error: 'rtc_gpio_desc' undeclared (first use in this function); did you mean 'rtc_io_desc'?

bitmakk, 2020-02-16:
You have to enable it the rtc gpio in menuconfig

chegewara, 2020-02-16:
Are you using arduino as submodule with esp-idf v3.3 or v4.x (master)?

martinberlin, 2020-05-15:
Just in case this help someone the rtc gpio in menuconfig is in:
Component Config -> Driver configurations -> RTCIO configuration

closed 2020-05-26 by costaud