espressif/esp-idf #4770: When trying to run export.sh I get `no such file or directory: $HOME/esp/esp_idf/tools/idf_tools.py`

vmgarcia, 2020-02-15:
## Environment
- IDF version: v4.1-dev-2263-g605da33c3
- Operating System:     macOS
- Using an IDE?: No

## Problem Description

When running export.sh I get an error saying 

```
Adding ESP-IDF tools to PATH...
idf_export_main:33: no such file or directory: /Users/victorgarcia/esp/esp_idf/tools/idf_tools.py
```

Before this error I get an error saying that IDF_PATH is not set although I'm using the path that is described in the getting started guide.

igrr, 2020-02-15:
Hi @vmgarcia, thanks for reporting this. Couple of questions: if you run `/Users/victorgarcia/esp/esp_idf/tools/idf_tools.py export` from the shell, does it report the same error? Which shell are you using (bash, zsh)? What does `which python` report in your shell?

vmgarcia, 2020-02-15:
hi @igrr thanks for the quick response!

When I run `/Users/victorgarcia/esp/esp_idf/tools/idf_tools.py export` I get a similar error
`zsh: no such file or directory: /Users/victorgarcia/esp/esp_idf/tools/idf_tools.py`

`which python` returns `/usr/bin/python`

igrr, 2020-02-15:
Okay, what about if you run `python /Users/victorgarcia/esp/esp_idf/tools/idf_tools.py export`?

vmgarcia, 2020-02-15:
"Also please check if you have esp-idf or esp_idf in the path (I see both in your last post)."

Nice callout, sorry I just copy pasted your command but when I ran export.sh using the previous command you told me to run I got a valid output
`/Users/victorgarcia/esp/esp_idf/tools/idf_tools.py export` returned 

```
export IDF_PYTHON_ENV_PATH="/Users/victorgarcia/.espressif/python_env/idf4.1_py2.7_env";export OPENOCD_SCRIPTS="/Users/victorgarcia/.espressif/tools/openocd-esp32/v0.10.0-esp32-20191114/openocd-esp32/share/openocd/scripts";export PATH="/Users/victorgarcia/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin:/Users/victorgarcia/.espressif/tools/xtensa-esp32s2-elf/esp-2019r2-8.2.0/xtensa-esp32s2-elf/bin:/Users/victorgarcia/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/victorgarcia/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/victorgarcia/.espressif/tools/openocd-esp32/v0.10.0-esp32-20191114/openocd-esp32/bin:/Users/victorgarcia/.espressif/python_env/idf4.1_py2.7_env/bin:/Users/victorgarcia/esp/esp-idf/tools:$PATH"```

igrr, 2020-02-15:
> Nice callout, sorry I just copy pasted your command 

Well, I just copy-pasted the path from your original post :)

> no such file or directory: /Users/victorgarcia/esp/esp_idf/tools/idf_tools.py

So is IDF_PATH variable set to `esp_idf` or `esp-idf`?.. (check by `echo $IDF_PATH`)

vmgarcia, 2020-02-15:
>So is IDF_PATH variable set to esp_idf or esp-idf?.. (check by echo $IDF_PATH)

I had set it to with an underscore.... 

When I do correct it I get 

```
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Traceback (most recent call last):
  File "/Users/victorgarcia/esp/esp-idf/tools/check_python_dependencies.py", line 42, in <module>
    default_requirements_path = os.path.join(idf_path, 'requirements.txt')
  File "/Users/victorgarcia/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/posixpath.py", line 70, in join
    elif path == '' or path.endswith('/'):
AttributeError: 'NoneType' object has no attribute 'endswith'```

igrr, 2020-02-26:
Just to check, are you exporting IDF_PATH variable, or just setting it? I.e. `export IDF_PATH=$HOME/esp/esp-idf` versus `IDF_PATH=$HOME/esp/esp-idf`. It seems that `check_python_dependencies.py` can return this error if somehow `IDF_PATH` is not found in its environment.

igrr, 2020-03-30:
@vmgarcia does the issue still exist?

closed 2020-04-09 by espressif-bot