espressif/esp-idf #4771: rtcio_hal_isolate in ULP assembly code?

mahesh2000, 2020-02-15:
hi, i found `rtcio_hal_isolate(int rtcio_num)` and `rtcio_hal_isolate()` very useful in reducing the power usage of the ULP.is there a ULP assembly version?

i need to have the core sleep most of the time. the ULP wakes every few seconds, turns on a GPIO pin & sets it HIGH. it then reads an ADC pin. after that, it sets the GPIO pin low. and then it should isolate all pins (it seems that this is best to save power), go to sleep. and then wake up again.

i've seen that the power usage of the ULP when this C code is run by the main core reduces power use to 40uA.
```
gpio_num_t ulp_blink_pin = GPIO_NUM_2; // this is also RTC_IO12
```

```
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
  WiFi.mode(WIFI_OFF);
  btStop();
  adc_power_off();  

  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_0_GPIO_NUM);  //    36
  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_1_GPIO_NUM); //    37
  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_2_GPIO_NUM); //    38
  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_3_GPIO_NUM); //    39
  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_4_GPIO_NUM); //    34
  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_5_GPIO_NUM); //    35
  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_6_GPIO_NUM); //    25
  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_7_GPIO_NUM); //    26
  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_8_GPIO_NUM); //    33
  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_9_GPIO_NUM); //    32
  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_10_GPIO_NUM);  //   4
  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_11_GPIO_NUM);  //   0
  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_12_GPIO_NUM);  //   2
  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_13_GPIO_NUM);  //   15
  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_14_GPIO_NUM);  //   13
  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_15_GPIO_NUM);  //   12
  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_16_GPIO_NUM);  //   14
  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_17_GPIO_NUM);  //   27

  rtc_gpio_force_hold_dis_all();
  rtc_gpio_pulldown_dis(ulp_blink_pin); // disable VCOM pulldown (saves 80µA). what's this about?
  rtc_gpio_set_direction(ulp_blink_pin, RTC_GPIO_MODE_OUTPUT_ONLY);

```

sch0bert, 2020-02-15:
This is a great request, I'm also interested.

mahesh2000, 2020-02-17:
Hi @boarchuz, I need to minimize power so that the ESP32 can run on 2xAA batteries for *5 years*. In my device, the core is mostly in deep sleep mode. the ulp wakes every 2 seconds, reads the ADC on a pin, flashes an LED (10ms) on another RTC GPIO pin, then calls `halt`. A few times every day it `wake`s the core so that the core can send a message via WiFi and then go right back to deep sleep. In my experiments, isolating the pins brought the power down to 40uA during core deep sleep with no ULP usage on a board eviscerated of the UART, power regulator and most other parts by de-soldering and/or ripping off with pliers. Highly satisfying after a long day.

What if I isolate the pins on the first run of the ULP, thereafter after every core awakening? Would I even need to do the latter?

mahesh2000, 2020-02-17:
Hi @boarchuz, is there a macro into which I could put the instructions you've given? I can then run it for all appropriate pins.

Also, is there a reference for the assembler for the esp32 of which you know? I have the instruction set but I've been piecing together the assembler instructions from reading projects on github and elsewhere. Thanks!

chegewara, 2020-02-17:
> I need to minimize power so that the ESP32 can run on 2xAA batteries for _5 years_. 

Is it even possible with esp32 in deep sleep mode without running any code and without driving any pins (just idle run)?

boarchuz, 2020-02-17:
There's nothing to gain by doing this with the ULP. Use your rtc_gpio_isolate code above on reset (no need to redo it every deep sleep wakeup).

mahesh2000, 2020-02-17:
@boarchuz, wow. if that's true, you just made my life awesome! i'm now turning off WiFi, BT, some doing some hodgepodge disabling of pins & isolation, except for 3 pins that the ULP code needs. is this sufficient? the main items and then the code follow. i'm a noob with the ESP32 so this is really helpful. also, is there a guide to the ESP's assembler? it seems similar to the old Sun micro but isn't entirely the same.

**Overview:**

```
/*
 * First time, after boot, the core does this:
 *   1. set pinMode(0, OUTPUT) before calling ulptool_load_binary(). why? took this from another program
 *   2. init ULP, call ulptool_load_binary()
 *   3. mess with some GPIOs after ulptool_load_binary()
 *   4. set ULP wakeup period to 5 secs: ulp_set_wakeup_period()
 *   5. mess with adc & adc1 after setting the ulp wake period:
 *   6. turn WiFi and BT off
 *   7. call esp_sleep_pd_config(). why?
 *   8. isolate most pins except the 3 we're using (1 for ADC, 2 for OUTPUT):
 *   9. call rtc_gpio_force_hold_dis_all(). why?
 *   10. Start the ULP program.
 * 
 * And then,
 *   enable wake up by the ULP, then make the core go to sleep. This code 
 *   will be run every time the core wakes up.
 */
```

**The code:**
```
void setup(){
  Serial.begin(115200); delay(100);
  esp_sleep_wakeup_cause_t cause = esp_sleep_get_wakeup_cause();
  // print_wakeup_cause(cause); 

  if (cause != ESP_SLEEP_WAKEUP_ULP) {
    // 1. set pinMode(0, OUTPUT) before calling ulptool_load_binary(). took this from another program
    pinMode(0, OUTPUT); // don't change, -10µA???
    
    // 2. init ULP program:
    esp_err_t err = ulptool_load_binary(0, ulp_main_bin_start, (ulp_main_bin_end - ulp_main_bin_start) / sizeof(uint32_t));
    ESP_ERROR_CHECK(err);

    // 3. mess with some GPIOs after ulptool_load_binary()
    rtc_gpio_init(ulp_blink_pin);
    rtc_gpio_pulldown_dis(ulp_blink_pin); // disable VCOM pulldown (saves 80µA). what's this about?
    rtc_gpio_set_direction(ulp_blink_pin, RTC_GPIO_MODE_OUTPUT_ONLY);

    // 4. set ULP wakeup period
    #define FIVE_SECONDS 5
    ulp_set_wakeup_period(0, FIVE_SECONDS * 1000 * 1000);

    // 5. mess with adc & adc1 after setting the ulp wake period:
    adc_power_off();
    adc1_config_channel_atten(ADC1_CHANNEL_4, ADC_ATTEN_DB_11);
    adc1_config_width(ADC_WIDTH_BIT_12);
    adc1_ulp_enable();
    
    // 6. turn WiFi and BT off:
    WiFi.disconnect(true);
    WiFi.mode(WIFI_OFF);
    btStop();
    esp_bt_controller_disable();
    
    // 7. call esp_sleep_pd_config()
    esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);

    // 8. isolate most pins except the 3 we're using (1 for ADC, 2 for OUTPUT):
    rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_0_GPIO_NUM);  //    36
    rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_1_GPIO_NUM); //    37
    rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_2_GPIO_NUM); //    38
    rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_3_GPIO_NUM); //    39
    rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_4_GPIO_NUM); //    34
    rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_5_GPIO_NUM); //    35
    rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_6_GPIO_NUM); //    25
    rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_7_GPIO_NUM); //    26
    rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_8_GPIO_NUM); //    33

    // using this pin for OUTPUT by the ULP:
    //  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_9_GPIO_NUM); //    32
    rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_10_GPIO_NUM);  //   4
    rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_11_GPIO_NUM);  //   0

    // using this pin for OUTPUT by the ULP:
    //  rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_12_GPIO_NUM);  //   2
    rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_13_GPIO_NUM);  //   15
    rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_14_GPIO_NUM);  //   13
    rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_15_GPIO_NUM);  //   12
    rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_16_GPIO_NUM);  //   14
    rtc_gpio_isolate((gpio_num_t) RTCIO_CHANNEL_17_GPIO_NUM);  //   27

    // 9. do we need this the following?
    rtc_gpio_force_hold_dis_all();

    /* 10. Start the program */
    esp_err_t err = ulp_run((&ulp_entry - RTC_SLOW_MEM) / sizeof(uint32_t));
    ESP_ERROR_CHECK(err);
 }

  // Enable wake up by the ULP, then make the core go to sleep
  esp_sleep_enable_ulp_wakeup();  // must be executed every time the core goes to sleep so that it can wake on a signal from the ULP.
  esp_deep_sleep_start();
  Serial.println("exiting goToDeepSleep(). This should never be printed.");
}
```

mahesh2000, 2020-02-17:
@chegewara, i was able to get consumption down to 40uA on idle deep sleep. so, i think the answer is yes.

chegewara, 2020-02-17:
> @chegewara, i was able to get consumption down to 40uA on idle deep sleep. so, i think the answer is yes.

Nice. Now add 10mA for LED for 10ms. Its 1800 per hour:
1800 x 10 ms = 18s 
18s x 10 mA = 180 mAh
180 x 24 = 4320 mAh per day.
Maybe my math is wrong, im not good at hardware.

mahesh2000, 2020-02-17:
the math is this: the LED comes on for 10ms every 2 secs (200ms), so a duty cycle of 0.005. each day, 24 hours x 0.005 = 0.12h, i.e. 7.2 mins, i.e. 432 *seconds*. at 10mA, the power usage per day is 0.12h x 10mAh = 1.2mAh. with a 2000mAh battery, that's 2000/1.2 = 4.5 years. if _nothing else happens_ other than the LED turning on. to which we need to add WiFi use, ULP current consumption, sensors' power consumption, etc.

chegewara, 2020-02-17:
@mahesh2000 Dont take me wrong, im not criticizing you or something, just trying to show you it may be not possible to achieve with esp32 your goals. If you can, i would like to learn something from your experience.

mahesh2000, 2020-02-17:
@chegewara, it was a good question, it made me recheck my numbers :-). the core & wifi take up most of the power consumption. so if you do the numbers, if you turn on wifi only a few times a day, optimize it to use a static IP address, etc., you should be able to run for years on a battery. and, yes, turning on the LED once every 10 seconds instead of 2 makes a good bit of difference, too.

mahesh2000, 2020-02-19:
i just came across this in the ESP32's [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf): "The internal LDO can be configured as having 1.8 V, or the same voltage as VDD3P3_RTC. It can be powered off via software to minimize the current of flash/SRAM during the Deep-sleep mode." btw, i think we're going to can the ESP32 because its ADC sensor is too noisy.