espressif/esp-idf #4773: Errors while interacting with microSD card

katzuv, 2020-02-16:
### Hardware:
Board:							ESP32-CAM
Core Installation/update date:			
IDE name:						Arduino IDe
Flash Frequency:					don't know
PSRAM enabled:                                          yes
Upload Speed:						115200
Computer OS:                                            Ubuntu 16.04

### Description:
I have an ESP32-CAM board with a microSD card attached (16GB). In my `setup` function I create a directory. In `loop` the ESP32-CAM takes images and stores in the directory.
The code successfully creates the directory and captures one image inside the directory. However, when the `loop` function runs the second time, I get the following errors:
```
E (10911) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0xffffffff
E (10912) diskio_sdmmc: sdmmc_read_blocks failed (-1)
E (11913) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (11913) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (11914) diskio_sdmmc: sdmmc_read_blocks failed (263)
```
What can I do to fix those?
Thank you!

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
[Link to gist](url)
### Debug Messages:
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
Can't find "debug" in tools menu of Arduino IDE.

igrr, 2020-02-26:
@katzuv Do you have a logic analyzer of a scope to capture the exchange on the SD card bus? From the log it seems that the card hasn't responded to the read command. It could be a problem in the IDF (i.e. the read command was not sent correctly) or a problem with the card (e.g. power supply issue, card controller went into brownout). Checking the exchange with a logic analyzer might help.

igrr, 2020-03-02:
Hi @katzuv, in your code you are re-configuring and isolating GPIO4:

```c++
  pinMode(4, OUTPUT);              //GPIO for LED flash
  digitalWrite(4, LOW);            //turn OFF flash LED
  rtc_gpio_hold_en(GPIO_NUM_4);    //make sure flash is held LOW in sleep
```

keep in mind that GPIO4 is part of SD interface, that's HS2_DATA1 line in the [schematic](https://github.com/SeeedDocument/forum_doc/raw/master/reg/ESP32_CAM_V1.6.pdf). You shouldn't reconfigure this pin if you are using the SD card in 4-line mode.

You can still use the SD card in 1-line mode (with lower throughput), please see https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/sdmmc_host.html#configuring-bus-width-and-frequency.

Alvin1Zhang, 2020-05-14:
@katzuv Thanks for reporting. Would you please help share if any updates or details for this issue? Thanks.

closed 2021-02-17 by igrr

dpnebert, 2021-07-30:
Yes, I commented out
`
pinMode(4, OUTPUT);              //GPIO for LED flash
  digitalWrite(4, LOW);            //turn OFF flash LED
  rtc_gpio_hold_en(GPIO_NUM_4);    //make sure flash is held LOW in sleep
`
And now it worked.