espressif/esp-idf #4774: DEPRECATION: Python 2.7... reach end of life

beta-tester, 2020-02-16:
i guess python 2.7 reached end of life...

a new Python environment in /home/pi/.espressif/python_env/idf4.2_py2.7_env
Installing virtualenv
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting virtualenv

bitmakk, 2020-02-16:
The idf with Cmake and idf.py uses python3

dobairoland, 2020-02-17:
Hi @beta-tester. Thank you for the notice. We are aware of this fact but we are still supporting Python 2.7 in order to support developers with older versions of IDF. However, we are encouraging developers to switch to Python 3.

So you can either switch to Python 3 or you can safely ignore the deprecation message.

If you need assistance for switching then please state your OS (and version) and the IDF version you are intending to use.

closed 2020-02-17 by dobairoland

beta-tester, 2020-02-17:
yes, at the moment it is only a notification... without impact.
i can compile and flash without problems as far as i could see.

i use Raspberry Pi + [Raspbian (2020-02-05)](https://downloads.raspberrypi.org/raspbian/images/raspbian-2020-02-14/) + [SBC-NodeMCU-ESP32](http://anleitung.joy-it.net/wp-content/uploads/2018/07/SBC-NodeMCU-ESP32-Datasheet_V1.3.pdf)
Raspbian and esp-idf is the newest available downloaded and upgraded yesterday.
followed the instructions [Get Started ([Linux] [Debian] [armel])](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/)

Raspbian:
```
pi@raspberrypi:~/esp/esp-idf $ uname -a
Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
```

ESP-IDF:
```
pi@raspberrypi:~/esp/esp-idf $ cd ~/esp/esp-idf/
pi@raspberrypi:~/esp/esp-idf $ idf.py --version
fatal: No names found, cannot describe anything.
WARNING: Git version unavailable, reading from source
ESP-IDF v4.2.0
pi@raspberrypi:~/esp/esp-idf $ git describe --tags
fatal: No names found, cannot describe anything.
pi@raspberrypi:~/esp/esp-idf $ git rev-parse HEAD
605da33c3d4fa93c6c98da2622f053ecbe5cfbf4
pi@raspberrypi:~/esp/esp-idf $ git version
git version 2.20.1
```

dobairoland, 2020-02-17:
Yes, the `export.sh` has to been called not just after each reboot but for every new terminal window. It can guess the `IDF_PATH` but it is better to set it up in `.bashrc` or `.bash_profile`. 

You should be able to install Python 3 on raspbian. I cannot test it now but I assume that `sudo apt install python3` should work. After that `python3 --version` can be used to test it.

You can setup `alias python=python3` temporarily only once and run `install.sh`. This will create a Python3-base virtual environment in the `~/.espressif` folder.

But as I sad before, you can just ignore the deprecation warning an use the default Python environment.

beta-tester, 2020-02-17:
python 3.7.3 is already installed in Raspbian out of the box, but the command `python` and `pip` is linked to 2.7.16 by default.

```
pi@raspberrypi:~/esp/esp-idf $ python --version
Python 2.7.16
pi@raspberrypi:~/esp/esp-idf $ python3 --version
Python 3.7.3
pi@raspberrypi:~/esp/esp-idf $ pip --version
pip 20.0.2 from /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages/pip (python 2.7)
pi@raspberrypi:~/esp/esp-idf $ pip3 --version
pip 18.1 from /usr/lib/python3/dist-packages/pip (python 3.7)
```

i will ignore the DEPRECATION notification for the moment...

thank you for your help.

dobairoland, 2020-02-17:
In this the following would create a Python3-based ESP-IDF environment:
```bash
alias python=python3
$IDF_PATH/install.sh
```

After `export.sh` the output of `python --version` would be `3.7.3` and that would be used as the default one with ESP-IDF.

beta-tester, 2020-02-17:
i tried your suggestion...
closed the terminal window and opened a new one...
but at `install.sh` i still see the DEPRECATION notification and
```bash
pi@raspberrypi:~ $ cd ~/esp/esp-idf/
pi@raspberrypi:~/esp/esp-idf $ python --version
Python 2.7.16
pi@raspberrypi:~/esp/esp-idf $ alias python=python3
pi@raspberrypi:~/esp/esp-idf $ python --version
Python 3.7.3
pi@raspberrypi:~/esp/esp-idf $ . ./install.sh 
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2019r2-8.2.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2019r2-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20191114 (already installed)
Installing Python environment and packages
fatal: No names found, cannot describe anything.
WARNING: Git describe was unsuccessul: Command '['git', 'describe', '--tags']' returned non-zero exit status 128
Installing Python packages from /home/pi/esp/esp-idf/requirements.txt
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Ignoring None: markers 'sys_platform == "win32"' don't match your environment
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: setuptools>=21 in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 4)) (44.0.0)
Requirement already satisfied: click>=5.0 in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 8)) (7.0)
Requirement already satisfied: pyserial>=3.0 in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 9)) (3.4)
Requirement already satisfied: future>=0.15.2 in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 11)) (2.8)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 13)) (0.26)
Requirement already satisfied: ipaddress; python_version < "3" in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/pi/esp/esp-idf/requirements.txt (line 11)) (1.0.23)
Requirement already satisfied: enum34; python_version < "3" in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/pi/esp/esp-idf/requirements.txt (line 11)) (1.1.6)
Requirement already satisfied: six>=1.4.1 in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/pi/esp/esp-idf/requirements.txt (line 11)) (1.14.0)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/pi/esp/esp-idf/requirements.txt (line 11)) (1.14.0)
Requirement already satisfied: pycparser in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r /home/pi/esp/esp-idf/requirements.txt (line 11)) (2.19)
All done! You can now run:

  . ./export.sh

pi@raspberrypi:~/esp/esp-idf $ . ./export.sh &>> ~/log-export.txt
```

even worse, when i now try to run `export.sh`, it will close the terminal window unexpectedly...
to watch the output, i redirect the output of export.sh to a file...
```
Adding ESP-IDF tools to PATH...
fatal: No names found, cannot describe anything.
WARNING: Git describe was unsuccessul: Command '['git', 'describe', '--tags']' returned non-zero exit status 128
Checking if Python packages are up to date...
The following Python requirements are not satisfied:
future>=0.15.2
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
To install the missing packages, please run "/home/pi/esp/esp-idf/install.sh"
```

so i guess i have to install other python3 specific libraries

dobairoland, 2020-02-17:
`install.sh` should be run directly without `.` (dot) before it. Dot is necessary only before `export.sh`.

beta-tester, 2020-02-17:
i tried different compinations...
the dot doesn't make a different...

```
pi@raspberrypi:~/esp/esp-idf $ python --version
Python 2.7.16
pi@raspberrypi:~/esp/esp-idf $ alias python=python3
pi@raspberrypi:~/esp/esp-idf $ python --version
Python 3.7.3

pi@raspberrypi:~/esp/esp-idf $ install.sh
bash: install.sh: command not found

pi@raspberrypi:~/esp/esp-idf $ ./install.sh
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2019r2-8.2.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2019r2-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20191114 (already installed)
Installing Python environment and packages
fatal: No names found, cannot describe anything.
WARNING: Git describe was unsuccessul: Command '['git', 'describe', '--tags']' returned non-zero exit status 128
Installing Python packages from /home/pi/esp/esp-idf/requirements.txt
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Ignoring None: markers 'sys_platform == "win32"' don't match your environment
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: setuptools>=21 in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 4)) (44.0.0)
Requirement already satisfied: click>=5.0 in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 8)) (7.0)
Requirement already satisfied: pyserial>=3.0 in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 9)) (3.4)
Requirement already satisfied: future>=0.15.2 in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 11)) (2.8)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 13)) (0.26)
Requirement already satisfied: ipaddress; python_version < "3" in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/pi/esp/esp-idf/requirements.txt (line 11)) (1.0.23)
Requirement already satisfied: enum34; python_version < "3" in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/pi/esp/esp-idf/requirements.txt (line 11)) (1.1.6)
Requirement already satisfied: six>=1.4.1 in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/pi/esp/esp-idf/requirements.txt (line 11)) (1.14.0)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/pi/esp/esp-idf/requirements.txt (line 11)) (1.14.0)
Requirement already satisfied: pycparser in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r /home/pi/esp/esp-idf/requirements.txt (line 11)) (2.19)
All done! You can now run:

  . ./export.sh

pi@raspberrypi:~/esp/esp-idf $ bash install.sh 
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2019r2-8.2.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2019r2-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20191114 (already installed)
Installing Python environment and packages
fatal: No names found, cannot describe anything.
WARNING: Git describe was unsuccessul: Command '['git', 'describe', '--tags']' returned non-zero exit status 128
Installing Python packages from /home/pi/esp/esp-idf/requirements.txt
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Ignoring None: markers 'sys_platform == "win32"' don't match your environment
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[... cut ...] (1.14.0)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/pi/esp/esp-idf/requirements.txt (line 11)) (1.14.0)
Requirement already satisfied: pycparser in /home/pi/.espressif/python_env/idf4.2_py2.7_env/lib/python2.7/site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r /home/pi/esp/esp-idf/requirements.txt (line 11)) (2.19)
All done! You can now run:

  . ./export.sh

```

beta-tester, 2020-02-22:
i tried a new installation of ESP-IDF
to a virgine installation of Ubuntu 20.04 LTS (Focal Fossa) Daily Build.

with success, but some changes were needed.

i have to adjust the pre-requirements from `python*` to `python3*` stuff:
```
sudo apt install git wget flex bison gperf python3 python3-pip python3-setuptools python3-serial python3-click python3-cryptography python3-future python3-pyparsing python3-pyelftools cmake ninja-build ccache libffi-dev libssl-dev
```

on Ubuntu 20.04 there is no alias or symbolic link for `phython`, only `python3`.
and it seems, a normal `alias python=python3` will not be visible inside `install.sh`.
so i created some symbolic links as workaround until the official esp-idf is able to handle python3 by itself:
```
###
sudo rm /usr/bin/python
sudo ln -s python3 /usr/bin/python
#
sudo rm /usr/bin/pdb /usr/bin/pip /usr/bin/pydoc /usr/bin/pygettext /usr/bin/python-config
sudo ln -s pdb3 /usr/bin/pdb
sudo ln -s pip3 /usr/bin/pip
sudo ln -s pydoc3 /usr/bin/pydoc
sudo ln -s pygettext3 /usr/bin/pygettext
###
```

the rest of the installation will be as usual:
```
mkdir -p ~/esp
cd ~/esp
git clone --depth=1 --recursive https://github.com/espressif/esp-idf.git

cd ~/esp/esp-idf
./install.sh
```

one error showed up during installation, but it did not interfer the installation process:
```
Building wheels for collected packages: distlib
  Running setup.py bdist_wheel for distlib ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-8f40b8jy/distlib/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-hys8om7i --python-tag cp38:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for distlib
  Running setup.py clean for distlib
Failed to build distlib
```

and also building will be as usual:
```
.  ~/esp/esp-idf/export.sh
idf.py build
```

dobairoland, 2020-02-22:
There is also this solution https://github.com/espressif/esp-idf/blob/master/tools/docker/Dockerfile using `update-alternatives --install /usr/bin/python python /usr/bin/python3 10`

gojimmypi, 2020-05-23:
none of the alias or linking options worked for me on Ubuntu 20.04. I'm seeing this error when running `install.sh` for the idf:

```
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200420 (already installed)
Installing Python environment and packages
Creating a new Python environment in /home/gojimmypi/.espressif/python_env/idf4.2_py2.7_env
Installing virtualenv
/usr/bin/python: No module named pip
Traceback (most recent call last):
  File "/home/gojimmypi/esp/esp-idf/tools/idf_tools.py", line 1492, in <module>
    main(sys.argv[1:])
  File "/home/gojimmypi/esp/esp-idf/tools/idf_tools.py", line 1488, in main
    action_func(args)
  File "/home/gojimmypi/esp/esp-idf/tools/idf_tools.py", line 1205, in action_install_python_env
    stdout=sys.stdout, stderr=sys.stderr)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python', '-m', 'pip', 'install', '--user', 'virtualenv']' returned non-zero exit status 1
```

The only thing that worked for me is this [horrible hack from stackoverflow](https://stackoverflow.com/questions/26266437/how-to-use-python2-7-pip-instead-of-default-pip/26267333#26267333)

```
$ wget https://bootstrap.pypa.io/get-pip.py
$ sudo python2.7 get-pip.py
```
but at least it gives a warning:
```
Creating a new Python environment in /home/gojimmypi/.espressif/python_env/idf4.2_py2.7_env
Installing virtualenv
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Collecting virtualenv
  Downloading virtualenv-20.0.21-py2.py3-none-any.whl (4.7 MB)
```
what are the plans for this?

dobairoland, 2020-05-23:
@gojimmypi The recommendation is to use Python 3 as described here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup.html#setting-up-python-3-as-default-for-ubuntu-and-debian

gojimmypi, 2020-05-23:
@dobairoland that worked perfectly! thank you. :)

for reference:
```
sudo apt-get install python3 python3-pip python3-setuptools
sudo update-alternatives --install /usr/bin/python python /usr/bin/python3 10
```
(not just for [docker files](https://github.com/espressif/esp-idf/issues/4774#issuecomment-589971273))

bear24rw, 2020-12-11:
Everyone on my project hit this issue too and had to manually symlink `python3` to `python` (which can break other things depending on python 2). Isn't the official python binary called `python3`? Should the `#!/usr/bin/env python` not specify that explicitly?

dobairoland, 2020-12-12:
@bear24rw ESP-IDF still works with Python 2 despite the deprecation warning. Using `#!/usr/bin/env python3` would mean breaking the system for those users who doesn't have Python 3.

bear24rw, 2020-12-12:
@dobairoland I see. The usage of the print() function, the fact that the idf documentation recommends python3 as the default python [[here]](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup.html#setting-up-python-3-as-default-for-ubuntu-and-debian) , and the fact that we hit errors pip installing the dependancies led me to believe python2 was not intended to be supported.

I just tired again in a clean ubuntu 20.04 docker and got the same error we got on macOS:
```
ERROR: Could not find a version that satisfies the requirement bidict>=0.21.0 (from python-socketio>=1.6.1->Flask-SocketIO<3.0,>=2.9->gdbgui==0.13.2.0->-r /esp-idf/requirements.txt (line 14)) (from versions: 0.1.5, 0.2.1, 0.3.0, 0.3.1, 0.9.0rc0, 0.9.0.post1, 0.10.0, 0.10.0.post1, 0.11.0, 0.12.0.post1, 0.13.0, 0.13.1, 0.14.0, 0.14.1, 0.14.2, 0.15.0.dev0, 0.15.0.dev1, 0.15.0rc1, 0.15.0, 0.16.0, 0.17.0, 0.17.1, 0.17.2, 0.17.3, 0.17.4, 0.17.5, 0.18.0, 0.18.1, 0.18.2, 0.18.3, 0.18.4)
ERROR: No matching distribution found for bidict>=0.21.0 (from python-socketio>=1.6.1->Flask-SocketIO<3.0,>=2.9->gdbgui==0.13.2.0->-r /esp-idf/requirements.txt (line 14))
```

Maybe I should open a new issue for this?

dobairoland, 2020-12-12:
@bear24rw I'm sorry for the issue. I already fixed it but it haven't made into the Github repo yet.

The fix is to add the following line into the `requirements.txt` file:
`python-socketio<5; python_version<="2.7"`

bear24rw, 2020-12-12:
@dobairoland no worries! Thanks for the fix 👍🏻

KaeLL, 2020-12-18:
> @bear24rw I'm sorry for the issue. I already fixed it but it haven't made into the Github repo yet.
> 
> The fix is to add the following line into the `requirements.txt` file:
> `python-socketio<5; python_version<="2.7"`

Will it be backported to v4.2?

dobairoland, 2020-12-18:
@KaeLL The same applies for v4.2. It is already done but it is not yet on Github.