espressif/esp-idf #4775: DownloadError while installing esp-idf tools, when running install.sh. MacOS Mojave 10.14.6

mattkefford, 2020-02-16:
## Environment
- Development Kit: ESP32-Wrover-Kit (not got that far though)
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used:  ESP32-WROVER
- IDF version: v4.1-dev-2263-g605da33c3
- Build System: Not sure - no instructions to find [Make|CMake|idf.py]
- Compiler version: -bash: xtensa-esp32-elf-gcc: command not found
- Operating System: macOS
- Using an IDE?: Have not got that far yet.
- Power Supply: USB (Have not got that far yet)

## Problem Description
I am trying to install the ESP32 IDF tools on my Mac for the first time, following the guide https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#get-started-get-esp-idf.

I have got to step 3 where you are to run the install.sh script and this is the output:
Keffords-Mac:esp-idf Kefford$ ./install.sh 
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2019r2-8.2.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2019r2-8.2.0 (already installed)
Installing esp32ulp-elf@2.28.51-esp-20191205
Downloading binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/Kefford/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1076)>
WARNING: Failed to download https://github.com/espressif/binutils-esp32ulp/releases/download/v2.28.51-esp-20191205/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/Kefford/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz.tmp
Downloading binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/Kefford/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1076)>
WARNING: Failed to download https://github.com/espressif/binutils-esp32ulp/releases/download/v2.28.51-esp-20191205/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/Kefford/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz.tmp
Downloading binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/Kefford/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1076)>
WARNING: Failed to download https://github.com/espressif/binutils-esp32ulp/releases/download/v2.28.51-esp-20191205/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/Kefford/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz.tmp
ERROR: Failed to download, and retry count has expired
Traceback (most recent call last):
  File "/Users/Kefford/esp/esp-idf/tools/idf_tools.py", line 1487, in <module>
    main(sys.argv[1:])
  File "/Users/Kefford/esp/esp-idf/tools/idf_tools.py", line 1483, in main
    action_func(args)
  File "/Users/Kefford/esp/esp-idf/tools/idf_tools.py", line 1181, in action_install
    tool_obj.download(tool_version)
  File "/Users/Kefford/esp/esp-idf/tools/idf_tools.py", line 598, in download
    raise DownloadError()
__main__.DownloadError

Taking a suggestion from #4364 I tried adding python3 to the start of the two lines in install.sh:
python3 ${IDF_PATH}/tools/idf_tools.py install
python3 ${IDF_PATH}/tools/idf_tools.py install-python-env

This did not fix the problem for me though.

I also tried creating the ".espressif" folder as I noticed it was not created yet, but this is not allowed so I created an "espressif" folder instead (and the dist folder too) and changed my IDF_TOOLS_PATH by adding to ~/.bash_profile the following:
export IDF_TOOLS_PATH=/Users/Kefford/espressif

This didn't fix the problem either and produced the following output:
Keffords-Mac:esp-idf Kefford$ ./install.sh 
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Installing xtensa-esp32-elf@esp-2019r2-8.2.0
Downloading xtensa-esp32-elf-gcc8_2_0-esp-2019r2-macos.tar.gz to /Users/Kefford/espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-macos.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1076)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-macos.tar.gz to /Users/Kefford/espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-macos.tar.gz.tmp
Downloading xtensa-esp32-elf-gcc8_2_0-esp-2019r2-macos.tar.gz to /Users/Kefford/espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-macos.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1076)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-macos.tar.gz to /Users/Kefford/espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-macos.tar.gz.tmp
Downloading xtensa-esp32-elf-gcc8_2_0-esp-2019r2-macos.tar.gz to /Users/Kefford/espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-macos.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1076)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-macos.tar.gz to /Users/Kefford/espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-macos.tar.gz.tmp
ERROR: Failed to download, and retry count has expired
Traceback (most recent call last):
  File "/Users/Kefford/esp/esp-idf/tools/idf_tools.py", line 1487, in <module>
    main(sys.argv[1:])
  File "/Users/Kefford/esp/esp-idf/tools/idf_tools.py", line 1483, in main
    action_func(args)
  File "/Users/Kefford/esp/esp-idf/tools/idf_tools.py", line 1181, in action_install
    tool_obj.download(tool_version)
  File "/Users/Kefford/esp/esp-idf/tools/idf_tools.py", line 598, in download
    raise DownloadError()
__main__.DownloadError

Now I am stuck after hours of trying to get the tools installed and not sure what to do. My background is in electronics hardware and embedded C so I have done my best but I feel you have to be an expert in python and OS based development just to get going with the ESP32!

Hope someone can help, thanks.

### Expected Behavior
That following the Getting Started guide allows installation of the tools.

### Actual Behavior
Install script does not install tools.

### Steps to reproduce
As mentioned, this is the first time installing these tools so should be a nice clean install. I do have python 2 and python 3 both installed and then for the rest just follow the Getting Started guide.

igrr, 2020-02-16:
Hi @mattkefford, sorry that it didn't work out of the box. It looks like the problem is related to SSL certificates:

> <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1076)

Could you please give a couple extra details:

1. What do `which python` and `which python3` report?
2. Your version of macOS
3. How did you install python3 (Homebrew, macports, python.org installer, something else?)

igrr, 2020-02-16:
Also, until we figure out the cause of the issue, there is a workaround which should let you continue: copy the URL printed in the console into your web browser, and download the file into the `dist` directory. Then run the install script again. You may have to repeat this a few times as there are multiple archives to be downloaded. But once all archives are downloaded, the install script should be able to continue.

mattkefford, 2020-02-16:
Hi Ivan thanks for the help!
1. 
Keffords-Mac:esp-idf Kefford$ which python
/Library/Frameworks/Python.framework/Versions/2.7/bin/python
Keffords-Mac:esp-idf Kefford$ which python3
/Library/Frameworks/Python.framework/Versions/3.7/bin/python3

2. MacOS Mojave Version 10.14.6

3. I believe I installed it from the https://www.python.org/downloads/ installer. When I do 'brew list' it doesn't show python in there.

Regarding the ".espressif" folder; I couldn't create it with that name as it said "You can’t use a name that begins with a dot because these names are reserved for the system. Please choose another name." So will I be ok continuing with the "espressif" folder for your workaround?

igrr, 2020-02-16:
> You can’t use a name that begins with a dot because these names are reserved for the system

You can still do this in Terminal: run `mkdir -p ~/.espressif` and `mkdir -p ~/.espressif/dist`. Although continuing with `espressif` is also okay, as long as IDF_TOOLS_PATH is set consistently. I'd probably recommend the default (starting with a dot) since it should work better if you try to use Eclipse or VS Code IDF plugins later.

igrr, 2020-02-16:
By the way, this article looks like it addresses the certificate problem with python.org python installations: https://www.dev2qa.com/how-to-fix-python-error-certificate-verify-failed-unable-to-get-local-issuer-certificate-in-mac-os/. Can you please try that?

mattkefford, 2020-02-16:
OK I will try that. 
Just FYI when I created the .espressif folder I went to Finder to view it, did cmd+shift+(dot) to show hidden folders and it seems it was already there, with contents too:
[image]

mattkefford, 2020-02-16:
That appears to have fixed the download issue :)
[image]

I will now fix the path and restart the download to the right folder, thanks!

mattkefford, 2020-02-16:
Hi, just wanted to add that after this step I get a message saying that python requirements are not satisfied and that I should follow step 3 of Getting Started. However there are no instructions on updating python dependencies. I googled and found on the esp32 forum that I needed to run the command "python -m pip install --user -r $IDF_PATH/requirements.txt". I think it would be a good idea to add this to the getting started guide, otherwise the user would end up trying ./install.sh again and going round in circles.

[image]

Thanks again :)

closed 2020-02-16 by mattkefford

krzychb, 2020-04-26:
Linking similar issue #4081

ingconti, 2021-03-22:
my 2 cents: for bigger 
https://www.dev2qa.com/how-to-fix-python-error-certificate-verify-failed-unable-to-get-local-issuer-certificate-in-mac-os/. 
did the job!

TobyBorland, 2021-11-25:
Same error under a fresh Ubuntu 20.04 LTS installation,

WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1131)>

Resolved via,

$ sudo update-ca-certificates --fresh
$ export SSL_CERT_DIR=/etc/ssl/certs

FalconBFR, 2021-12-11:
> https://www.dev2qa.com/how-to-fix-python-error-certificate-verify-failed-unable-to-get-local-issuer-certificate-in-mac-os/

Fixed with this. This is the solution.

quark-modularity, 2022-02-04:
Hello, 
Thanks for the explanation. Is still working and resolving the issue. M1 MacBook Air, Python 3.10

ErickGiffoni, 2022-04-22:
I've just had the same issue in macOS Big Sur 11.5.2.

Thanks for the help !

> By the way, this article looks like it addresses the certificate problem with python.org python installations: https://www.dev2qa.com/how-to-fix-python-error-certificate-verify-failed-unable-to-get-local-issuer-certificate-in-mac-os/. Can you please try that?

That article solved for me !

ArturoHaber, 2022-06-05:
> my 2 cents: for bigger https://www.dev2qa.com/how-to-fix-python-error-certificate-verify-failed-unable-to-get-local-issuer-certificate-in-mac-os/. did the job!

Thank you... This is the official fix for M1 Mac

SiboVG, 2022-07-18:
> By the way, this article looks like it addresses the certificate problem with python.org python installations: https://www.dev2qa.com/how-to-fix-python-error-certificate-verify-failed-unable-to-get-local-issuer-certificate-in-mac-os/. Can you please try that?

Just want to add a side note: this initially didn't work for me because running `Install Certificates.command` by double-clicking it didn't run the command with the correct permissions. So I had to open terminal and execute the command `sudo <path to Install Certificates.command>`.

After that, everything worked perfectly :)

Tangerino, 2022-11-01:
Fix the problem in Venturas as well (https://www.dev2qa.com/how-to-fix-python-error-certificate-verify-failed-unable-to-get-local-issuer-certificate-in-mac-os/)

ssvolans, 2023-04-11:
I believe working on the esp32 idf is very worst it took a lot of time developers to fix issues of the esp32 installation.

babldev, 2024-04-08:
For me above steps did not work to resolve the certificate issues on macOS 14 Sonoma, but `brew reinstall openssl` did.

https://github.com/Homebrew/homebrew-core/issues/42198#issuecomment-564181886

awayings, 2025-01-21:
for my old mac 11, I have to install openssl by compiling. I need to copy original mac's pem file (/private/etc/ssl/cert.pem) to new openssl config's location. then I can run the Install Certificates.command .