espressif/esp-idf #4776: ULP instruction REG_WR description

boarchuz, 2020-02-17:
https://github.com/espressif/esp-idf/blob/master/docs/en/api-guides/ulp_instruction_set.rst#reg_wr--write-to-peripheral-register
The description for REG_WR is currently:
"The instruction writes up to 8 bits **from a general purpose register** into a peripheral register."

However, it is (unfortunately) impossible to write to a peripheral register from a general purpose register. It should convey something more like  "... up to 8 bits from [the provided]/[an immediate] value ... " instead.

boarchuz, 2020-02-17:
This is also in the TRM. I realise that this is probably not the right place for documentation feedback but it was unclear how to report this through that link.

krzychb, 2020-02-17:
Hi @boarchuz, thank you for reporting this issue!

closed 2020-03-02 by espressif-bot