espressif/esp-idf #4778: esp_wifi_connect 1150 wifi not start - ESP32 wifi init seems to fail.

gecko242, 2020-02-17:
- [ X ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ X ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ X ] I have searched the issue tracker for a similar issue and not found a similar issue.

**Environment**

- Custom board, using an ESP32-WROOM-32U 16MB Module.

- IDF Version : v4.1-dev-1935-g647cb628a

- Using idf.py on Windows 10, CMD 

- Compiler Version : esp-2019r2 8.2.0

- Using VScode, compiling and monitoring externally.

- Power supply is bomber - Custom board with 5A switchmode

**Preface:**
This appears to be some sort of memory corruption / overflow issue. This issue does not always manifest, and changing some of the code in a completely unrelated location will cause it to go away, hence I think another piece of code is interfering with the wifi init stuff. 

**The Issue:**
Sometimes on boot I am met with:

`
E (1092) wifi: esp_wifi_connect 1150 wifi not start
`

Repeatedly followed by:
`
W (2242) wifi: Haven't to connect to a suitable AP now!
`

Here is my wifi init code:

```cpp

    wifi_config_t wifi_ap_config = {0};
    wifi_config_t wifi_sta_config = {0};

    wifiEventState = xEventGroupCreate();
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_netif_create_default_wifi_ap();
    esp_netif_create_default_wifi_sta();

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &wifi_event_handler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &wifi_event_handler, NULL));

    

    wifi_ap_config.ap.max_connection = MAX_AP_CONNECTIONS;
      
    sprintf(wifiNameBuffer, "productName- %05d", configuration.serialNumber); // 5 digit serial number, with leading zeros

    wifiNameLen = strlen(wifiNameBuffer);
    strncpy((char *)wifi_ap_config.ap.ssid, wifiNameBuffer, wifiNameLen);
    wifi_ap_config.ap.ssid_len = wifiNameLen;

    

    strncpy((char*)wifi_sta_config.sta.ssid, "SSID", sizeof(wifi_sta_config.sta.ssid));
    strncpy((char*)wifi_sta_config.sta.password, "PSK",sizeof(wifi_sta_config.sta.password));
    ESP_LOGI(TAG, "Trying to connect to %s with %s", wifi_sta_config.sta.ssid, wifi_sta_config.sta.password);

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_ap_config));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_sta_config));

    ESP_ERROR_CHECK(esp_wifi_start());

```

Like I said, this works perfectly sometimes, with no errors. I can change a completely unrelated portion of the code, and it will reliably fail. I feel that this must be something small, but I have been struggling with this for a few days now. There is no other WiFi stuff going on in my code, and this does not produce a crash, just the error described above.

zhangyanjiaoesp, 2020-02-21:
@gecko242 

The log `E (1092) wifi: esp_wifi_connect 1150 wifi not start` shows that you called `esp_wifi_connect()` before `esp_wifi_start()`.

The log `W (2242) wifi: Haven't to connect to a suitable AP now!` shows that you called the function `esp_wifi_get_ap_info()` before you had connected to an AP.

gecko242, 2020-02-21:
@zhangyanjiaoesp 
I have been doing a bit more digging on this. Your second point is correct, that was an error in my code. 

The first point is not quite that simple. I am calling `esp_wifi_connect()` from the event handler, on the following event:

`(event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START)`

You would expect that you can call `esp_wifi_connect()` straight after that event is fired, but that does not appear to be the case. 

Connecting straight away works most of the time, but not always. Delaying for a short time after a failed connect, and trying again, always seems to work.

liuzfesp, 2020-03-24:
Hi @gecko242 , 

**Sometimes on boot I am met with:

E (1092) wifi: esp_wifi_connect 1150 wifi not start**

I guess the log is displayed in following scenario:

- STA is connected to an AP
- Reboot the ESP32 (e.g. the application calls `esp_restart()` to restart the ESP32 etc)
- The error log should be displayed **before** the ESP32 is rebooted, the error log is NOT displayed **after** the ESP32 is rebooted

If my above suspicion is correct, then it's not a real bug, it's just a log. Because when you restart the ESP32 with `esp_restart()`:

 - `esp_restart()` call `esp_wifi_stop()` first, it stops WiFi and generates `WIFI_EVENT_STA_DISCONNECTED' event
 - on receiving the disconnect event, application call `esp_wifi_connect()` to reconnect the WiFi, but at that moment, the WiFi has already been stopped, then the error log is displayed.

To avoid the error log, the application should NOT call `esp_wifi_connect()` if the `WIFI_EVENT_STA_DISCONNECTED` is triggered because of WiFi stop. Anyway, you can also just ignore the log.

gecko242, 2020-03-24:
@liuzfesp 
Interesting, your explanation makes perfect sense. However I experience this problem after a hardware reset, often after flashing. I dont call `esp_restart()` anywhere in my code.

Here is the code in question:
`
    if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START) // connect on wifi start
    {
        xEventGroupSetBits(wifiEventState, WIFI_START_BIT);
        vTaskDelay(20 / portTICK_RATE_MS); // wait a bit, then try and connect. 
        if(esp_wifi_connect() != ESP_OK)
        {
            ESP_LOGE(TAG, "WIFI CONNECT FAILED"); // sometimes wifi connection seems to fail if initiated too soon. 
            vTaskDelay(1000 / portTICK_RATE_MS);    // if this is the case, wait a bit and try again. 
            esp_wifi_connect();
        }
    }
`

Occasionally after a hardware reset, the first `esp_wifi_connect()` fails. The second `esp_wifi_connect()` always succeeds.

liuzfesp, 2020-04-15:
Hi @gecko242 it will be helpful if you can provide whole log about this issue.

liuzfesp, 2020-04-30:
Hi @gecko242 can you provide the whole log?

gecko242, 2020-04-30:
@liuzfesp Unfortunately I am having a difficult time recreating the issue at the moment. As I mentioned, I think  this is very timing specific. I dont have time to directly investigate the issue at the moment, but if I see it crop up again, I will be sure to post the log here.

Alvin1Zhang, 2020-07-08:
@gecko242 Thanks for reporting, do you have any time slot for investigating/reproducing the issue? Would you please help share if any updates for the issue? Thanks.

closed 2020-07-13 by gecko242

Alvin1Zhang, 2020-07-14:
@gecko242 Thanks for sharing the updates.