espressif/esp-idf #4780: ESP_OTA: OTA image has invalid magic byte

tharunjoy, 2020-02-17:
## Environment

- Development Kit:     ESP32-DevKitC
- Module or chip used:  ESP32-WROOM-32
- IDF version:  v3.3-dev-270-g6f8e2b018
- Build System:        5.2.0
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):  crosstool-ng-1.22.0-80-g6c4433a
- Operating System:     Linux
- Using an IDE?:   Yes
- Power Supply:         USB
## Problem Description

//Detailed problem description goes here.
I am trying to implement the OTA update.
I ran an OpenSSL server locally and I was able to do OTA update smoothly. Then I tried deploying the same application firmware in onedrive, google drive and even in dropbox where I am getting some error as shown below. 

`I (641674) esp_https_ota: Starting OTA...
I (641674) esp_https_ota: Writing to partition subtype 16 at offset 0x170000
I (644734) esp_https_ota: esp_ota_begin succeeded
I (644744) esp_https_ota: Please Wait. This may take time
E (644744) esp_ota_ops: OTA image has invalid magic byte (expected 0xE9, saw 0x3c
E (644754) esp_https_ota: Error: esp_ota_write failed! err=0x0
E (644754) OTA_UPDATE :::: Firmware Upgrades Failed
`
I am using a custom partation_table is as below 
`
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,200K,
otadata,data,ota,0x3b000,8K,
phy_init,data,phy,0x3d000,4K,
factory,app,factory,0x40000,1200K,
ota_0,app,ota_0,0x170000,1200K,
ota_1,app,ota_1,0x2a0000,1200K,
`
Size of the image is 
`
Total sizes:
 DRAM .data size:   12976 bytes
 DRAM .bss  size:   68976 bytes
Used static DRAM:   81952 bytes (  42628 available, 65.8% used)
Used static IRAM:   77172 bytes (  53900 available, 58.9% used)
      Flash code:  658886 bytes
    Flash rodata:  153652 bytes
Total image size:~ 902686 bytes (.bin may be padded larger)
`

Why I am getting this issue and how to resolve this?

Thanks in advance.

chegewara, 2020-02-17:
`has invalid magic byte (expected 0xE9, saw 0x3c`
`<      60 0074 0x3c`
You can safely assume that response is HTML page with error code like error 404 or 500.

tharunjoy, 2020-02-17:
Is there a way I can verify that inside esp IDF, by checking the memory or finding the error code, etc.
I am able to download that from the browser using the same download link and I was able to flash it successfully.

chegewara, 2020-02-17:
Sure, you can log to serial. Here is code from native ota example i am using in app:
```
while (1) {
        int data_read = esp_http_client_read(client, ota_write_data, BUFFSIZE);
        if (data_read < 0) {
            ESP_LOGE(TAG, "Error: SSL data read error");
            http_cleanup(client);
            task_fatal_error();
        } else if (data_read > 0) {
             <------ ADD LOGS HERE to print data_read
            err = esp_ota_write( update_handle, (const void *)ota_write_data, data_read);
            if (err != ESP_OK) {
                http_cleanup(client);
                task_fatal_error();
            }
            binary_file_length += data_read;
            ESP_LOGD(TAG, "Written image length %d", binary_file_length);
        } else if (data_read == 0) {
            ESP_LOGI(TAG, "Connection closed,all data received");
            break;
        }
    }
```
Or even better, you could check `Content-Type: ` header (still part of example):
`esp_http_client_fetch_headers(client);`

shubhamkulkarni97, 2020-02-18:
@tharunjoy Which OTA Example are you using?

tharunjoy, 2020-02-18:
@shubhamkulkarni97 I am using simple_ota_example.

mahavirj, 2020-02-19:
@tharunjoy If you go to stable `v3.3` tag, https://github.com/espressif/esp-idf/tree/v3.3.1, do you still observe same issue? Just would like to understand if this is not a regression due to some of the recent changes that went in `esp_https_ota` component.

tharunjoy, 2020-02-19:
Thank you guys for your support, I was able to rectify and resolve the issue.
My inference is, curl and esp are not able to redirect the URL automatically just like a web browser. I found the redirected URL and was able to successfully do OTA update.

To know more, just goto the esp-idf forum
https://www.esp32.com/viewtopic.php?f=13&t=14299&p=55721#p55721

shubhamkulkarni97, 2020-02-25:
@tharunjoy URL redirection is supported in ESP-IDF. A intermediate commit causes regression in case URL is redirected.

I have attached a patch which fixes the issue. Can you please try this patch?

Let me know if this patch works for you.
[Fix-for-URL-redirection.patch.zip](https://github.com/espressif/esp-idf/files/4248172/Fix-for-URL-redirection.patch.zip)

closed 2020-02-26 by espressif-bot