espressif/esp-idf #4781: HTTP server BackEnd Authentication AP mode

rebeliousconformist, 2020-02-17:
In the example from 
https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/restful_server

I need an Authenticaion backend on the ESP-IDF side where all User requests from Vue.JS can processed. 

Current authentications are limited to the HTTP client mode the ESP32. Is there a possiblity to implement them us the HTTP server mode ?

Regards

Alan5142, 2020-02-20:
You can read the authentication header in the esp32, refer to [this example](https://github.com/espressif/esp-idf/blob/f91080637c054fa2b4107192719075d237ecc3ec/examples/protocols/http_server/simple/main/main.c)  to know how you can read headers

rebeliousconformist, 2020-02-27:
Okay. But then how do I maintain user access integrity across all the pages ? You can't expect the sure to enter a username and password for every weboage

Alan5142, 2020-03-09:
Well, thats the point of handling users in the backend, one way to solve this problem is to generate a token in the esp32 and send that token from the client to the esp32 in every request that needs authentication, take a look to [jwt](https://jwt.io/)

mahavirj, 2021-06-01:
Basic authentication support was added in HTTP server example sometimes back, please refer to https://github.com/espressif/esp-idf/commit/94512a76b763acb2c7f4edfaf77416efeb1e6e58. This may help to build on further use-case specified here.

closed 2021-06-01 by mahavirj