espressif/esp-idf #4783: ImportError: No module named tiny_test_fw

PranavD10, 2020-02-17:
Hi,
I am working on the simple OTA example in the IDF, And currently facing this issue.

Alvin1Zhang, 2020-02-18:
@PranavD10 Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

PranavD10, 2020-02-18:
Hi,
Thanks for replying. I just want to run the python script given in the simple OTA example. I dont have any issue with the ESP32 or the firmware compiled for that. I am using the latest idf branch in my project, Ubuntu version 18.04 and python 2.7. Here is the command and the output i got for that.

>> python example_test.py 
Traceback (most recent call last):
  File "example_test.py", line 9, in <module>
    from tiny_test_fw import DUT
ImportError: No module named tiny_test_fw

Do anyone knows why i got this error. I used the export.sh file to import the paths.

dobairoland, 2020-02-18:
Scripts in the example directory are for internal testing only and they are intended to be run in special testing environment. 

`export PYTHONPATH="$IDF_PATH/tools:$IDF_PATH/tools/ci/python_packages:$PYTHONPATH"` should solve the above issue but you may experience other failures, like Python packages not installed: https://github.com/espressif/esp-idf/blob/master/tools/ci/python_packages/tiny_test_fw/requirements.txt

closed 2020-02-18 by dobairoland

MaazSk, 2021-11-02:
I was also trying an OTA example and I am facing the below issue

`import ttfw_idf
ModuleNotFoundError: No module named 'ttfw_idf'`

I tried export PYTHONPATH="$IDF_PATH/tools:$IDF_PATH/tools/ci/python_packages:$PYTHONPATH" but still I am facing the same issue

hdpklm, 2021-11-29:
separator between lines should be semicolumn
```
export PYTHONPATH="$IDF_PATH/tools;$IDF_PATH/tools/ci/python_packages;$PYTHONPATH"
```

hfudev, 2021-12-10:
it is os independent. for windows it should be `;`, for linux it should be `:`

if you want to check it by yourself, you can run

```
import os
print(os.pathsep)
```

(cc @MaazSk @hdpklm)

MaazSk, 2021-12-14:
@hfudev, 

Thank you for your reply

stoymengermanov98, 2021-12-22:
Hi , I am working on ESP32 EVB on Simple HTTP-server example. I am trying to run a client.py script in same way (writing in terminal : python client.py)    . . ., but there is same problem. I receive the following message : 
Traceback (most recent call last):
  File "client.py", line 23, in <module>
    from tiny_test_fw import Utility
ModuleNotFoundError: No module named 'tiny_test_fw'

I tried export PYTHONPATH="$IDF_PATH/tools:$IDF_PATH/tools/ci/python_packages:$PYTHONPATH" but still I am facing the same issue , I also replaced the separators with ; but still nothing . Please, give me an advice what to do.

MaazSk, 2021-12-22:
@stoymengermanov98, I guess this [link](https://github.com/espressif/esp-idf/issues/7815#issuecomment-958691020) may help you.