espressif/esp-idf #4785: Esp-idf v4.0 ulp example issue

devUsama, 2020-02-17:
Iam trying to build the ulp example of given in esp-idf v4.0, its giving me the error:
**fatal error: ulp_main.h: No such file or directory**
[image]
Documentation explained that binutils-esp32ulp toolchain should be installed during ESP-IDF setup with CMake build system and Build system generate ${ULP_APP_NAME}.h and ${ULP_APP_NAME}.ld files, but in my case these files are not generated. Also when i build my code it open esp32ulp_mapgen.py file

igrr, 2020-02-17:
Thanks for reporting, will fix this soon.

devUsama, 2020-02-27:
Any updates on this issue?

igrr, 2020-02-27:
Hi @devUsama, sorry for the delay. The fix has been merged into release/v4.0 branch internally, and will soon show up on Github as commit a8fe083017057b6c2dbd76ef5fd060b16ff06790.

In the meantime, you can apply the attached patch locally.

[0001-cmake-Set-uninitialized-variable-warnings-in-ULP-boo.patch.zip](https://github.com/espressif/esp-idf/files/4260303/0001-cmake-Set-uninitialized-variable-warnings-in-ULP-boo.patch.zip)

devUsama, 2020-02-27:
Thanks for the update, i'll check it out!

devUsama, 2020-02-28:
Build successfully now, closing the issue!

closed 2020-02-28 by devUsama