espressif/esp-idf #4787: ESP_EVENT_ANY_ID - invalid conversion from 'int' to 'esp_mqtt_event_id_t'

klaassd, 2020-02-17:
This error occurs when compiling with C++, in plain C it works fine.

```
/opt/esp/idf/components/esp_event/include/esp_event_base.h:37:32: error: invalid conversion from 'int' to 'esp_mqtt_event_id_t' [-fpermissive]
 #define ESP_EVENT_ANY_ID       -1               /**< register handler for any event id */
                                ^~
../main/MQTT_Client.cpp:31:47: note: in expansion of macro 'ESP_EVENT_ANY_ID'
  esp_mqtt_client_register_event(this->client, ESP_EVENT_ANY_ID, mqtt_event_handler, this->client);
                                               ^~~~~~~~~~~~~~~~
```
Also explicit casting is not working, because -1 is not in the range of the enum.

0xjakob, 2020-02-19:
Hi @klaassd, thanks for posting this issue! Could you please tell us more about your environment? In particular the IDF version and code that leads to the issue would be helpful.

0xjakob, 2020-02-19:
OK, no need to post the environment, it's definitely an incompatibility between the MQTT event types and general event types. We're gonna have a look at that.

klaassd, 2020-02-19:
Hi,
I use espressif/idf docker container. Since I checked out the latest version I use the master branch from idf. In detail this is commit 93a8603c545fb8e54741d6685146e2f3b874378d from Nov 18th 2019. Sorry, that I can't tell a specific version.

This is my constructor causing the error:
```
WifiConnection::WifiConnection(wifi_config_t* config) :
	RtosThread(taskName, taskStackDepth, taskPrio),
	s_retry_num(0)
{
    s_wifi_event_group = xEventGroupCreate();

    esp_netif_init();

    ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_netif_create_default_wifi_sta();

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &wifi_event_handler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &wifi_event_handler, NULL));

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, config) );
}
```

klaassd, 2020-02-19:
Hi @Oxjakob, sorry didn't refresh the page before answering. Thanks for having a look at it!

closed 2020-03-04 by espressif-bot