espressif/esp-idf #4788: install.sh: problems with python2

szutenberg, 2020-02-17:
install.sh 

```
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, openocd-esp32
Installing xtensa-esp32-elf@esp-2019r2-8.2.0
Downloading xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz to /tue/esp/idf_tools_path/dist/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz.tmp
Done
Extracting /tue/esp/idf_tools_path/dist/xtensa-esp32-elf-gcc8_2_0-esp-2019r2-linux-amd64.tar.gz to /tue/esp/idf_tools_path/tools/xtensa-esp32-elf/esp-2019r2-8.2.0
Installing esp32ulp-elf@2.28.51.20170517
Downloading binutils-esp32ulp-linux64-2.28.51-esp32ulp-20180809.tar.gz to /tue/esp/idf_tools_path/dist/binutils-esp32ulp-linux64-2.28.51-esp32ulp-20180809.tar.gz.tmp
Done
Extracting /tue/esp/idf_tools_path/dist/binutils-esp32ulp-linux64-2.28.51-esp32ulp-20180809.tar.gz to /tue/esp/idf_tools_path/tools/esp32ulp-elf/2.28.51.20170517
Installing openocd-esp32@v0.10.0-esp32-20190313
Downloading openocd-esp32-linux64-0.10.0-esp32-20190313.tar.gz to /tue/esp/idf_tools_path/dist/openocd-esp32-linux64-0.10.0-esp32-20190313.tar.gz.tmp
Done
Extracting /tue/esp/idf_tools_path/dist/openocd-esp32-linux64-0.10.0-esp32-20190313.tar.gz to /tue/esp/idf_tools_path/tools/openocd-esp32/v0.10.0-esp32-20190313
Installing Python environment and packages
Creating a new Python environment in /tue/esp/idf_tools_path/python_env/idf4.0_py2.7_env
usage: virtualenv [--version] [--with-traceback] [-v | -q] [--discovery {builtin}] [-p py] [--creator {builtin,cpython2-posix}] [--seeder {app-data,pip}] [--no-seed] [--activators comma_separated_list] [--clear] [--system-site-packages]
                  [--symlinks | --copies] [--download | --no-download] [--extra-search-dir d [d ...]] [--pip version] [--setuptools version] [--wheel version] [--no-pip] [--no-setuptools] [--no-wheel] [--clear-app-data] [--symlink-app-data]
                  [--prompt prompt] [-h]
                  dest
virtualenv: error: unrecognized arguments: --no-site-packages
Traceback (most recent call last):
  File "/tue/esp/esp-idf/tools/idf_tools.py", line 1372, in <module>
    main(sys.argv[1:])
  File "/tue/esp/esp-idf/tools/idf_tools.py", line 1368, in main
    action_func(args)
  File "/tue/esp/esp-idf/tools/idf_tools.py", line 1184, in action_install_python_env
    stdout=sys.stdout, stderr=sys.stderr)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python', '-m', 'virtualenv', '--no-site-packages', '/tue/esp/idf_tools_path/python_env/idf4.0_py2.7_env']' returned non-zero exit status 2
```

Reason: python --version =2.7.17

I'd recommend to run python scripts in install.sh via "python3".

igrr, 2020-02-17:
Hi @szutenberg, the cause of this issue is the deprecated --no-site-packages argument which idf_tools.py was passing to `virtualenv`. This happens when virtualenv package installed is later than v20.0.0. In your specific case, switching to python3 might be solving the issue if the virtualenv package installed in py3 environment is of an older version. The issue was first raised in https://github.com/espressif/esp-idf/issues/4744 and is now fixed in master and most release branches. In release/v4.0 branch which you are using, it was fixed in commit https://github.com/espressif/esp-idf/commit/8cb3cc515477e777144acb3b58f5156c4ad69124. Please update to the latest release/v4.0 branch and the issue should be fixed. The fix will be part of the next bugfix release (v4.0.1).

closed 2020-02-17 by igrr

ozarchie, 2020-04-16:
I have tried the instller, selected the v4 idf, upgrade to python3.7 and upgrade git.
Yet still this gives the "--no-site-packages argument" fault.

Not sure that this is really fixed"

```
C:\Users\Anne\esp\esp-idf>install
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, ninja, idf-exe, ccache
Installing xtensa-esp32-elf@esp-2019r2-8.2.0
file xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip is already downloaded
Extracting C:\Users\Anne\dist\xtensa-esp32-elf-gcc8_2_0-esp-2019r2-win32.zip to C:\Users\Anne\tools\xtensa-esp32-elf\esp-2019r2-8.2.0
Installing esp32ulp-elf@2.28.51.20170517
file binutils-esp32ulp-win32-2.28.51-esp32ulp-20180809.zip is already downloaded
Extracting C:\Users\Anne\dist\binutils-esp32ulp-win32-2.28.51-esp32ulp-20180809.zip to C:\Users\Anne\tools\esp32ulp-elf\2.28.51.20170517
Installing cmake@3.13.4
file cmake-3.13.4-win64-x64.zip is already downloaded
Extracting C:\Users\Anne\dist\cmake-3.13.4-win64-x64.zip to C:\Users\Anne\tools\cmake\3.13.4
Installing openocd-esp32@v0.10.0-esp32-20190313
file openocd-esp32-win32-0.10.0-esp32-20190313.zip is already downloaded
Extracting C:\Users\Anne\dist\openocd-esp32-win32-0.10.0-esp32-20190313.zip to C:\Users\Anne\tools\openocd-esp32\v0.10.0-esp32-20190313
Installing mconf@v4.6.0.0-idf-20190628
file mconf-v4.6.0.0-idf-20190628-win32.zip is already downloaded
Extracting C:\Users\Anne\dist\mconf-v4.6.0.0-idf-20190628-win32.zip to C:\Users\Anne\tools\mconf\v4.6.0.0-idf-20190628
Installing ninja@1.9.0
file ninja-1.9.0-win64.zip is already downloaded
Extracting C:\Users\Anne\dist\ninja-1.9.0-win64.zip to C:\Users\Anne\tools\ninja\1.9.0
Installing idf-exe@1.0.1
file idf-exe-v1.0.1.zip is already downloaded
Extracting C:\Users\Anne\dist\idf-exe-v1.0.1.zip to C:\Users\Anne\tools\idf-exe\1.0.1
Installing ccache@3.7
file ccache-3.7-w64.zip is already downloaded
Extracting C:\Users\Anne\dist\ccache-3.7-w64.zip to C:\Users\Anne\tools\ccache\3.7
Setting up Python environment
Creating a new Python environment in C:\Users\Anne\python_env\idf4.0_py3.7_env
usage: virtualenv [--version] [--with-traceback] [-v | -q] [--app-data APP_DATA] [--clear-app-data] [--discovery {builtin}] [-p py] [--creator {builtin,cpython3-win,venv}] [--seeder {app-data,pip}] [--no-seed] [--activators comma_sep_list]
                  [--clear] [--system-site-packages] [--copies] [--download | --no-download] [--extra-search-dir d [d ...]] [--pip version] [--setuptools version] [--wheel version] [--no-pip] [--no-setuptools] [--no-wheel]
                  [--symlink-app-data] [--prompt prompt] [-h]
                  dest
virtualenv: error: unrecognized arguments: --no-site-packages
Traceback (most recent call last):
  File "C:\Users\Anne\esp\esp-idf\tools\idf_tools.py", line 1372, in <module>
    main(sys.argv[1:])
  File "C:\Users\Anne\esp\esp-idf\tools\idf_tools.py", line 1368, in main
    action_func(args)
  File "C:\Users\Anne\esp\esp-idf\tools\idf_tools.py", line 1184, in action_install_python_env
    stdout=sys.stdout, stderr=sys.stderr)
  File "C:\Users\Anne\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['C:\\Users\\Anne\\AppData\\Local\\Programs\\Python\\Python37\\python.exe', '-m', 'virtualenv', '--no-site-packages', 'C:\\Users\\Anne\\python_env\\idf4.0_py3.7_env']' returned non-zero exit status 2.

C:\Users\Anne\esp\esp-idf>python -V
Python 3.7.3
```

igrr, 2020-04-16:
@ozarchie could you please check IDF version `git describe --tags` and post it here? Perhaps your branch is older than commit 8cb3cc5?

ozarchie, 2020-04-16:
Hi Ivan,
I downloaded esp-idf-tools-setup-2.3.exe and installed from that exe.

I hope this is what you are looking for:
C:\Users\Anne\esp\esp-idf>git describe --tags
v4.0
C:\Users\Anne\esp\esp-idf
(I am not sure which git repository you are asking about but assumed it was the esp_idf.)

I finally stepped through the manual install, and deleted the "--no-site-packages" argument in the script.
It installed and is working now.
Cheers
John

igrr, 2020-04-16:
Right, looks like you are using v4.0 tagged version, which has this bug. Removing the --no-site-packages argument manually, like you did, or pulling the latest release/v4.0 branch, are both valid solutions. The next bugfix release, v4.0.1, will include the fix as well.

manoj153, 2020-04-16:
Thanks were having this -no-site-packages issue as well on my ubuntu 18, is there any pitfall if I go to release/v4.0 branch ? are the codes there committed daily that could potentially break my code?

igrr, 2020-04-16:
@manoj153 The tradeoffs associated with tags/branches are explained here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#which-version-should-i-start-with. In short, yes, there is some risk, but don't expect any major changes on a release branch. If you don't feel confident switching to the release branch, you can also cherry-pick the fix on top of v4.0 tag (`git cherry-pick 8cb3cc5`).

dipen-1337lab, 2020-04-30:
Relieved to find this link on the issue. Spent almost a week trying to debug the issue.

Thank you guys!

**Removing the --no-site-packages argument manually** worked for me! Below is the log at the end of the installation.

...
...
Adding ESP-IDF tools to PATH...
    C:\Users\Dipen Shah\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin
    C:\Users\Dipen Shah\.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin
    C:\Users\Dipen Shah\.espressif\tools\cmake\3.13.4\bin
    C:\Users\Dipen Shah\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin
    C:\Users\Dipen Shah\.espressif\tools\mconf\v4.6.0.0-idf-20190628\
    C:\Users\Dipen Shah\.espressif\tools\ninja\1.9.0\
    C:\Users\Dipen Shah\.espressif\tools\idf-exe\1.0.1\
    C:\Users\Dipen Shah\.espressif\tools\ccache\3.7\
    C:\Users\Dipen Shah\.espressif\python_env\idf4.0_py2.7_env\Scripts
    G:\gitHub\gitHubLibraries\espressif\esp-idf\tools

Checking if Python packages are up to date...
Python requirements from G:\gitHub\gitHubLibraries\espressif\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

manoj153, 2020-05-02:
> @manoj153 The tradeoffs associated with tags/branches are explained here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#which-version-should-i-start-with. In short, yes, [...]

Thanks, that was informative