espressif/esp-idf #4790: Docker image rebuilds all every time, doesn't do incremental build

ftab, 2020-02-17:
## Environment

- Development Kit:      ESP32-LyraT
- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it): v3.3.1 docker image - espressif/idf:v3.3.1
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB
- Docker version: 18.09.9

## Problem Description

When building an application (e.g. hello world example) it does not do an incremental build on future runs.

### Expected Behavior

Incremental build, future runs should not have to rebuild if the source did not change

### Actual Behavior

Full project is built every time.

### Steps to reproduce

1. Plug ESP32 board into /dev/ttyUSB0
2. Install docker-ce (18.09.9 in my case)
3. Check out esp-idf at v3.3.1
4. cd to examples/getting-started/hello_world or cp -a examples/getting-started/hello_world elsewhere and then cd to there
5. Set up default config: `docker run -it -v $PWD:/project -w /project --device=/dev/ttyUSB0 espressif/idf:v3.3.1 make defconfig`
6. Build, flash, monitor: `docker run -it -v $PWD:/project -w /project --device=/dev/ttyUSB0 espressif/idf:v3.3.1 make -j flash monitor`
Make sure to do the boot button dance if you do not have the jumpers set for automatic boot/flashing
7. Ctrl+] to exit after satisfied it is running
8. Run `docker run -it -v $PWD:/project -w /project --device=/dev/ttyUSB0 espressif/idf:v3.3.1 make monitor`

It tries to rebuild the whole project (all components) again.

### Code to reproduce this issue

https://github.com/espressif/esp-idf/tree/v3.3.1/examples/get-started/hello_world

## Debug Logs

The code runs fine, it is the build process that does not reuse the existing objects. A full rebuild is done every time, and that takes quite some time on some computers.

```
$ docker run -it -v $PWD:/project -w /project --device=/dev/ttyUSB0 espressif/idf:v3.3.1 make -j flash monitor
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Python requirements from /opt/esp/idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /opt/esp/idf/components/esptool_py/esptool
  /opt/esp/idf/components/espcoredump
  /opt/esp/idf/components/partition_table/
  /opt/esp/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a-5.2.0/xtensa-esp32-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin
  /opt/esp/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/bin
  /opt/esp/python_env/idf3.3_py3.6_env/bin
  /opt/esp/idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Toolchain path: /opt/esp/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
make[1]: Entering directory '/opt/esp/idf/tools/kconfig'
cc -c  -DCURSES_LOC="<ncurses.h>" -DLOCALE -MMD -MP  /opt/esp/idf/tools/kconfig/mconf.c -o mconf.o
[... 769 lines cut ...]
I (194) cpu_start: Application information:
I (199) cpu_start: Project name:     hello-world
I (205) cpu_start: App version:      1
I (209) cpu_start: Compile time:     Feb 17 2020 20:10:52
I (215) cpu_start: ELF file SHA256:  9135d5b140877d2f...
I (221) cpu_start: ESP-IDF:          v3.3.1
I (226) cpu_start: Starting app cpu, entry point is 0x40081228
0x40081228: call_start_cpu1 at /opt/esp/idf/components/esp32/cpu_start.c:269

I (0) cpu_start: App cpu up.
I (236) heap_init: Initializing. RAM available for dynamic allocation:
I (243) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (249) heap_init: At 3FFB2EC0 len 0002D140 (180 KiB): DRAM
I (255) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (262) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (268) heap_init: At 40088404 len 00017BFC (94 KiB): IRAM
I (274) cpu_start: Pro cpu start user code
I (293) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Hello world!
This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external flash
Restarting in 10 seconds...
Restarting in 9 seconds...
Restarting in 8 seconds...
```

igrr, 2020-02-17:
Unfortunately this is related to the fact that in Make build system, Kconfig tools are built in the ESP-IDF directory, not in the project build directory. Therefore the `conf` binary gets rebuilt every time you start the build in a container. Since sdkconfig and sdkconfig.h are its dependencies, they get rebuilt as well, and because of them, all the project sources get rebuilt.

This problem (kconfig tools built in IDF directory) is solved in the CMake build system. If you are starting a new project, I would recommend starting with CMake.

If that is not an option, you can start a named container (and without the --rm flag), and then execute the build in the running container using `docker exec`.

ftab, 2020-02-17:
@igrr Thanks so much for the quick response!

I'd love to use CMake instead, but it looks like ESP-ADF hasn't been updated for it (or at least some examples aren't).

I have esp-idf and esp-adf natively on my main machine so I've tried that before messing with the Docker image further, but I get errors trying to do idf.py build on esp-adf/examples/get-started/pipeline_a2dp_sink_and_hfp with either esp-adf master or v2.0-beta2

This is the one my current project is most like.

Running the named container, how would that look? I tried this but didn't get very far

```
docker run --name idf -it -v $PWD:/project -w /project --device=/dev/ttyUSB0 espressif/idf:v3.3.1
```

I get to a bash terminal and can build and flash it once through `make -j flash monitor`, but I've got to stay in the container. It stops the container when I exit. If I try docker start idf I can't exec in after the fact.

```
build@build:~/esp/hello_world$ docker start idf
idf
build@build:~/esp/hello_world$ docker exec -it idf make -j flash monitor
make: xtensa-esp32-elf-gcc: Command not found
expr: syntax error
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
Toolchain path: 
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
Project is not inside a git repository, will not use 'git describe' to determine PROJECT_VER.
App "hello-world" version: 1
make[1]: xtensa-esp32-elf-gcc: Command not found
expr: syntax error
The following Python requirements are not satisfied:
pyserial>=3.0
future>=0.15.2
pyparsing>=2.0.3,<2.4.0
Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
Alternatively, you can run "/usr/bin/python -m pip install --user -r /opt/esp/idf/requirements.txt" for resolving the issue.
/opt/esp/idf/make/project.mk:501: recipe for target 'check_python_dependencies' failed
make: *** [check_python_dependencies] Error 1
make: *** Waiting for unfinished jobs....
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
```

Similar path-style issues with idf.py
```
build@build:~/esp/hello_world$ docker start idf
idf
build@build:~/esp/hello_world$ docker exec -it idf idf.py build flash monitor
OCI runtime exec failed: exec failed: container_linux.go:346: starting container process caused "exec: \"idf.py\": executable file not found in $PATH": unknown
```

Did I do that wrong?

igrr, 2020-02-17:
Ah, I see, the entry point script doesn't get executed when doing `docker exec`... Okay, as another option, you may try to mount a named volume into the /opt/esp/idf/tools/kconfig of the container. The first time the empty volume is mounted, it will be initialized with the data from the container. To do this, add another -v argument `-v kconfig:/opt/esp/idf/tools/kconfig` to your original docker run command.

igrr, 2020-02-18:
I've tested the following and it seems to work:
```
docker run --rm -it -v $PWD:/project -v kconfig:/opt/esp/idf/tools/kconfig -w /project espressif/idf make
```

Will add this recipe to the documentation.

ftab, 2020-02-18:
Yep, that seems to do the trick!

With /dev/ttyUSB0 hooked up:

```
docker run --rm -it -v $PWD:/project -v kconfig:/opt/esp/idf/tools/kconfig -w /project --device=/dev/ttyUSB0 espressif/idf:v3.3.1 make -j flash monitor
```

Subsequent runs just flash and monitor.

I've also tested this with my custom combined IDF/ADF image and verified it works with `make -j flash monitor` and `make monitor`

I will try to update my project with cmake so I can just use `idf.py build flash monitor`

Thanks!

closed 2020-02-23 by espressif-bot

joeatbayes, 2025-03-06:
This is a real challenge when wanting to use this from a docker image running in github actions as the re-download step is super slow and we pay by the second when running git actions.