espressif/esp-idf #4791: Unable to receive  data through Uart above 120 bytes

manojsai17, 2020-02-18:
I am using ESP-WROOM-32 module of size 16Mb  with esp-idf 3.3 version .
we connected a gsm module to ESP-WROOM-32 through Uart .
Unable to receive  data through Uart above 120 bytes . I will receive maximum of 900 bytes through mqtt calls.
please suggest a solution or any code reference  to receive data above 120 bytes through Uart .

,

macdroid53, 2020-02-18:
This sounds related to issues: https://github.com/espressif/esp-idf/issues/4537

There is a lot of detail about this in that issue, but, the short version: 120 is the threshold trigger to move received data from the FIFO to a buffer which is presented to the user code by uart.c. Assuming the processor can keep up with the baud rate and remove 120 bytes for each event then the user code has to remove 120 bytes and clear the buffer.

But, the issue above will cause you another problem, since it is only triggered each time 120 bytes are received, any number of received bytes not evenly divisible by 120 will loose the remaining bytes in the buffer. In your case, if your code was to remove 120 bytes each trigger, when the break is detected there will be 60 bytes left in the FIFO that are never transferred to the buffer and since the 120 threshold is not met, those 60 bytes will be lost.

koobest, 2020-02-19:
Hi, @manojsai17 
Is your problem the same as mentioned by macdroid53 ？would be greatly appreciated if you can provide more details?

thanks !!

manojsai17, 2020-02-21:
> Hi, @manojsai17
> Is your problem the same as mentioned by macdroid53 ？would be greatly appreciated if you can provide more details?
> 
> thanks !!

could you provide the sample reference code for above logic

koobest, 2020-02-24:
Hi,
This issue is the same with these two:
https://github.com/espressif/esp-idf/issues/4820
https://github.com/espressif/esp-idf/issues/4807 
please close it.

thanks !!.

Alvin1Zhang, 2020-03-17:
@manojsai17 Thanks for reporting. As the issue is duplicate, will close. Thanks.

closed 2020-03-17 by Alvin1Zhang