espressif/esp-idf #4792: Using esp-idf command prompt for building projects in folders other than esp-idf folder

Zeni241, 2020-02-18:
A rather basic (stupid) question.
I used ESP-IDF Tools Installer version 2.3 to successfully install required tools and esp-idf 4.0.
I can configure, build, flash projects if they are in some sub folder of esp-idf folder. 
But I can't figure out how to work on projects which are in some folder other than esp-idf folder.

The great thing about " esp-idf command prompt" is that its blazing fast as compared to working in eclipse or MINGW32.

igrr, 2020-02-18:
Could you please explain what issue you are facing? You can't change into the directory of your project, or the build returns some error if the project is not within IDF directory?

Zeni241, 2020-02-18:
I can't change into directory other than C:\Users\xxxx\esp-idf>.

igrr, 2020-02-18:
Can you paste your command and the output? I.e. what happens when you are trying to change to another directory?

igrr, 2020-02-18:
Keep in mind that if you are changing to a directory on a different drive, you need to change both the directory and the drive, e.g.
```
d:
cd d:\work
```

Zeni241, 2020-02-18:
C:\Users\xxxx\esp-idf>cd d:
D:\

C:\Users\xxxx\esp-idf>cd aaa
The system cannot find the path specified.

C:\Users\xxxx\esp-idf>cd d:\aaa

C:\Users\xxxx\esp-idf>ls
'ls' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\xxxxx\esp-idf>

Zeni241, 2020-02-18:
Using Python in C:\Users\xxxxx\AppData\Local\Programs\Python\Python38\
Python 3.8.1
Using Git in C:\Program Files\Git\cmd\
git version 2.23.0.windows.1
Setting IDF_PATH: C:\Users\xxxxx\esp-idf

Adding ESP-IDF tools to PATH...
    C:\Users\xxxxx\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin
    C:\Users\xxxxx\.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin
    C:\Users\xxxxx\.espressif\tools\cmake\3.13.4\bin
    C:\Users\xxxxx\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin
    C:\Users\xxxxx\.espressif\tools\mconf\v4.6.0.0-idf-20190628\
    C:\Users\xxxxx\.espressif\tools\ninja\1.9.0\
    C:\Users\xxxxx\.espressif\tools\idf-exe\1.0.1\
    C:\Users\xxxxx\.espressif\tools\ccache\3.7\
    C:\Users\xxxxx\.espressif\python_env\idf4.0_py3.8_env\Scripts
    C:\Users\xxxxx\esp-idf\tools

Checking if Python packages are up to date...
Python requirements from C:\Users\xxxxx\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\Users\xxxxx\esp-idf>cd d:
D:\

C:\Users\xxxxx\esp-idf>cd aaa
The system cannot find the path specified.

C:\Users\xxxxx\esp-idf>cd d:\aaa

C:\Users\xxxxx\esp-idf>ls
'ls' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\xxxxx\esp-idf>

Zeni241, 2020-02-18:
**C:\Users\xxxxx\esp-idf>cd D:\aaa\TestDrive**

**C:\Users\xxxxx\esp-idf>idf.py build**
Checking Python dependencies...
Python requirements from C:\Users\xxxxx\esp-idf\requirements.txt are satisfied.
Executing action: all (aliases: build)
**Running cmake in directory c:\users\xxxxx\esp-idf\build**
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 c:\users\xxxxx\esp-idf"...
Warn about uninitialized values.
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Didn't find assembler
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_ASM_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "C:/Users/xxxxx/esp-idf/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/xxxxx/esp-idf/build/CMakeFiles/CMakeError.log".
**cmake failed with exit code 1**

**C:\Users\xxxxx\esp-idf>**

igrr, 2020-02-18:
You need to type two lines, as follows:
```
cd d:\aaa
d:
```

Zeni241, 2020-02-18:
**_C:\Users\xxxxx\esp-idf>cd D:\aaa\TestDrive_**

**_C:\Users\xxxxx\esp-idf>d:_**

D:\aaa\TestDrive>idf.py build
Checking Python dependencies...
Python requirements from C:\Users\xxxxx\esp-idf\requirements.txt are satisfied.
Executing action: all (aliases: build)
Running ninja in directory d:\aaa\testdrive\build
Executing "ninja all"...
[1/3] Performing build step for 'bootloader'
ninja: no work to do.

Project build complete. To flash, run this command:
'
'
'
'
or run 'idf.py -p (PORT) flash'

D:\aaa\TestDrive>idf.py -p COM4 flash
Checking Python dependencies...
Python requirements from C:\Users\xxxxx\esp-idf\requirements.txt are satisfied.
Adding flash's dependency "all" to list of actions
Executing action: all (aliases: build)
Running ninja in directory d:\aaa\testdrive\build
Executing "ninja all"...
[1/3] Performing build step for 'bootloader'
ninja: no work to do.
Executing action: flash
Running esptool.py in directory d:\aaa\testdrive\build
.
.
.
.
.
esptool.py v2.8
Serial port COM4
Connecting.....
Detecting chip type... ESP32
.......
.
.
.
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 3072 bytes to 103...
_**Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 1581.4 kbit/s)...**_
Hash of data verified.
Compressed 25344 bytes to 14948...
**_Wrote 25344 bytes (14948 compressed) at 0x00001000 in 0.4 seconds (effective 568.5 kbit/s)..._**
Hash of data verified.
Compressed 147216 bytes to 76545...
**_Wrote 147216 bytes (76545 compressed) at 0x00010000 in 1.8 seconds (effective 653.1 kbit/s)..._**
Hash of data verified.

Leaving...

It works!!!! Thanks you. (But it used to be simple in mingw32. I hope other commands (like ls) will also in it in future).

The fact is esp-idf command prompt is blazing fast as compared to eclipse etc.

**Thanks again igrr.**

closed 2020-02-18 by Zeni241