espressif/esp-idf #4793: unknown type name 'ip4_addr_t' and ip6_addr_t

andrewcch1226, 2020-02-18:
Hi, I'm trying to migrate my existing firmware from esp-idf v3 to esp-idf v4. After that I encounter unknown type name 'ip4_addr_t' and ip6_addr_t error which come from esp-idf tcpip_adapter itself.

[image]

david-cermak, 2020-02-21:
Hi @andrewcch1226 

Please make sure that the compatibility option is enabled in the menuconfig (this should be enabled by default):

`ESP NETIF Adapter` -> `Enable backward compatible tcpip_adapter interface`

WIth that option ON, you should be able to compile existing applications, with deprecation warnings though. For more details please refer to the migration manual [here](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/tcpip_adapter_migration.html) to completely move from tcpip-adapter to esp-netif.

andrewcch1226, 2020-02-24:
Hi @david-cermak 
Thanks for the helping out, I'm using stable version v4.0 for my development and i did not found the option from the menuconfig.

david-cermak, 2020-02-25:
@andrewcch1226 Ahh, sorry I was under the impression you're on IDF v4.1 and using the esp-netif instead of `tcpip_adapter`.

In case of v4.0 the tcpip_adapter is still the IDF network interface layer and should include the `lwip/ip_addr.h`. Can you please post the error you are getting when compiling in command line, just to rule out potential eclipse plugin issues?

andrewcch1226, 2020-02-26:
@david-cermak I have included the `lwip/ip_addr.h `. By the way, the error was dismissed by changing `#if LWIP_IPV4` to `#if 1` from `lwip/ip4_addr.h` and `#if LWIP_IPV6` to `#if 1` from `lwip/ip6_addr.h` .

david-cermak, 2020-02-28:
This is really strange, as both these options are hardcoded to `1` in the version you've referenced, `release/v4.0`. Here are the links to the code @`v4.0`:
* https://github.com/espressif/esp-idf/blob/release/v4.0/components/lwip/port/esp32/include/lwipopts.h#L654
* https://github.com/espressif/esp-lwip/blob/066ffe0abb83eee47808bf77c1bcfef51ad077fe/src/include/lwip/opt.h#L730

Is it not possible that you redefine these flags somewhere? project makefile, environment, lwip-hook-include...?

david-cermak, 2020-03-05:
@andrewcch1226 Do you have any other comment or request or could the issue be closed?

andrewcch1226, 2020-03-06:
ya, I think the issue was solved. thanks for helping out. :)

closed 2020-03-06 by andrewcch1226