espressif/esp-idf #4794: HFP_AG example crash

KollarRichard, 2020-02-18:
## Environment

- Development Kit :       ESP32-Wrover-Kit
- Kit version :                  v4.1
- Module or chip used : ESP32-WROVER-B]
- IDF version :                 v4.1-dev-1935-g647cb628a
- Build System :              Make
- Compiler version :       xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System :     macOS
- Using an IDE? :            No
- Power Supply :            USB

## Problem Description

Example crashes little while after audio connection.

### Expected Behavior

Incoming callback sends sine wave.

### Actual Behavior

Example crashes.

### Steps to reproduce

1. Run example with HCI and wide band speech.
2. Apply fix from #4660 
3. Connect to your handsfree device.

### Code to reproduce this issue

HFP_AG [example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/hfp_ag) 

## Debug Logs

```

CORRUPT HEAP: multi_heap.c:477 detected at 0x3f808fd4
abort() was called at PC 0x40091ea9 on core 0
0x40091ea9: multi_heap_assert at /Users/richard.kollar/esp/esp-idf/components/heap/multi_heap_platform.h:67
 (inlined by) multi_heap_free_impl at /Users/richard.kollar/esp/esp-idf/components/heap/multi_heap.c:477

ELF file SHA256: c452bead2707bd2ccb99df623cce27a06addcadd0bff3fbda97f76f3b5198b1e

Backtrace: 0x400925cd:0x3ffd1120 0x40092975:0x3ffd1140 0x40091ea9:0x3ffd1160 0x4008731e:0x3ffd1180 0x40095909:0x3ffd11a0 0x400fe771:0x3ffd11c0 0x400ff3c5:0x3ffd11e0 0x400fda05:0x3ffd1200 0x400fdb6e:0x3ffd1220 0x401282de:0x3ffd1240 0x40126469:0x3ffd1260 0x40095b49:0x3ffd1280
0x400925cd: invoke_abort at /Users/richard.kollar/esp/esp-idf/components/esp32/panic.c:157

0x40092975: abort at /Users/richard.kollar/esp/esp-idf/components/esp32/panic.c:174

0x40091ea9: multi_heap_assert at /Users/richard.kollar/esp/esp-idf/components/heap/multi_heap_platform.h:67
 (inlined by) multi_heap_free_impl at /Users/richard.kollar/esp/esp-idf/components/heap/multi_heap.c:477

0x4008731e: heap_caps_free at /Users/richard.kollar/esp/esp-idf/components/heap/heap_caps.c:273

0x40095909: free at /Users/richard.kollar/esp/esp-idf/components/newlib/heap.c:47

0x400fe771: bta_ag_sco_event at /Users/richard.kollar/esp/esp-idf/components/bt/host/bluedroid/bta/hf_ag/bta_ag_sco.c:772

0x400ff3c5: bta_ag_ci_sco_data at /Users/richard.kollar/esp/esp-idf/components/bt/host/bluedroid/bta/hf_ag/bta_ag_sco.c:1672

0x400fda05: bta_ag_sm_execute at /Users/richard.kollar/esp/esp-idf/components/bt/host/bluedroid/bta/hf_ag/bta_ag_main.c:910 (discriminator 2)

0x400fdb6e: bta_ag_hdl_event at /Users/richard.kollar/esp/esp-idf/components/bt/host/bluedroid/bta/hf_ag/bta_ag_main.c:965 (discriminator 3)

0x401282de: bta_sys_event at /Users/richard.kollar/esp/esp-idf/components/bt/host/bluedroid/bta/sys/bta_sys_main.c:499

0x40126469: osi_thread_run at /Users/richard.kollar/esp/esp-idf/components/bt/common/osi/thread.c:68

0x40095b49: vPortTaskWrapper at /Users/richard.kollar/esp/esp-idf/components/freertos/port.c:143

```

### Question

From backtrace i found this line and i dont understand what is happening [here](https://github.com/espressif/esp-idf/blob/ca735340f93eec0d14fb301a9d9d7305d700b410/components/bt/host/bluedroid/bta/hf_ag/bta_ag_sco.c#L1668).
int UNUSED() function is called and from description i read that it is returning void. Why? I believe that could be the reason of crashing.

AbnerFederer, 2020-02-20:
Hi, 

I think you are right.
The `p_sbc` cannot be `UNUSED`. 

Thanks
Abner

AbnerFederer, 2020-02-24:
Hi, @KollarRichard 

Please try this patch.

[ag_crash.zip](https://github.com/espressif/esp-idf/files/4243591/ag_crash.zip)

Thanks
Abner

KollarRichard, 2020-02-26:
Thank you, I will let you know if it helps.

KollarRichard, 2020-03-11:
works with version4.1 . Thank you.

closed 2020-03-11 by KollarRichard